lwfinger/rtw89 #191: Drivers install don't work after kernel rebuild

jjepson2, 2022-11-17:
Here is the order I've done things in:

1. Install Ubuntu 22.04.1 on a Lenovo Slim 7 Pro X with default install, default kernel, etc.
2. Try installing wifi drivers from rtw89 - this worked and let me use wifi

Then I decided to make some patches to the kernel to be able to use laptop keyboard and microphone. This led to the following steps:

1. Fresh Ubuntu 22.04.1 installed over the top of the old one.
2. I then downloaded the kernel source using sudo apt source linux-image-unsigned-$(uname -r) .  Apt proceeded to download the source from the 'linux' package to my home directory.  (The 'linux' source was the same source version as uname -r).
3. I then modified two files in the source in my home directory.  Namely I modified sound/soc/amd/yc/acp6x-mach.c and drivers/acpi/resource.c as follows:

diff --git a/drivers/acpi/resource.c b/drivers/acpi/resource.c
index c2d4947844250..510cdec375c4d 100644
--- a/drivers/acpi/resource.c
+++ b/drivers/acpi/resource.c
@@ -416,6 +416,16 @@ static bool acpi_dev_irq_override(u32 gsi, u8 triggering, u8 polarity,
 {
 	int i;
 
+#ifdef CONFIG_X86
+	/*
+	 * IRQ override isn't needed on modern AMD Zen systems and
+	 * this override breaks active low IRQs on AMD Ryzen 6000 and
+	 * newer systems. Skip it.
+	 */
+	if (boot_cpu_has(X86_FEATURE_ZEN))
+		return false;
+#endif
+
 	for (i = 0; i < ARRAY_SIZE(skip_override_table); i++) {
 		const struct irq_override_cmp *entry = &skip_override_table[i];

[diff](https://lore.kernel.org/all/20220920171320.7776-1-mario.limonciello@amd.com/#iZ2e.:20220920171320.7776-1-mario.limonciello::40amd.com:1sound:soc:amd:yc:acp6x-mach.c) --git a/sound/soc/amd/yc/acp6x-mach.c b/sound/soc/amd/yc/acp6x-mach.c
index 5eab3baf3573..2cb50d5cf1a9 100644
--- a/sound/soc/amd/yc/acp6x-mach.c
+++ b/sound/soc/amd/yc/acp6x-mach.c
@@ -171,6 +171,13 @@ static const struct dmi_system_id yc_acp_quirk_table[] = {
 			DMI_MATCH(DMI_PRODUCT_NAME, "21J6"),
 		}
 	},
+	{
+		.driver_data = &acp6x_card,
+		.matches = {
+			DMI_MATCH(DMI_BOARD_VENDOR, "LENOVO"),
+			DMI_MATCH(DMI_PRODUCT_NAME, "82"),
+		}
+	},
 	{
 		.driver_data = &acp6x_card,
 		.matches = {

4.  Then I built the kernel using the instructions given here: https://wiki.ubuntu.com/Kernel/BuildYourOwnKernel with no configuration changes (so didn't need to run 'LANG=C fakeroot debian/rules editconfigs')
5. Then I invoked apt remove --purge to remove the current linux-headers, linux-headers-generic, linux-image, linux-module, and linux-module-extra packages.  
6. I then installed all the debian files which were produced by step 4 above using 'dpkg -i *.deb' .  Specifically, these packges: 
linux-buildinfo-5.15.0-53-generic_5.15.0-53.59_amd64.deb
linux-cloud-tools-5.15.0-53_5.15.0-53.59_amd64.deb
linux-cloud-tools-5.15.0-53-generic_5.15.0-53.59_amd64.deb
linux-headers-5.15.0-53_5.15.0-53.59_all.deb
linux-headers-5.15.0-53-generic_5.15.0-53.59_amd64.deb
linux-image-unsigned-5.15.0-53-generic_5.15.0-53.59_amd64.deb
linux-modules-5.15.0-53-generic_5.15.0-53.59_amd64.deb
linux-modules-extra-5.15.0-53-generic_5.15.0-53.59_amd64.deb
linux-modules-iwlwifi-5.15.0-53-generic_5.15.0-53.59_amd64.deb
linux-tools-5.15.0-53_5.15.0-53.59_amd64.deb
linux-tools-5.15.0-53-generic_5.15.0-53.59_amd64.deb

7. I then securely signed the kernel and restarted (which also logged the MOK key)
8. Then upon doing the exact same install instructions for rtw89, now I am getting the following errors when trying to load the module:

Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152042] rtw89core: disagrees about version of symbol ieee80211_stop_tx_ba_session
[... 46 lines cut ...]
Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152206] rtw89core: Unknown symbol ieee80211_alloc_hw_nm (err -22)
Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152208] rtw89core: disagrees about version of symbol ieee80211_tx_dequeue
Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152208] rtw89core: Unknown symbol ieee80211_tx_dequeue (err -22)
Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152215] rtw89core: disagrees about version of symbol ieee80211_stop_queues
Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152216] rtw89core: Unknown symbol ieee80211_stop_queues (err -22)
Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152223] rtw89core: disagrees about version of symbol ieee80211_scan_completed
Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152224] rtw89core: Unknown symbol ieee80211_scan_completed (err -22)
Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152226] rtw89core: disagrees about version of symbol ieee80211_iterate_active_interfaces_atomic
Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152226] rtw89core: Unknown symbol ieee80211_iterate_active_interfaces_atomic (err -22)
Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152234] rtw89core: disagrees about version of symbol ieee80211_unregister_hw
Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152234] rtw89core: Unknown symbol ieee80211_unregister_hw (err -22)
Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152241] rtw89core: disagrees about version of symbol ieee80211_beacon_get_tim
Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152241] rtw89core: Unknown symbol ieee80211_beacon_get_tim (err -22)
Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152245] rtw89core: disagrees about version of symbol ieee80211_channel_to_freq_khz
Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152246] rtw89core: Unknown symbol ieee80211_channel_to_freq_khz (err -22)
Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152254] rtw89core: disagrees about version of symbol cfg80211_chandef_create
Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152254] rtw89core: Unknown symbol cfg80211_chandef_create (err -22)
Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152258] rtw89core: disagrees about version of symbol ieee80211_return_txq
Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152259] rtw89core: Unknown symbol ieee80211_return_txq (err -22)
Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152261] rtw89core: disagrees about version of symbol ieee80211_queue_work
Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152261] rtw89core: Unknown symbol ieee80211_queue_work (err -22)
Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152270] rtw89core: disagrees about version of symbol ieee80211_rx_napi
Nov 17 13:51:47 joseph-Slim-7-ProX-14ARH7 kernel: [   80.152270] rtw89core: Unknown symbol ieee80211_rx_napi (err -22)

Would you have any ideas of how to fix this issue??  Thanks.

lwfinger, 2022-11-17:
Your kernel headers do not match the running kernel.

Does 'ls -l /lib/modules/$(uname -r)/build' point to your new headers?

The Debian way is way too complicated. I have one system running Debian and all I do is cd to the kernel source, make sure the configuration is correct, and then do 'make -jX' where X is 2*#CPU +1. Once that finishes correctly, then 'sudo make modules_install install'. That works every time.

jjepson2, 2022-11-17:
Yes, that command gives as output: lrwxrwxrwx 1 root root 40 Nov 17 12:47 /lib/modules/5.15.0-53-generic/build -> /usr/src/linux-headers-5.15.0-53-generic .

But of course my recompiled version has the same version number.  How can I show the versions of both the headers and running kernel to compare and see if the versions don't match??

lwfinger, 2022-11-17:
The 'uname -r' part of that command shows the ID of the running kernel. The whole command I gave you points to the headers.

When you do a 'make' in the rtw89 directory, the -C part will show the headers that are being used. It should be the same.

jjepson2, 2022-11-17:
Looking at the Makefile in rtw89, all of this looks consistent.  Is there any other things I could try to debug this issue?? Also, are there any commands I can run while making or installing rtw89 to help me debug?

jjepson2, 2022-11-17:
Also, using lsmod shows that I have many other drivers seemingly loaded fine into memory.

lwfinger, 2022-11-17:
Do a 'make clean', 'make', and 'sudo make install', then reboot. Does that help?

jjepson2, 2022-11-17:
Just tried that.  It didn't seem to fix the issue.

lwfinger, 2022-11-17:
Are you getting modules from the in-kernel driver? What does 'lsmod | grep rtw' show?

jjepson2, 2022-11-18:
When I do that command it doesn't show anything.

lwfinger, 2022-11-18:
OK, you do not have the old drivers.

Make and install the kernel the way I said.

jjepson2, 2022-11-18:
Doing it that way made my kernel unbootable.

jjepson2, 2022-11-18:
There is an “out of ram” error on the boot page, which is weird.

jjepson2, 2022-11-18:
I reverted back to the kernel that works (and replaced the source and other files). Interestingly though when I did it that way, it installed a different version of the kernel.

jjepson2, 2022-11-18:
(The way you had mentioned earlier I mean)

jjepson2, 2022-11-18:
So now I can boot, but Wi-Fi still doesn’t work because the module won’t load for the same reason.  (version disagrees error)

jjepson2, 2022-11-18:
I figured out what was causing the version error.  The Debian build process produced a pkg entitled linux-modules-iwlwifi-5.15.0-53-generic_5.15.0-53.59_amd64.deb .  I uninstalled this, and it was able to find the symbol definitions from one of the other default mods that had been built, without version conflicts.  However, now it can load the modules, but there is still no wifi after reboot, even with the modules loaded.  Would you have any thing I could look at regarding that??

jjepson2, 2022-11-18:
specifically if looks like the modules loaded are rtw_8852be , rtw_8852b, rtw89pci, rtw89core .  

I also did try another make clean, make, sudo make install, and then reboot.  This didn’t fix this problem.

jjepson2, 2022-11-18:
It looks like the issue is maybe that it can’t find the appropriate bin files in the firmware directory?? Is it supposed to install more than just the rtw8852a_fw file in that directory?? Because it’s getting a “direct firmware load failed with error -2” error trying to load rtw8852b_fw .

jjepson2, 2022-11-18:
How do I get it to install all the .bin files in that directory? (If needed)

lwfinger, 2022-11-18:
You need to install a package with a name like linux-firmware-realtek. That will have the .bin files that are needed.

jjepson2, 2022-11-18:
It works now!  I just downloaded the cached (but deleted) .bin files from the repository here.  Just FYI, the ubuntu linux-firmware package still does not contain all the needed .bin firmware files yet for rtw89 (even though the linux-firmware package on kernel.org does).

lwfinger, 2022-11-18:
That is another reason I do not use Ubuntu!!

Glad you got it to work.

closed 2022-11-18 by lwfinger