lwfinger/rtw89 #193: Module rtw89_8852be missing after install on Arch via AUR

eqkessel, 2022-11-20:
I installed this package on Arch (kernel version 6.0.9) via the instructions listed on the README:
```bash
git clone https://aur.archlinux.org/rtw89-dkms-git.git
cd rtw89-dkms-git
makepkg -sri
```

However the module for the 8852BE (WiFi module included with my motherboard) didn't get included with the install it seems:
```bash
$ find /lib/modules/$(uname -r) -type f -name '*.ko*' | grep 'rtw\|8852'
/lib/modules/6.0.9-arch1-1/kernel/drivers/net/wireless/realtek/rtw89/rtw89_8852ce.ko.zst
/lib/modules/6.0.9-arch1-1/kernel/drivers/net/wireless/realtek/rtw89/rtw89_8852a.ko.zst
/lib/modules/6.0.9-arch1-1/kernel/drivers/net/wireless/realtek/rtw89/rtw89_8852ae.ko.zst
/lib/modules/6.0.9-arch1-1/kernel/drivers/net/wireless/realtek/rtw89/rtw89_pci.ko.zst
/lib/modules/6.0.9-arch1-1/kernel/drivers/net/wireless/realtek/rtw89/rtw89_8852c.ko.zst
/lib/modules/6.0.9-arch1-1/kernel/drivers/net/wireless/realtek/rtw89/rtw89_core.ko.zst
/lib/modules/6.0.9-arch1-1/kernel/drivers/net/wireless/realtek/rtw88/rtw88_8821c.ko.zst
/lib/modules/6.0.9-arch1-1/kernel/drivers/net/wireless/realtek/rtw88/rtw88_8821ce.ko.zst
/lib/modules/6.0.9-arch1-1/kernel/drivers/net/wireless/realtek/rtw88/rtw88_8822ce.ko.zst
/lib/modules/6.0.9-arch1-1/kernel/drivers/net/wireless/realtek/rtw88/rtw88_8822b.ko.zst
/lib/modules/6.0.9-arch1-1/kernel/drivers/net/wireless/realtek/rtw88/rtw88_8822c.ko.zst
/lib/modules/6.0.9-arch1-1/kernel/drivers/net/wireless/realtek/rtw88/rtw88_pci.ko.zst
/lib/modules/6.0.9-arch1-1/kernel/drivers/net/wireless/realtek/rtw88/rtw88_8723de.ko.zst
/lib/modules/6.0.9-arch1-1/kernel/drivers/net/wireless/realtek/rtw88/rtw88_core.ko.zst
/lib/modules/6.0.9-arch1-1/kernel/drivers/net/wireless/realtek/rtw88/rtw88_8822be.ko.zst
/lib/modules/6.0.9-arch1-1/kernel/drivers/net/wireless/realtek/rtw88/rtw88_8723d.ko.zst
```

This evidently causes `modprobe` to fail:
```bash
$ sudo modprobe rtw89_8852be
modprobe: FATAL: Module rtw89_8852be not found in directory /lib/modules/6.0.9-arch1-1
```

The install process doesn't seem to produce any errors:
```bash
$ makepkg -sri
==> Making package: rtw89-dkms-git 1:r274.b8a0d44-2 (Sun 20 Nov 2022 01:07:51 AM EST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Cloning rtw89 git repo...
Cloning into bare repository '/home/qynn/aur/rtw89-dkms-git/rtw89'...
remote: Enumerating objects: 1777, done.
remote: Counting objects: 100% (260/260), done.
remote: Compressing objects: 100% (99/99), done.
remote: Total 1777 (delta 190), reused 177 (delta 159), pack-reused 1517
Receiving objects: 100% (1777/1777), 4.82 MiB | 21.76 MiB/s, done.
Resolving deltas: 100% (1316/1316), done.
  -> Found dkms.conf
==> Validating source files with sha256sums...
    rtw89 ... Skipped
    dkms.conf ... Passed
==> Extracting sources...
  -> Creating working copy of rtw89 git repo...
Cloning into 'rtw89'...
done.
Switched to a new branch 'makepkg'
==> Starting pkgver()...
==> Updated version: rtw89-dkms-git 1:r297.a2c1135-1
==> Starting build()...
==> Entering fakeroot environment...
==> Starting package()...
removed '/home/qynn/aur/rtw89-dkms-git/pkg/rtw89-dkms-git/usr/src/rtw89-r297.a2c1135/.git/hooks/pre-commit.sample'
removed '/home/qynn/aur/rtw89-dkms-git/pkg/rtw89-dkms-git/usr/src/rtw89-r297.a2c1135/.git/hooks/fsmonitor-watchman.sample'
removed '/home/qynn/aur/rtw89-dkms-git/pkg/rtw89-dkms-git/usr/src/rtw89-r297.a2c1135/.git/hooks/commit-msg.sample'
removed '/home/qynn/aur/rtw89-dkms-git/pkg/rtw89-dkms-git/usr/src/rtw89-r297.a2c1135/.git/hooks/pre-push.sample'
removed '/home/qynn/aur/rtw89-dkms-git/pkg/rtw89-dkms-git/usr/src/rtw89-r297.a2c1135/.git/hooks/pre-rebase.sample'
removed '/home/qynn/aur/rtw89-dkms-git/pkg/rtw89-dkms-git/usr/src/rtw89-r297.a2c1135/.git/hooks/applypatch-msg.sample'
removed '/home/qynn/aur/rtw89-dkms-git/pkg/rtw89-dkms-git/usr/src/rtw89-r297.a2c1135/.git/hooks/prepare-commit-msg.sample'
[... 71 lines cut ...]
warning: rtw89-dkms-git-1:r297.a2c1135-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (1) rtw89-dkms-git-1:r297.a2c1135-1

Total Installed Size:  5.71 MiB
Net Upgrade Size:      0.00 MiB

:: Proceed with installation? [Y/n] y
(1/1) checking keys in keyring                     [######################] 100%
(1/1) checking package integrity                   [######################] 100%
(1/1) loading package files                        [######################] 100%
(1/1) checking for file conflicts                  [######################] 100%
(1/1) checking available disk space                [######################] 100%
:: Running pre-transaction hooks...
(1/1) Remove upgraded DKMS modules
:: Processing package changes...
(1/1) reinstalling rtw89-dkms-git                  [######################] 100%
:: Running post-transaction hooks...
(1/2) Arming ConditionNeedsUpdate...
(2/2) Install DKMS modules
```

Any idea why this doesn't include the 8852be module? Other issues in this repository suggest it exists. How can I can I get this module installed on my system? Thanks.

Greek64, 2022-11-20:
> Any idea why this doesn't include the 8852be module?

Probably because the latest commit by the author (a2c113501a5a988683f236df1aa79ce40f29e1be) removes all firmware from being installed, because, and I quote:
> The linux-firmware repo has caught up

Now, don't ask me why the author would in essence break functionality, just because upstream has caught up.
I personally would just archive this repo, but whatever.

Just checkout a previous commit and compile again...

lwfinger, 2022-11-20:
@Greek64 is not correct. I did not break functionality. The firmware I deleted is now available from linux-firmware, and should be included with the kernel-firmware-realtek package of all up to date distros. The problem is that driver rtw89-8852be was not built as seen below.

@eqkessel - The driver for the RTW8852BE will not be in the kernel until 6.1.0 is released. Until you have that kernel, you need to use this repo.

piyushmanolkar, 2022-12-05:
> @Greek64 is not correct. I did not break functionality. The firmware I deleted is now available from linux-firmware, and should be included with the kernel-firmware-realtek package of all up to date [...]

I'm having the same problem, My wifi adapter suddenly stopped getting recognized, So I reinstalled this driver, rebooted and still the device is not getting detected, Can you please define the steps needed to get this working.

I have Manjaro Linux, 5.15 Kernel.

`ip a` output -
`1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eno1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether e0:70:ea:c7:52:7e brd ff:ff:ff:ff:ff:ff
    altname enp2s0
    inet 192.168.1.9/24 brd 192.168.1.255 scope global dynamic noprefixroute eno1
       valid_lft 84593sec preferred_lft 84593sec
    inet6 2401:4900:1c9a:5ab0:306c:5841:98e3:82da/64 scope global dynamic noprefixroute 
       valid_lft 76857sec preferred_lft 76857sec
    inet6 fe80::38c2:60c2:b152:50d/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
`

ferrao, 2022-12-09:
> The firmware I deleted is now available from linux-firmware, and should be included with the kernel-firmware-realtek package of all up to date distros. 

Same problem is present on a recent (few days..) ArchLinux install.

> If the above fails, go a 'git pull' and look at the end of README.md. It provides a link that you may use to download and install the firmware.

What link are you referring to @lwfinger ? I do not see any link regarding realtek firmware on the README file.

lwfinger, 2022-12-10:
You should complain to ArchLinux about the missing linux-firmware-realtek. My distro has had it ever since linux-firmware was split into different packages.

I may have missed a push. Now the last part of README.md should say

```bash
Missing firmware for RTW8852BE

The necessary firmware file should be in package firmware-realtek or linux-firmware-realtek;
however some versions of some distros have been extremely slow to pick up this firmware file,
even though it has been in the official linux-firmware repo at vger.kernel.org since
Oct. 27, 2022. If your distro is one of these, you can download the firmware from
https://lwfinger.com/download/rtw8852b_fw.bin, and copy it to /lib/firmware/rtw89/.
```

ferrao, 2022-12-10:
> My distro has had it ever since linux-firmware was split into different packages.

Out of curiosity, what is the distro you are using @lwfinger ?

> I may have missed a push. 

No problem, thank you so much for your quickly reply and action!

ferrao, 2022-12-10:
@lwfinger Actually I just realized the firmware **is** already on Arch:

`/lib/firmware/rtw89/rtw8852b_fw.bin.xz`

But for some reason, the AUR `rtw89-dkms-git` package does not build the driver, I'll keep digging and post a solution as soon as I find it..

lwfinger, 2022-12-10:
I use openSUSE Tumbleweed. The kernel does not yet include rtw89-8852be, which is likely why the AUR package does not build a driver for the RTW8852BE.

ferrao, 2022-12-10:
> The kernel does not yet include rtw89-8852be, which is likely why the AUR package does not build a driver for the RTW8852BE.

I don't get that.. The AUR package clones your repo. Why would it not build that specific driver, when I can build it manually without a problem?

It uses `dkms`, which I have not yet fully understood, but on the `dkms.conf` file inside the AUR package, it does exclude that module. But I tried to add it manually, did not made any difference.

lwfinger, 2022-12-10:
You would need to reinstall dkms.conf after changing it, but beyond that I have no knowledge of dkms.

ferrao, 2022-12-10:
Anyway, I have this up&running **without** using the `AUR` package... Just compiling & installing by hand.
So I guess the problem is really up to the AUR package maintainer @lwfinger

closed 2022-12-10 by lwfinger

lwfinger, 2022-12-16:
The driver will first appear in the kernel in version 6.2, probably in mid February. I was wrong above.