lwfinger/rtw89 #196: latest update 2cad718

salimp2009, 2022-11-24:
hi, 
I am on Manjaro and the latest update with commit  #2cad718 and the wireless is gone.
I went into the directory and tried to run "make" manually
 but it gives this error ; 

```
usr/src/rtw89-r305.2cad718/phy.c: In function ‘rtw89_phy_ul_tb_ctrl_check’:
/usr/src/rtw89-r305.2cad718/phy.c:2993:17: error: ‘struct ieee80211_vif’ has no member named ‘cfg’
 2993 |         if (!vif->cfg.assoc)
      |                 ^~
make[2]: *** [scripts/Makefile.build:289: /usr/src/rtw89-r305.2cad718/phy.o] Error 1
make[1]: *** [Makefile:1900: /usr/src/rtw89-r305.2cad718] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.15.78-1-MANJARO/build'
make: *** [Makefile:86: all] Error 2
```
salim

lwfinger, 2022-11-24:
Yes, vif->cfg was not added until kernel 6.0.

Do a pull, and it should build now.

salimp2009, 2022-11-24:
i believe the function in the error message 
` rtw89_phy_ul_tb_ctrl_check` 
was added thru commit #ad6f83c

[](https://github.com/lwfinger/rtw89/commit/ad6f83cecc51f2b7680a686a4c16d4b4f070cd14#:~:text=if%20(!vif%2D%3E,return%3B)

lwfinger, 2022-11-24:
I fixed it right now, perhaps. Do a pull and try again.

salimp2009, 2022-11-24:
should i use package manager version or the one from this repo .

salimp2009, 2022-11-24:
> I fixed it right now, perhaps. Do a pull and try again.

i did pull from the this git repo and it works.

shall I use the Arch version since I am on Manjaro ;

closed 2022-11-24 by salimp2009