lwfinger/rtw89 #198: rtw89 for Yoga 6

Shahin-rmz, 2022-11-26:
Hello, first of all thanks for providing Driver for Lenovo devices.
I am using lenovo Yoga 6, and installed Ubuntu 22.04 LTS on it.
Wifi does not work, so I pull and `make`the repo.
There was an issue finding `rtw_8852ae`
loading the module with ` modeprobe`  did not work.
however, from [this](https://askubuntu.com/a/1337456) community answer, I thought that I should load rtw89, **not rtw88**. But there is no such file,
would be thankful if got any helps.

Can´t live in Windows environment with WSL and such things.

[image]

Shahin-rmz, 2022-11-26:
```
[sudo] password for shahin: 
insmod /lib/modules/5.15.0-53-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw_8852a.ko 
insmod /lib/modules/5.15.0-53-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw_8852ae.ko 
``` 
this is what I got after loading the module.
still no working.

lwfinger, 2022-11-26:
Please post the output of 'lspci -nn'.

Shahin-rmz, 2022-11-27:
You diagnosed it correctly.
The driver was not compatible with my device.
Here is the result of the command that you told me.

```
03:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. Device [10ec:b852]
```
after searching for the driver of the device from [this ](https://askubuntu.com/a/1415817) link and install it, it is now working seamlessly.
Thanks.
Issue solved and closed

closed 2022-11-27 by Shahin-rmz

lwfinger, 2022-11-27:
For your information, the RTW8852BE driver in this repo is much, much better than the one from HRex39.

Shahin-rmz, 2022-11-28:
> the RTW8852BE driver in this repo is much, much better than the one from HRex39.

So I have to uninstall the current driver, and at the end do `sudo modprobe rtw8852` right?

lwfinger, 2022-11-28:
Yes.

Shahin-rmz, 2022-12-01:
Would you please tell me which driver is exact match of my wifi module?
It seems I can't make it run.

lwfinger, 2022-12-01:
You need to make and install the code from this repo. Your drivers will be rtw_8852be, rtw_8852b, rtw89core, and rtw89pci. If those drivers are loaded (use 'lsmod | grep rtw' to see), then you need to check dmesg to see what happened.