lwfinger/rtw89 #202: Loading wifi modules had no effects

hilliao, 2022-12-11:
on my Ubuntu 22.04.1 Kernel 5.15.0-56-generic, [Lenovo - Ideapad 1i 14.0" HD Laptop - Celeron N4020 - 4GB Memory - 64GB eMMC - Cloud Grey](https://www.bestbuy.com/site/lenovo-ideapad-1i-14-0-hd-laptop-celeron-n4020-4gb-memory-64gb-emmc-cloud-grey/6511947.p?skuId=6511947). All the steps succeeded from build to installation to load kernel modules. However, no wifi is detected via command ip a or from the Unity UI. Security boot is disabled. Do you want to see the full command line output?

lwfinger, 2022-12-11:
You do not have the firmware installed. Ubuntu has been very slow to pick up the firmware from the linux-firmware repo.

In README.md is an alternative solution.

hilliao, 2022-12-11:
Thanks for the quick response. I am trying Ubuntu 22.10 with a newer kernel now. I followed the README.md steps to add options:

$ uname -a
Linux hil-ideapad 5.19.0-21-generic #21-Ubuntu SMP PREEMPT_DYNAMIC Wed Oct 12 18:33:17 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
$ cat /etc/modprobe.d/rtw_8852be.conf
options rtw89pci disable_aspm_l1
options rtw89pci disable_aspm_l1ss

$ sudo modprobe -v rtw_8852be
insmod /lib/modules/5.19.0-21-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw89pci.ko disable_aspm_l1 disable_aspm_l1ss 
insmod /lib/modules/5.19.0-21-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw_8852b.ko 
insmod /lib/modules/5.19.0-21-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw_8852be.ko

### $ sudo dmesg
[ 4813.125590] CPU: 1 PID: 16837 Comm: modprobe Tainted: G        W  OE     5.19.0-21-generic #21-Ubuntu
[ 4813.125594] Hardware name: LENOVO 82V6/LNVNB161216, BIOS KKCN12WW 06/21/2022
[ 4813.125596] RIP: 0010:dev_addr_check.cold+0x65/0x96
[ 4813.125600] Code: 01 e8 92 fe f4 ff 0f 0b 49 c7 c4 4f ec e5 99 80 3b 00 75 27 48 c7 c6 5a ec e5 99 4c 89 e2 48 c7 c7 e8 e5 ef 99 e8 6e fe f4 ff <0f> 0b e9 5b e2 d1 ff 4c 8b 24 c5 a0 21 b4 99 eb d4 be 25 00 00 00
[ 4813.125603] RSP: 0018:ffffaaf8836f7be8 EFLAGS: 00010246
[ 4813.125605] RAX: 0000000000000000 RBX: ffff9e850ff8f000 RCX: 0000000000000000
[ 4813.125607] RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
[ 4813.125609] RBP: ffffaaf8836f7bf8 R08: 0000000000000000 R09: 0000000000000000
[ 4813.125610] R10: 0000000000000000 R11: 0000000000000000 R12: ffffffff99e5eedc
[ 4813.125612] R13: ffff9e850ff8f000 R14: dead000000000122 R15: ffff9e850101d150
[ 4813.125614] FS:  00007f82880b2040(0000) GS:ffff9e857c080000(0000) knlGS:0000000000000000
[ 4813.125617] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 4813.125618] CR2: 00007f065ade1478 CR3: 000000000eae2000 CR4: 0000000000350ee0
[ 4813.125621] Call Trace:
[ 4813.125624]  <TASK>
[ 4813.125629]  dev_addr_flush+0x29/0xa0
[ 4813.125635]  free_netdev+0x88/0x1f0
[ 4813.125640]  rtw_free_netdev+0x1d/0x2c [8852be]
[ 4813.125781]  rtw_os_ndev_free+0x3f/0x56 [8852be]
[ 4813.125900]  rtw_pci_primary_adapter_deinit+0x1a/0x50 [8852be]
[ 4813.126023]  rtw_dev_remove+0x84/0x137 [8852be]
[ 4813.126148]  pci_device_remove+0x36/0xb0
[ 4813.126152]  device_remove+0x40/0x80
[ 4813.126156]  device_release_driver_internal+0x222/0x2a0
[ 4813.126159]  ? klist_remove+0x93/0x100
[ 4813.126163]  driver_detach+0x4a/0xa0
[ 4813.126166]  bus_remove_driver+0x6c/0x100
[ 4813.126169]  driver_unregister+0x2f/0x60
[ 4813.126171]  pci_unregister_driver+0x40/0x90
[ 4813.126176]  rtw_drv_halt+0x39/0xd2c [8852be]
[ 4813.126311]  __do_sys_delete_module.constprop.0+0x18c/0x280
[ 4813.126317]  __x64_sys_delete_module+0x12/0x20
[ 4813.126319]  do_syscall_64+0x58/0x90
[ 4813.126322]  ? do_syscall_64+0x67/0x90
[ 4813.126325]  ? __rseq_handle_notify_resume+0x37/0x70
[ 4813.126329]  ? exit_to_user_mode_loop+0xe0/0x130
[ 4813.126333]  ? exit_to_user_mode_prepare+0x30/0xb0
[ 4813.126336]  ? syscall_exit_to_user_mode+0x26/0x50
[ 4813.126339]  ? __x64_sys_close+0x11/0x50
[ 4813.126343]  ? do_syscall_64+0x67/0x90
[ 4813.126345]  entry_SYSCALL_64_after_hwframe+0x63/0xcd
[ 4813.126349] RIP: 0033:0x7f828791f82b
[ 4813.126354] Code: 73 01 c3 48 8b 0d d5 65 0d 00 f7 d8 64 89 01 48 83 c8 ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa b8 b0 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d a5 65 0d 00 f7 d8 64 89 01 48
[ 4813.126356] RSP: 002b:00007ffec98131e8 EFLAGS: 00000206 ORIG_RAX: 00000000000000b0
[ 4813.126359] RAX: ffffffffffffffda RBX: 0000555aacd57110 RCX: 00007f828791f82b
[ 4813.126361] RDX: 0000000000000000 RSI: 0000000000000800 RDI: 0000555aacd57178
[ 4813.126363] RBP: 0000000000000000 R08: 00007f82879f72d0 R09: 0000000000000073
[ 4813.126364] R10: 00000000ffffffff R11: 0000000000000206 R12: 00007ffec9813230
[ 4813.126366] R13: 0000555aacd57110 R14: 0000000000000000 R15: 0000000000000000
[ 4813.126369]  </TASK>
[ 4813.126371] ---[ end trace 0000000000000000 ]---
[ 4813.149424] PHL: [MAC] [ERR] aoac report pointer null
[ 4813.176607] RTW: module exit success
[ 4813.176617] RTW: ===================== MSTAT =====================
[ 4813.176620] RTW:  TAG      alloc       peak   aloc_cnt    err_cnt
[ 4813.176622] RTW: -------------------------------------------------
[ 4813.176623] RTW:  VIR          0   19995415          0          0
[ 4813.176627] RTW:  PHY          0   21919899          0          0
[ 4813.176629] RTW:  SKB          0   18337280          0          0
[ 4813.176631] RTW:  USB          0          0          0          0
[ 4825.108126] rtw89_8852be 0000:03:00.0: Direct firmware load for rtw89/rtw8852b_fw.bin failed with error -2
[ 4825.108138] rtw89_8852be 0000:03:00.0: failed to early request firmware: -2
[ 4825.108273] rtw89_8852be 0000:03:00.0: Direct firmware load for rtw89/rtw8852b_fw.bin failed with error -2
[ 4825.108277] rtw89_8852be 0000:03:00.0: failed to request firmware
[ 4825.110740] rtw89_8852be 0000:03:00.0: failed to wait firmware completion
[ 4825.110757] rtw89_8852be 0000:03:00.0: failed to setup chip information
[ 4825.111358] rtw89_8852be: probe of 0000:03:00.0 failed with error -22

It appears the options did not solve the firmware loading issue but the error from `dmesg` appeared to be more verbose. The README.md shows the firmware errors are more about BIOS related bugs, not Ubuntu related. Please advise.

lwfinger, 2022-12-12:
A newer kernel could not possible help, and seem to have introduced a new problem that I have not seen before.

The following lines from dmesg indicate that the firmware file is not available. Options will not help that either. Follow the instructions in my previous comment!

hilliao, 2022-12-12:
I missed the last section of the README.md file. I downloaded the firmware .bin file and executed the following line:

> sudo  mv '/home/hil/Downloads/rtw8852b_fw.bin'   /lib/firmware/rtw89/
> sudo modprobe -v rtw_8852be

Then I finally got to see wifi visible on the top right. It works like the kernel module at https://github.com/HRex39/rtl8852be. Thanks for the quick responses.

closed 2022-12-12 by hilliao