lwfinger/rtw89 #203: Xiaomi with Realtek b852

Jusix, 2022-12-12:
Hello,

on my new Xiaomi Book Pro 14 Ryzen with Realtek b852 using newest kernel Linux 6.1 Wifi is not recognized at all. Is the PCIe device ID missing in the driver?

```
01:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device b852
        Subsystem: Device 1d3e:c826
        Flags: fast devsel, IRQ 255, IOMMU group 11
        I/O ports at 2000 [disabled] [size=256]
        Memory at c0900000 (64-bit, non-prefetchable) [disabled] [size=1M]
        Capabilities: [40] Power Management version 3
        Capabilities: [50] MSI: Enable- Count=1/1 Maskable- 64bit+
        Capabilities: [70] Express Endpoint, MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [148] Device Serial Number 00-e0-4c-ff-fe-88-52-01
        Capabilities: [158] Latency Tolerance Reporting
        Capabilities: [160] L1 PM Substates
```

KutouAkira, 2022-12-13:
I have the same problem and the result of dmesg is
```
[    5.848748] rtw89core: loading out-of-tree module taints kernel.
[    5.866609] rtw89_8852be 0000:01:00.0: firmware: failed to load rtw89/rtw8852b_fw.bin (-2)
[    5.866616] rtw89_8852be 0000:01:00.0: Direct firmware load for rtw89/rtw8852b_fw.bin failed with error -2
[    5.866618] rtw89_8852be 0000:01:00.0: failed to early request firmware: -2
[    5.866678] rtw89_8852be 0000:01:00.0: enabling device (0000 -> 0003)
[    5.866696] rtw89_8852be 0000:01:00.0: firmware: failed to load rtw89/rtw8852b_fw.bin (-2)
[    5.866704] rtw89_8852be 0000:01:00.0: firmware: failed to load rtw89/rtw8852b_fw.bin (-2)
[    5.866705] rtw89_8852be 0000:01:00.0: Direct firmware load for rtw89/rtw8852b_fw.bin failed with error -2
[    5.866707] rtw89_8852be 0000:01:00.0: failed to request firmware
[    5.868405] rtw89_8852be 0000:01:00.0: failed to wait firmware completion
[    5.868415] rtw89_8852be 0000:01:00.0: failed to setup chip information
[    5.869076] rtw89_8852be: probe of 0000:01:00.0 failed with error -22
```
I have saw this issue #172 but my firmware has not been compressed
[image]

lwfinger, 2022-12-14:
The README.md tells you how to get the firmware if your distro refuses to provide firmware even though it is in the official linux-firmware repo.

closed 2022-12-14 by lwfinger