lwfinger/rtw89 #204: When the 8852be module operates in AP mode, how many stations can be connected?

SEUNGYEOP-KANG, 2022-12-14:
-. kernel version : 5.19.16
-. os : ubuntu 22.04.1
-. AP application : hostapd 2.10

When the 8852be module operates in AP mode, 32 stations can be connected.
It looks like there is a limitation within the driver.
Is there any way to connect more than 33 stations?

lwfinger, 2022-12-14:
What software are you using to create the AP?

SEUNGYEOP-KANG, 2022-12-14:
it is hostapd. version is 2.10.

lwfinger, 2022-12-16:
I sent your question to my contact at Realtek in Taiwan. His reply was

"We never connect more than 8 stations in AP mode, and the test is only ping okay,
so I'm not sure if it can work well with more than 32 stations.

Logically, a station occupies a MAC_ID and two security CAM entries, and current
definition are: 

  #define RTW89_MAX_MAC_ID_NUM 128

  struct rtw89_chip_info {
	u8 scam_num;	// 128

So, I think it may support up to 128, and later 64 stations use software
encryption/decryption.
But, again I'm not sure for this kind of high-end application.
"
 
My search of the source could not find any arrays dimensioned at 32.

Sorry, we do not have an answer for you.

Larry