lwfinger/rtw89 #206: RTL8852AE firmware binaries not found on Fedora 36 with kernel 6.0.15 and how to fix it.

luisfelipehsr, 2022-12-28:
I have a Lenovo laptop and I installed Fedora 36 in it. Apparently the .bin files should be provided by the distro, but they aren't. I tried for a long time to make it work and at the end of the day the problem was that after commit a275186 the .bin files were removed.

For anyone with the same error, in dmesg I got `Direct firmware load for rtw89/rtw8852a_fw.bin failed with error -` and in lspci my Wifi controller showed that there were kernel modules available, but no driver in use.

To fix it I did the usual instructions here, but I checked out commit a275186 before doing `make`:

`git checkout a275186`

Maybe the Makefile should check if the .bin are in the directory before deciding to not copy them anymore?

lijun1234567, 2022-12-28:
You can try it in kernel  6.1.1-1

luisfelipehsr, 2022-12-28:
> You can try it in kernel 6.1.1-1

I will try it when DNF decides that it wants to update the kernel. For now I am leaving as it is since it works and I am afraid of breaking Nvidia stuff with another kernel update.

lwfinger, 2022-12-28:
On Fedora, you need to install package linux-firmware-vendor. It has ALL the rtw89 firmware.

closed 2022-12-28 by lwfinger

Stenfrank, 2023-03-13:
> On Fedora, you need to install package linux-firmware-vendor. It has ALL the rtw89 firmware.

Hello, I installed linux-firmware-vendor and this same package but I can't get the RTL8852BE wifi card to work, I have this kernel 6.1.15-200.fc37.x86_64

Stenfrank, 2023-03-13:
> Are you sure that the driver is part of 6.1.15? The driver for rtw8852be was not released as part of the kernel until 6.2! Without any log information, there is no way to tell what is wrong!

How can I validate it?

lwfinger, 2023-03-13:
Look at the output os 'sudo dmesg | grep 8852'.

Stenfrank, 2023-03-13:
> Look at the output os 'sudo dmesg | grep 8852'.

I attach all the outputs:

make:
make -C /lib/modules/6.1.15-200.fc37.x86_64/build M=/home/frank/rtw89 modules
make[1]: se entra en el directorio '/usr/src/kernels/6.1.15-200.fc37.x86_64'
  CC [M]  /home/frank/rtw89/core.o
  CC [M]  /home/frank/rtw89/chan.o
  CC [M]  /home/frank/rtw89/mac80211.o
  CC [M]  /home/frank/rtw89/mac.o
  CC [M]  /home/frank/rtw89/phy.o
  CC [M]  /home/frank/rtw89/fw.o
  CC [M]  /home/frank/rtw89/cam.o
  CC [M]  /home/frank/rtw89/efuse.o
  CC [M]  /home/frank/rtw89/regd.o
  CC [M]  /home/frank/rtw89/sar.o
  CC [M]  /home/frank/rtw89/coex.o
  CC [M]  /home/frank/rtw89/ps.o
  CC [M]  /home/frank/rtw89/debug.o
  CC [M]  /home/frank/rtw89/ser.o
  CC [M]  /home/frank/rtw89/wow.o
  LD [M]  /home/frank/rtw89/rtw89core.o
  CC [M]  /home/frank/rtw89/rtw8852a.o
  CC [M]  /home/frank/rtw89/rtw8852a_table.o
  CC [M]  /home/frank/rtw89/rtw8852a_rfk.o
  CC [M]  /home/frank/rtw89/rtw8852a_rfk_table.o
  LD [M]  /home/frank/rtw89/rtw_8852a.o
  CC [M]  /home/frank/rtw89/rtw8852ae.o
  LD [M]  /home/frank/rtw89/rtw_8852ae.o
  CC [M]  /home/frank/rtw89/rtw8852b.o
  CC [M]  /home/frank/rtw89/rtw8852b_table.o
  CC [M]  /home/frank/rtw89/rtw8852b_rfk.o
  CC [M]  /home/frank/rtw89/rtw8852b_rfk_table.o
  LD [M]  /home/frank/rtw89/rtw_8852b.o
  CC [M]  /home/frank/rtw89/rtw8852be.o
  LD [M]  /home/frank/rtw89/rtw_8852be.o
  CC [M]  /home/frank/rtw89/rtw8852c.o
  CC [M]  /home/frank/rtw89/rtw8852c_table.o
  CC [M]  /home/frank/rtw89/rtw8852c_rfk.o
  CC [M]  /home/frank/rtw89/rtw8852c_rfk_table.o
  LD [M]  /home/frank/rtw89/rtw_8852c.o
  CC [M]  /home/frank/rtw89/rtw8852ce.o
  LD [M]  /home/frank/rtw89/rtw_8852ce.o
  CC [M]  /home/frank/rtw89/pci.o
  LD [M]  /home/frank/rtw89/rtw89pci.o
  MODPOST /home/frank/rtw89/Module.symvers
  CC [M]  /home/frank/rtw89/rtw89core.mod.o
  LD [M]  /home/frank/rtw89/rtw89core.ko
  BTF [M] /home/frank/rtw89/rtw89core.ko
Skipping BTF generation for /home/frank/rtw89/rtw89core.ko due to unavailability of vmlinux
  CC [M]  /home/frank/rtw89/rtw89pci.mod.o
  LD [M]  /home/frank/rtw89/rtw89pci.ko
  BTF [M] /home/frank/rtw89/rtw89pci.ko
Skipping BTF generation for /home/frank/rtw89/rtw89pci.ko due to unavailability of vmlinux
  CC [M]  /home/frank/rtw89/rtw_8852a.mod.o
  LD [M]  /home/frank/rtw89/rtw_8852a.ko
  BTF [M] /home/frank/rtw89/rtw_8852a.ko
Skipping BTF generation for /home/frank/rtw89/rtw_8852a.ko due to unavailability of vmlinux
  CC [M]  /home/frank/rtw89/rtw_8852ae.mod.o
  LD [M]  /home/frank/rtw89/rtw_8852ae.ko
  BTF [M] /home/frank/rtw89/rtw_8852ae.ko
Skipping BTF generation for /home/frank/rtw89/rtw_8852ae.ko due to unavailability of vmlinux
  CC [M]  /home/frank/rtw89/rtw_8852b.mod.o
  LD [M]  /home/frank/rtw89/rtw_8852b.ko
  BTF [M] /home/frank/rtw89/rtw_8852b.ko
Skipping BTF generation for /home/frank/rtw89/rtw_8852b.ko due to unavailability of vmlinux
  CC [M]  /home/frank/rtw89/rtw_8852be.mod.o
  LD [M]  /home/frank/rtw89/rtw_8852be.ko
  BTF [M] /home/frank/rtw89/rtw_8852be.ko
Skipping BTF generation for /home/frank/rtw89/rtw_8852be.ko due to unavailability of vmlinux
  CC [M]  /home/frank/rtw89/rtw_8852c.mod.o
  LD [M]  /home/frank/rtw89/rtw_8852c.ko
  BTF [M] /home/frank/rtw89/rtw_8852c.ko
Skipping BTF generation for /home/frank/rtw89/rtw_8852c.ko due to unavailability of vmlinux
  CC [M]  /home/frank/rtw89/rtw_8852ce.mod.o
  LD [M]  /home/frank/rtw89/rtw_8852ce.ko
  BTF [M] /home/frank/rtw89/rtw_8852ce.ko
Skipping BTF generation for /home/frank/rtw89/rtw_8852ce.ko due to unavailability of vmlinux
make[1]: se sale del directorio '/usr/src/kernels/6.1.15-200.fc37.x86_64'

sudo make install:
make -C /lib/modules/6.1.15-200.fc37.x86_64/build M=/home/frank/rtw89 modules
make[1]: se entra en el directorio '/usr/src/kernels/6.1.15-200.fc37.x86_64'
make[1]: se sale del directorio '/usr/src/kernels/6.1.15-200.fc37.x86_64'
Install rtw89 SUCCESS

uname -r
6.1.15-200.fc37.x86_64

sudo dmesg | grep 8852
[    7.204677] Bluetooth: hci0: RTL: examining hci_ver=0b hci_rev=000b lmp_ver=0b lmp_subver=8852
[    7.205673] Bluetooth: hci0: RTL: loading rtl_bt/rtl8852bu_fw.bin
[    7.210717] Bluetooth: hci0: RTL: loading rtl_bt/rtl8852bu_config.bin

On the other hand, I did the tests [Kernel 6.2 test](https://fedoraproject.org/wiki/Test_Day:2023-03-05_Kernel_6.2_Test_Week) week which ended yesterday and when testing the new kernel 6.2 the wifi works, the problem is that currently the last stable version is 6.1.15-200.fc37.x86_64

lwfinger, 2023-03-14:
I have no idea why you posted under an issue for the RTL8852AE. Your device is an RTW8852BU.

Your device was not added to the kernel until version 6.2 - it is not in version 6.1!

That driver is in this repo, but you have apparently skipped a 'git pull' as none of the USB drivers are being built. If you do that pull, followed by a 'make' and a sudo make install', your device will work with kernel 6.1.15. On my system, I get the following output:

finger@localhost:~/rtw88>make
make -C /lib/modules/6.3.0-rc2/build M=/home/finger/rtw88 modules
  CC [M]  /home/finger/rtw88/main.o
  CC [M]  /home/finger/rtw88/mac80211.o
  CC [M]  /home/finger/rtw88/util.o
  CC [M]  /home/finger/rtw88/debug.o
  CC [M]  /home/finger/rtw88/tx.o
  CC [M]  /home/finger/rtw88/rx.o
  CC [M]  /home/finger/rtw88/mac.o
  CC [M]  /home/finger/rtw88/phy.o
  CC [M]  /home/finger/rtw88/coex.o
  CC [M]  /home/finger/rtw88/efuse.o
  CC [M]  /home/finger/rtw88/fw.o
  CC [M]  /home/finger/rtw88/ps.o
  CC [M]  /home/finger/rtw88/sec.o
  CC [M]  /home/finger/rtw88/wow.o
  CC [M]  /home/finger/rtw88/bf.o
  CC [M]  /home/finger/rtw88/regd.o
  CC [M]  /home/finger/rtw88/sar.o
  LD [M]  /home/finger/rtw88/rtw_core.o
  CC [M]  /home/finger/rtw88/rtw8822b.o
  CC [M]  /home/finger/rtw88/rtw8822b_table.o
  LD [M]  /home/finger/rtw88/rtw_8822b.o
  CC [M]  /home/finger/rtw88/rtw8822be.o
  LD [M]  /home/finger/rtw88/rtw_8822be.o
  CC [M]  /home/finger/rtw88/rtw8822bu.o
  LD [M]  /home/finger/rtw88/rtw_8822bu.o
  CC [M]  /home/finger/rtw88/rtw8822bs.o
  LD [M]  /home/finger/rtw88/rtw_8822bs.o
  CC [M]  /home/finger/rtw88/rtw8822c.o
  CC [M]  /home/finger/rtw88/rtw8822c_table.o
  LD [M]  /home/finger/rtw88/rtw_8822c.o
  CC [M]  /home/finger/rtw88/rtw8822ce.o
  LD [M]  /home/finger/rtw88/rtw_8822ce.o
  CC [M]  /home/finger/rtw88/rtw8822cu.o
  LD [M]  /home/finger/rtw88/rtw_8822cu.o
  LD [M]  /home/finger/rtw88/rtw_8822cs.o
  CC [M]  /home/finger/rtw88/rtw8723d.o
  CC [M]  /home/finger/rtw88/rtw8723d_table.o
  LD [M]  /home/finger/rtw88/rtw_8723d.o
  CC [M]  /home/finger/rtw88/rtw8723de.o
  LD [M]  /home/finger/rtw88/rtw_8723de.o
  CC [M]  /home/finger/rtw88/rtw8723du.o
  LD [M]  /home/finger/rtw88/rtw_8723du.o
  CC [M]  /home/finger/rtw88/rtw8821c.o
  CC [M]  /home/finger/rtw88/rtw8821c_table.o
  LD [M]  /home/finger/rtw88/rtw_8821c.o
  CC [M]  /home/finger/rtw88/rtw8821ce.o
  LD [M]  /home/finger/rtw88/rtw_8821ce.o
  LD [M]  /home/finger/rtw88/rtw_8821cs.o
  CC [M]  /home/finger/rtw88/rtw8821cu.o
  LD [M]  /home/finger/rtw88/rtw_8821cu.o
  CC [M]  /home/finger/rtw88/pci.o
  LD [M]  /home/finger/rtw88/rtw_pci.o
  CC [M]  /home/finger/rtw88/sdio.o
  LD [M]  /home/finger/rtw88/rtw_sdio.o
  CC [M]  /home/finger/rtw88/usb.o
  LD [M]  /home/finger/rtw88/rtw_usb.o
  MODPOST /home/finger/rtw88/Module.symvers
  CC [M]  /home/finger/rtw88/rtw_core.mod.o
  LD [M]  /home/finger/rtw88/rtw_core.ko
  CC [M]  /home/finger/rtw88/rtw_8822b.mod.o
  LD [M]  /home/finger/rtw88/rtw_8822b.ko
  CC [M]  /home/finger/rtw88/rtw_8822be.mod.o
  LD [M]  /home/finger/rtw88/rtw_8822be.ko
  CC [M]  /home/finger/rtw88/rtw_8822bu.mod.o
  LD [M]  /home/finger/rtw88/rtw_8822bu.ko
  CC [M]  /home/finger/rtw88/rtw_8822bs.mod.o
  LD [M]  /home/finger/rtw88/rtw_8822bs.ko
  CC [M]  /home/finger/rtw88/rtw_8822c.mod.o
  LD [M]  /home/finger/rtw88/rtw_8822c.ko
  CC [M]  /home/finger/rtw88/rtw_8822ce.mod.o
  LD [M]  /home/finger/rtw88/rtw_8822ce.ko
  CC [M]  /home/finger/rtw88/rtw_8822cu.mod.o
  LD [M]  /home/finger/rtw88/rtw_8822cu.ko
  CC [M]  /home/finger/rtw88/rtw_8822cs.mod.o
  LD [M]  /home/finger/rtw88/rtw_8822cs.ko
  CC [M]  /home/finger/rtw88/rtw_8723d.mod.o
  LD [M]  /home/finger/rtw88/rtw_8723d.ko
  CC [M]  /home/finger/rtw88/rtw_8723de.mod.o
  LD [M]  /home/finger/rtw88/rtw_8723de.ko
  CC [M]  /home/finger/rtw88/rtw_8723du.mod.o
  LD [M]  /home/finger/rtw88/rtw_8723du.ko
  CC [M]  /home/finger/rtw88/rtw_8821c.mod.o
  LD [M]  /home/finger/rtw88/rtw_8821c.ko
  CC [M]  /home/finger/rtw88/rtw_8821ce.mod.o
  LD [M]  /home/finger/rtw88/rtw_8821ce.ko
  CC [M]  /home/finger/rtw88/rtw_8821cs.mod.o
  LD [M]  /home/finger/rtw88/rtw_8821cs.ko
  CC [M]  /home/finger/rtw88/rtw_8821cu.mod.o
  LD [M]  /home/finger/rtw88/rtw_8821cu.ko
  CC [M]  /home/finger/rtw88/rtw_pci.mod.o
  LD [M]  /home/finger/rtw88/rtw_pci.ko
  CC [M]  /home/finger/rtw88/rtw_sdio.mod.o
  LD [M]  /home/finger/rtw88/rtw_sdio.ko
  CC [M]  /home/finger/rtw88/rtw_usb.mod.o
  LD [M]  /home/finger/rtw88/rtw_usb.ko

Notice the rtw_8xxx*u.ko (for USB), and rtw8xxx*s.ko (for SDIO) that you are missing. The SDIO drivers will not be in the kernel until v6.4 or later.

Stenfrank, 2023-03-22:
Thank you so muc