lwfinger/rtw89 #209: How to install on Ubuntu

jmwileyms, 2023-01-04:
Hello, I have followed the instructions to build the files for Ubuntu. I am trying to get Ubuntu running on my Lenovo Slim 7 ProX which has the RTL 8852be. I am building it in a VM running on Windows and trying to get a dual boot version of Ubuntu to use the driver I build. After running the make command how can I get the driver to install on the dual booted OS?

Kind regards,
J.M. Wiley

Ps. Obviously a Ubuntu/Unix newbie.

lwfinger, 2023-01-04:
You can build the driver in the VM, but it takes some effort to transfer it to the real Ubuntu copy. Copying the kernel modules will not work unless the kernel version AND the compiler version are the same in both the VM and the real machine.

Does your dual-booted Ubuntu have network? If so, then just clone the repo on it and build the code there. If no network on the real machine, you could pass a USB stick to the VM, and copy the contents of your rtw89 folder to it, then reboot into the real machine. Then mount that stick on it, and copy the files to the disk. That is a fairly complicated set of steps for a newbie.

An easier solution would be to buy a USB donble such as the RTL8188EU (https://www.ebay.com/itm/283687393583?hash=item420d16192f:g:qP8AAOSwWhxd2xy3&amdata=enc%3AAQAHAAAA4IAwCL2yAY5l9CKPypewbhIuaJZZ46m38NA4ZLfwzi6PfhO%2F7qvBrITxZr0%2BNbtSMrI2JETUjoiQE7%2By1NJEsY%2FQHXvQiWXJj9L7dJBpJ%2FATrG8P1BPzkRiJAuEZaoqL9K6rtR5bJ7y61SqRA7VVp2iL2DKvydr7%2FcjmZONb8Loq9BvKNBKcB5y9Yz7lzBdoIjDgypNxYj%2B%2B0jWlrA8HpkOqYe8yTGln7T3zXXaYG8tWwwlEyeGTAyjQdTyKMGCSkNwed5QgS8GC2%2BLGMSst%2FiE4FTrvBCPUyfl2yJgmGJ3Y%7Ctkp%3ABFBMxIS78a9h) for $2.11. You can just plug it in, configure, and your Internet will work. At that point, you can download and build the rtw89 driver.

jmwileyms, 2023-01-05:
Hello, Thanks for the quick response. I never thought about buying a USB dongle I will give that a shot since it seems easiest.

jmwileyms, 2023-01-12:
Hello, I ordered the USB dongle you mentioned above and the driver does not seem to be installed. It seems the one sent to me was an 8188FTV and not the 8188EU. Should it work? If so how can I get the driver installed for it. It is definitely recognized. When I run lsusb I see the following in the output:
Bus 003 Device 002: ID 0bda:f179 Realtek Semiconductor Corp. RTL8188FTV 802.11b/g/n 1T1R 2.4G WLAN Adapter

Is there anyway to get a prebuilt driver package for this, or am I in the same situation as before?

Thanks,
J.M.

Jusix, 2023-01-24:
@jmwileyms The mainline kernel 6.2 should work. https://kernel.ubuntu.com/~kernel-ppa/mainline/v6.2-rc5/

jmwileyms, 2023-01-25:
@lwfinger - I had the seller send me a replacement and it turns out to be the same one :-( @Jusix, Thanks I will give that a shot.

jmwileyms, 2023-01-26:
@Jusix - I tried installing the latest kernel and it is still not recognizing either of my network adapters. It does show the USB device as unclaimed when I run lshw -C network.

lwfinger, 2023-01-26:
We need to see what devices you have now. Post the output of 'lsusb'. I did not expect the new kernel to change anything. None of these drivers have recently been added to the kernel.

jmwileyms, 2023-01-26:
Realtek RTL8188FTV Wireless LAN 802.11n USB 2.0 Network Adapter
Realtek RTL8852BE WiFi 6 802.11ax PCIe Adapter

Are the two devices I currently have. And I am now running the 6.2 Ubuntu kernel

Jusix, 2023-01-26:
[Buildlog](https://kernel.ubuntu.com/~kernel-ppa/mainline/v6.2-rc5/amd64/log) tells  rtw89_8852be.ko is part of the build. Have you copied the firmware which is still missing in Ubuntu?

```
Missing firmware for RTW8852BE

Thae necessary firmwar3e file should be in package firmware-realtek of linux-firmware-realtek;
however some versions of some distros have been extremely slow to pick up this firmware file,
even though it has been in the official linux-firmware repo at vger.kernel.org since
Oct. 27, 2022. If your distro is one of these, you can download the firmware from
https://lwfinger.com/download/rtw8852b_fw.bin, and copy it to /lib/firmware/rtw89/.
```

jmwileyms, 2023-01-26:
@lwfinger I am running Ubuntu 22.04

jmwileyms, 2023-01-26:
I assume I just copy the file into the /lib/firmware/rtw89 directory. Is there anything else I need to do?

Jusix, 2023-01-26:
Yes just copy the file and reboot.

lwfinger, 2023-01-26:
File README.md also has this info, but a 6.2 kernel has the driver I am using.

You managed to delete the USB IDs for the RTL8188FTV, thus I have no idea what driver it uses! I get tired of getting incomplete information.

By the way, the output of the dmesg command would have shown the errors thrown by the missing firmware. Too bad that Ubuntu is so slow in packageing the firmware that has been in the linux-firmware official repository since October. Yet another reason for not using Ubuntu!!!

jmwileyms, 2023-01-26:
The firmware seemed to do the trick. For the first time I am sending this message from Ubuntu instead of Windows. Thanks for the help.

lwfinger, 2023-01-26:
Of course it was the missing piece after you got the 6.2 kernel. The rtw89 repo would have fixed it for earlier kernels.

closed 2023-01-26 by lwfinger