lwfinger/rtw89 #211: My wifi is often disconnected and cannot be reconnected

fengwk, 2023-01-11:
My wifi is often disconnected and cannot be reconnected. At present, one recovery method found is to shut down and restart to recover. This bothers me, but I don't know the problem. Here are some logs I intercepted.

kernel：Linux arch 6.1.4-redmipatch

```
1月 11 21:46:54 arch kernel: hub 6-0:1.0: config failed, hub doesn't have any ports! (err -19)
1月 11 21:46:55 arch bluetoothd[548]: src/plugin.c:plugin_init() Failed to init vcp plugin
1月 11 21:46:55 arch bluetoothd[548]: src/plugin.c:plugin_init() Failed to init mcp plugin
1月 11 21:46:55 arch bluetoothd[548]: src/plugin.c:plugin_init() Failed to init bap plugin
1月 11 21:46:55 arch kernel: Bluetooth: hci0: Failed to read codec capabilities (-22)
1月 11 21:46:56 arch kernel: ucsi_acpi USBC000:00: PPM init failed (-110)
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a107
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0f9
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0fb
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a015
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0f1
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a00b
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a02f
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0e5
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0ef
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a009
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a02d
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0e7
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0ef
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a00b
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a033
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a01f
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a00f
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0d5
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0f5
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a00d
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a033
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a103
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a01b
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0e7
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb8989ffd
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0db
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a021
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0d9
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a021
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0df
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb89acaaf
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb897d16f
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb89a8dbb
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb89a8df3
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb8935ed3
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb89a978d
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb890cd1b
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb89a8fdf
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb892055b
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb89a8d9b
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb8935f2d
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb8935e25
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb89a39f7
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb89ac527
1月 11 21:50:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb893434d
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a03b
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a037
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a025
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0e7
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0ff
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a017
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a023
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a037
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0d9
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0eb
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0fb
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a003
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a01f
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a021
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0d9
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a01f
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0ff
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a01f
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a033
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0d7
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0ed
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a103
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a00b
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a01f
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a035
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0db
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0ef
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a035
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a0d5
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb898a105
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb892058f
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb89a3c43
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb8920561
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb8935e13
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb89a8dd1
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb89343f1
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb89a39c7
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb893438b
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb89a3b7f
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb897d175
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb8935ebf
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb890cce7
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb8935e4f
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb89360d5
1月 11 21:52:05 arch kernel: rtw89_8852be 0000:01:00.0: [ERR]fw PC = 0xb8935ded
1月 11 22:06:46 arch pulseaudio[12590]: Lost I/O connection in module "module-gsettings"
1月 11 22:06:47 arch kernel: watchdog: watchdog0: watchdog did not stop!
```

fengwk, 2023-01-11:
I found this, https://github.com/lwfinger/rtw89/issues/55

I will try it

lwfinger, 2023-01-11:
Those options should fix the firmware errors.

fengwk, 2023-01-12:
I tried these 3 options, but disconnection still occurs

fengwk, 2023-06-10:
Sorry, I have replaced the network card and cannot reproduce this issue again

closed 2023-06-10 by fengwk