lwfinger/rtw89 #212: Instructions on blacklisting kernel versions

blackshibe, 2023-01-11:
The readme reads the following:

This branch was created from the version merged into the wireless-drivers-next repo, which is in the 5.16 kernel. IF YOU USE DRIVERS FROM THIS REPO FOR KERNELS 5.16+, YOU MUST BLACKLIST THE KERNEL VERSIONS!!!! FAILING TO DO THIS WILL RESULT IN ALL MANNER OF STRANGE ERRORS.

I have no idea what this means, but it sure seems to mean that I can't use a 6.0 kernel without running something in a terminal., which is a problem. Please elaborate?

lwfinger, 2023-01-12:
New kernels have some of these drivers built in. If you build and install the drivers from this repo, then you run the risk of getting a mixed set of drivers with a kernel crash.

If you are using Linux, then you should spend some time learning how to run a terminal. There are some things that cannot be done with a GUI.

AGenews, 2023-01-13:
Dear lwfinger,
I highly appreciate your work and I am deeply thankful for your development. I am (was) able to run my wifi module with your driver. I typically learn new stuff in Linux when it is necessary (like now, blacklisting the RIGHT modules) ... this works best with a working internet connection. Therefore it would be of utmost generosity if you could share another glimpse of LinuxFu with us, so we are back in the game (without USB tethering), and can deepen our understanding of blacklisting kernel modules .... but for now .... a jumpstart would be really really really appreciated.
Thank you!

lwfinger, 2023-01-13:
The blacklist is just a list of modules that should not be loaded. As the kernel detects each PCI, USB, and SDIO device attached to the respective bus, it finds the driver that contains that ID in its tables. If that driver is not found in a blacklist, then it loads the module. If there are options files that mention this driver, then it applies those options when the module is loaded. It is as simple as that. Do a Google search for "How does blacklisting work". The article at https://linuxconfig.org/how-to-blacklist-a-module-on-ubuntu-debian-linux is pretty good - the information applies to other distros as well.

By the way: This is not my driver! It was wriitten by the PCIe Linux group at Realtek in Taiwan. I take that code that is in new kernels and make it available for older kernels.

closed 2023-01-13 by lwfinger