lwfinger/rtw89 #213: Driver stopped to work with kernel upgrade to  5.15.0-57-generic

LTepliakov, 2023-01-12:
With kernel 5.15.0-56-generic it worked fine. Her is the link to Linux probe https://linux-hardware.org/?probe=e228f3ea92
But with upgrade to 5.15.0-57-generic it is detected but does not work: https://linux-hardware.org/?probe=4606ff1dab
 
Thank you for this driver and appreciate your help!

lwfinger, 2023-01-12:
Your dmesg output shows the following:

[    3.167313] rtw89_8852be 0000:30:00.0: Direct firmware load for rtw89/rtw8852b_fw.bin failed with error -2
[    3.167318] rtw89_8852be 0000:30:00.0: failed to early request firmware: -2
[    3.167427] rtw89_8852be 0000:30:00.0: enabling device (0000 -> 0003)
[    3.167447] rtw89_8852be 0000:30:00.0: Direct firmware load for rtw89/rtw8852b_fw.bin failed with error -2
[    3.167449] rtw89_8852be 0000:30:00.0: failed to request firmware
[    3.170302] rtw89_8852be 0000:30:00.0: failed to wait firmware completion
[    3.170328] rtw89_8852be 0000:30:00.0: failed to setup chip information
[    3.172210] rtw89_8852be: probe of 0000:30:00.0 failed with error -22

You do not have the firmware installed. The reason is that your distro does not package it. The last section of README.md gives the prescription on how to download and install it. BTW, my distro includes the 8852be firmware as part of the firmware-realtek package. Yours should as well.

LTepliakov, 2023-01-13:
Thank you lwfinger!
Downloaded firmware, reinstalled the driver and it now works fine.

closed 2023-01-13 by LTepliakov