lwfinger/rtw89 #214: The RTL8853BE works erratically under 5Ghz WiFi

cmx0223, 2023-01-15:
The RTL8852BE works erratically under 5Ghz WiFi.2.4Ghz WiFi interferes with rtw89-BT.Connecting to 5G WiFi will have disconnection problems from time to time, and you will be repeatedly asked for a password when trying to reconnect.

Is this fixable?Thank you for your efforts.

cmx0223, 2023-01-15:
Here are logs.
[image]

THMonster, 2023-01-27:
I got the same problem. Did you fix it?

lwfinger, 2023-01-27:
This problem is due to your BIOS not handling the power changes on the PCIe bus. Create a file with
sudo touch /usr/lib/modprobe.d/70-rtw8852be.conf
Using your favorite editor, add the following lines to the file above.

options rtw89_pci  disable_clkreq=y disable_aspm_l1=y   disable_aspm_l1ss=y
options rtw89pci  disable_clkreq=y disable_aspm_l1=y   disable_aspm_l1ss=y
options rtw89_core disable_ps_mode=y
options rtw89core disable_ps_mode=y

Those options keep the PCI bus from entering the deepest idle mode, and keeps the AP from entering power-save mode.. That set of options will work for the kernel driver, AND the one from this repo. The downside is that your wireless card and computer will use more power, which is why these options are not standard.

lwfinger, 2023-04-24:
I just pushed a new version that may fix this problem. Do a 'git pull' and remake the driver. Report here if it helps or not.

lwfinger, 2023-07-11:
If disabling power save works, why did you post?

Have you tried the options for rtw89core and rtw89_core (the kernel version)? 'modinfo rtw89core' will show you the options.

Mingy123, 2023-07-11:
I was assuming based on your previous post that the module options was only a temporary fix, so I wanted to report.

I don't understand what you mean by trying the kernel version, but modinfo does show some parameters:
```
parm:           debug_mask:Debugging mask (uint)
parm:           disable_ps_mode:Set Y to disable low power mode (bool)
```

If I'm not mistaken, the drivers provided by the kernel does not work with rtl8852be, the wlan0 interface doesn't appear after uninstalling this driver. According to dmesg, there is a null pointer derefence error. But I might be doing something wrong in trying to switch to the kernel drivers, since lsmod still shows that rtw89_8852be is loaded:

```
lsmod | grep rtw 
rtw89_8852be           12288  1
rtw_8852b             360448  1 rtw89_8852be
rtw89_pci              86016  1 rtw89_8852be
rtw89_core            622592  2 rtw89_pci,rtw_8852b
mac80211             1527808  2 rtw89_core,rtw89_pci
cfg80211             1290240  3 rtw89_core,mac80211,rtw_8852b
```

<details>
<summary>dmesg log</summary>
[    3.441658] rtw_8852b: loading out-of-tree module taints kernel.
[    3.441663] rtw_8852b: module verification failed: signature and/or required key missing - tainting kernel
[    3.468546] rtw89_8852be 0000:01:00.0: loaded firmware rtw89/rtw8852b_fw-1.bin
[    3.468615] rtw89_8852be 0000:01:00.0: enabling device (0000 -> 0003)
[    3.470495] rtw89_8852be 0000:01:00.0: Firmware version 0.29.29.1, cmd version 0, type 5
[    3.470497] rtw89_8852be 0000:01:00.0: Firmware version 0.29.29.1, cmd version 0, type 3
[    3.470503] BUG: kernel NULL pointer dereference, address: 0000000000000058
[    3.470518] #PF: supervisor read access in kernel mode
[    3.470526] #PF: error_code(0x0000) - not-present page
[    3.470535] PGD 0 P4D 0
[    3.470540] Oops: 0000 [#1] PREEMPT SMP NOPTI
[    3.470547] CPU: 0 PID: 731 Comm: (udev-worker) Tainted: G           OE      6.4.1-artix1-1 #1 deff63865c2d0dea265e2ccc73f016d343a4f5a5
[    3.470561] Hardware name: HP HP Laptop 14s-dq5xxx/8A1F, BIOS F.17 10/20/2022
[    3.470570] RIP: 0010:rtw8852b_set_txpwr_ul_tb_offset+0x2a/0x170 [rtw_8852b]
</details>

lwfinger, 2023-07-11:
Those module options are not temporary fixes IF YOUR BIOS IS FAULTY!!! I think yours is.

Your module list shows some drim the kernel, and one from this repo. That combination is guaranteed to fail.

I gave you both repo and kernel options, which is why I mentioned the kernel version. Sorry that I confused you.

closed 2023-10-02 by lwfinger

naggie, 2024-03-31:
For anyone finding this -- I had highly unstable pings. I fixed it on nixos by bypassing power saving entirely: `networking.networkmanager.wifi.powersave = false;` -- that does it permanently. You can test it with `sudo iw dev wlp5s0 set power_save on `