lwfinger/rtw89 #217: Cannot seem to compile properly?

alexbahder, 2023-01-22:
Just updated my kernel version to 6.0.12-76060006-generic. 
Tried to find the old driver I was using here ( ) but was redirected to this one (rtw89). 
I cloned the repository, tried to make, and was confronted with this:

`make -C /lib/modules/6.0.12-76060006-generic/build M=/home/fourzerofour/rtw89/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-6.0.12-76060006-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-11 (Ubuntu 11.2.0-19ubuntu1) 11.2.0
  You are using:           gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
  CC [M]  /home/fourzerofour/rtw89/rtw89/core.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/chan.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/mac80211.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/mac.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/phy.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/fw.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/cam.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/efuse.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/regd.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/sar.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/coex.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/ps.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/debug.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/ser.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/wow.o
  LD [M]  /home/fourzerofour/rtw89/rtw89/rtw89core.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/rtw8852a.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/rtw8852a_table.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/rtw8852a_rfk.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/rtw8852a_rfk_table.o
  LD [M]  /home/fourzerofour/rtw89/rtw89/rtw_8852a.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/rtw8852ae.o
  LD [M]  /home/fourzerofour/rtw89/rtw89/rtw_8852ae.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/rtw8852b.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/rtw8852b_table.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/rtw8852b_rfk.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/rtw8852b_rfk_table.o
  LD [M]  /home/fourzerofour/rtw89/rtw89/rtw_8852b.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/rtw8852be.o
  LD [M]  /home/fourzerofour/rtw89/rtw89/rtw_8852be.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/rtw8852c.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/rtw8852c_table.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/rtw8852c_rfk.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/rtw8852c_rfk_table.o
  LD [M]  /home/fourzerofour/rtw89/rtw89/rtw_8852c.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/rtw8852ce.o
  LD [M]  /home/fourzerofour/rtw89/rtw89/rtw_8852ce.o
  CC [M]  /home/fourzerofour/rtw89/rtw89/pci.o
  LD [M]  /home/fourzerofour/rtw89/rtw89/rtw89pci.o
  MODPOST /home/fourzerofour/rtw89/rtw89/Module.symvers
  CC [M]  /home/fourzerofour/rtw89/rtw89/rtw89core.mod.o
  LD [M]  /home/fourzerofour/rtw89/rtw89/rtw89core.ko
  BTF [M] /home/fourzerofour/rtw89/rtw89/rtw89core.ko
Skipping BTF generation for /home/fourzerofour/rtw89/rtw89/rtw89core.ko due to unavailability of vmlinux
  CC [M]  /home/fourzerofour/rtw89/rtw89/rtw89pci.mod.o
  LD [M]  /home/fourzerofour/rtw89/rtw89/rtw89pci.ko
  BTF [M] /home/fourzerofour/rtw89/rtw89/rtw89pci.ko
Skipping BTF generation for /home/fourzerofour/rtw89/rtw89/rtw89pci.ko due to unavailability of vmlinux
  CC [M]  /home/fourzerofour/rtw89/rtw89/rtw_8852a.mod.o
  LD [M]  /home/fourzerofour/rtw89/rtw89/rtw_8852a.ko
  BTF [M] /home/fourzerofour/rtw89/rtw89/rtw_8852a.ko
Skipping BTF generation for /home/fourzerofour/rtw89/rtw89/rtw_8852a.ko due to unavailability of vmlinux
  CC [M]  /home/fourzerofour/rtw89/rtw89/rtw_8852ae.mod.o
  LD [M]  /home/fourzerofour/rtw89/rtw89/rtw_8852ae.ko
  BTF [M] /home/fourzerofour/rtw89/rtw89/rtw_8852ae.ko
Skipping BTF generation for /home/fourzerofour/rtw89/rtw89/rtw_8852ae.ko due to unavailability of vmlinux
  CC [M]  /home/fourzerofour/rtw89/rtw89/rtw_8852b.mod.o
  LD [M]  /home/fourzerofour/rtw89/rtw89/rtw_8852b.ko
  BTF [M] /home/fourzerofour/rtw89/rtw89/rtw_8852b.ko
Skipping BTF generation for /home/fourzerofour/rtw89/rtw89/rtw_8852b.ko due to unavailability of vmlinux
  CC [M]  /home/fourzerofour/rtw89/rtw89/rtw_8852be.mod.o
  LD [M]  /home/fourzerofour/rtw89/rtw89/rtw_8852be.ko
  BTF [M] /home/fourzerofour/rtw89/rtw89/rtw_8852be.ko
Skipping BTF generation for /home/fourzerofour/rtw89/rtw89/rtw_8852be.ko due to unavailability of vmlinux
  CC [M]  /home/fourzerofour/rtw89/rtw89/rtw_8852c.mod.o
  LD [M]  /home/fourzerofour/rtw89/rtw89/rtw_8852c.ko
  BTF [M] /home/fourzerofour/rtw89/rtw89/rtw_8852c.ko
Skipping BTF generation for /home/fourzerofour/rtw89/rtw89/rtw_8852c.ko due to unavailability of vmlinux
  CC [M]  /home/fourzerofour/rtw89/rtw89/rtw_8852ce.mod.o
  LD [M]  /home/fourzerofour/rtw89/rtw89/rtw_8852ce.ko
  BTF [M] /home/fourzerofour/rtw89/rtw89/rtw_8852ce.ko
Skipping BTF generation for /home/fourzerofour/rtw89/rtw89/rtw_8852ce.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-6.0.12-76060006-generic'
`

Of course, I tried to just go ahead and run 
`sudo make install`
But that didn't seem to work right, as despite trying to unload / load the module, rebooting, etc.; WiFi still didn't work at all. In fact, Pop OS refused to acknowledge (in settings) that a WiFi adapter was present in the system.

Also, upon trying to unload the 'rtw_core' part, it says it does not exist. I am assuming this is related to why it doesn't work, and I am assuming the lack of it working is due to the build errors / stuff that was skipped. 

But I could be wrong and have been known to be pretty dumb about this stuff.  Hope someone smarter than me can tell me what I'm doing wrong.

lwfinger, 2023-01-22:
If you look at the output of the dmesg command, you will likely see that the modules refuse to load due to the following:

directory '/usr/src/linux-headers-6.0.12-76060006-generic' warning: the compiler differs from the one used to build the kernel The kernel was built by: x86_64-linux-gnu-gcc-11 (Ubuntu 11.2.0-19ubuntu1) 11.2.0
You are using:           gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0 CC

Ubuntu is leaving you with a crippled system, which is very common for them. You need to get an updated kernel, or roll back gcc to  11.2.0.

To avoid kernel crashes, the kernel and any external driver must agree on the layout of critical structures internal to the kernel. this agreement will happen as long as the kernel headers match the running kernel, and both the kernel and the external module are compiled with EXACTLY the same compiler.

kavin-kr, 2023-04-05:
I'm also facing the same issue after updating Ubuntu.
```sh
❯ uname --all
Linux pavilion 5.19.0-38-generic #39~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Mar 17 21:16:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

❯ sudo make install
make -C /lib/modules/5.19.0-38-generic/build M=/home/username/.local/drivers/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.19.0-38-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
  You are using:           gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
make[1]: Leaving directory '/usr/src/linux-headers-5.19.0-38-generic'
Install rtw89 SUCCESS
```

But for me both the compiler version are same (assuming x86_64-linux-gnu-gcc and gcc represents same).

I have read the warnings about Ubuntu issues in README but any help in making it work will be useful.

kavin-kr, 2023-04-07:
For those who are experiencing the above issue due to Ubuntu kernal update, one solution which fixed my problem is to boot old kernal version.
1. Mark the packages of old kernal versions as manually installed so that it won't get removed by `apt autoremove`.
  `sudo apt-mark manual linux-image-5.19.0-35-generic linux-headers-5.19.0-35-generic`
2. Update the grub to use the old kernal versions.
  For how-to see: https://linux.how2shout.com/how-to-change-default-kernel-in-ubuntu-22-04-20-04-lts/#7_Change_or_Set_Default_Kernel_Version

closed 2023-04-24 by lwfinger