lwfinger/rtw89 #218: debian packaging/dkms issues

sney, 2023-01-30:
Hi,

3 issues -
* dh-modaliases is ubuntu-only, so dpkg-buildpackage fails until this is removed from d/control and d/rules (but it doesn't seem to be needed?)
* 3.0 (native) in d/source/format doesn't allow a package revision (this might be a new change, I think it did work previously)

and after tweaking d/control, d/rules, and d/changelog to adjust for the above, dpkg-buildpackage still:

* failed when it couldn't find the firmware blobs, even though your readme says they should be installed separately via the firmware-realtek package.

After copying the 3 files from /lib/firmware/rtw89 to the build directory, the packages were able to build, the dkms process succeeded, the module loaded, and wifi works. So it's just a few packaging issues. 

OS details:
Debian bookworm/sid
Linux 6.1.0-2 (6.1.7-1)
gcc 12.2.0-14
debhelper 13.11.4

You should be able to reproduce this sequence of events by testing the build in a debian testing/unstable VM. Debian has diverged a lot more from ubuntu over the past few years than many expect, and you're definitely not alone in assuming that something which works on ubuntu will work on debian. Unfortunately it's no longer true and testing on both platforms is needed if you're going to list both platforms as supported. 

I hope this is enough information for a fix. If not, I may have time to put together a PR with release-quality fixes in a few weeks, I can't promise anything at the moment but if you ping me after Feb 13th I'll know for sure. 

Thanks for your work,
sney

lwfinger, 2023-01-31:
I do not use either Ubuntu or Debian, nor do I tailor my repos for either. In addition, I do not explicitly suppport DKMS. If you have issues with any part of DKMS, then you should prepare patches, which I will apply if they are sound.

I will never support including the firmware in this repo. The distros should be following the official linux-firmware repository, at vger.kernel.org, which means that updates to the firmware are automatically updated whenever Realtek releases them. At that point, nothing needs to be done by me! I hope that Ubuntu and Debian will finally catch up with the rest of the world when kernel 6.2 is released and a driver for RTW8852BE is included in the kernel, but we will see!

sney, 2023-03-16:
The debian kernel team confirmed yesterday that debian 12 will release with a 6.1 (longterm) kernel, so this repo is still needed. I am just putting the finishing touches on some fixes for the debian packaging workflow and I will submit the pull request soon, likely tomorrow. 

sney

sney, 2023-03-17:
I thought this would close automatically when [the pull request](https://github.com/lwfinger/rtw89/pull/227) was merged. Guess not, here's me pressing the button.