lwfinger/rtw89 #219: Wifi is connected to network but no internet access

pmiss2, 2023-02-06:
Decided to do a Debian reinstall two days ago, before the reinstall the wifi was working fine. Ive followed the instructions with and without module signing for secureboot, and Ive done sudo modprobe -v rtw_8852ae after make install.
The laptop is connected to my wifi network but has no internet access. 
Still works with Ethernet though.
I'm using Debian 11 with non-free firmware included.

If anyone could help I would be forever grateful, Ive tried for many hours with no results, although i'm still new to linux.

Here is some useful information:

[iwconfig.txt](https://github.com/lwfinger/rtw89/files/10636544/iwconfig.txt)

[lsmod.txt](https://github.com/lwfinger/rtw89/files/10636545/lsmod.txt)

[lspci.txt](https://github.com/lwfinger/rtw89/files/10636552/lspci.txt)

[lspci -nnk | grep -iA2 net.txt](https://github.com/lwfinger/rtw89/files/10636553/lspci.-nnk.grep.-iA2.net.txt)

[nmcli dev show | grep DNS.txt](https://github.com/lwfinger/rtw89/files/10636554/nmcli.dev.show.grep.DNS.txt)

[ping_8.8.4.4.txt](https://github.com/lwfinger/rtw89/files/10636555/ping_8.8.4.4.txt)

[ping_8.8.8.8.txt](https://github.com/lwfinger/rtw89/files/10636556/ping_8.8.8.8.txt)

Thank you for reading this!!!

lwfinger, 2023-02-06:
I do not know how you configured your wireless connection, but I think you are missing a routing parameter. You can test this with 'route' or 'tracepath 8.8.8.8'.

pmiss2, 2023-02-07:
Im using a completely fresh install with nothing else configured. For my network configuration, i only connected to my network and typed the correct password.

[route.txt](https://github.com/lwfinger/rtw89/files/10671830/route.txt)
^this is the output of 'sudo route'

[tracepath.txt](https://github.com/lwfinger/rtw89/files/10671540/tracepath.txt)
^this is the output of 'tracepath 8.8.8.8' 

Thank you for helping me, if you ever need a poem, a song, or a couch moved please say so :)

lwfinger, 2023-02-07:
Your routing table is wrong. I have no idea how it was created, but mine is
finger@localhost:~/rtl8723bu>route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         192.168.0.1     0.0.0.0         UG    600    0        0 wls1
192.168.0.0     0.0.0.0         255.255.255.0   U     600    0        0 wls1

Yours is
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         _gateway        0.0.0.0         UG    600    0        0 wlp3s0
10.0.0.0        0.0.0.0         255.255.255.0   U     600    0        0 wlp3s0
link-local      0.0.0.0         255.255.0.0     U     1000   0        0 wlp3s0

I do not know what address that "_gateway" represents, but I think that default line should be
default        10.0.0.1        0.0.0.0         UG    600    0        0 wlp3s0

The command 'sudo route add default gw 10.0.0.1' should temporarily create the necessary route, but you need to find out why NetworkManager is not setting it up correctly. I run Debian 11 on a ppc32 host, and it got the routing right without me doing anything,

pmiss2, 2023-02-07:
after trying 'sudo route add default gw 10.0.0.1', my new routing table is:

Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         _gateway        0.0.0.0         UG    0      0        0 enp5s0
default         _gateway        0.0.0.0         UG    100    0        0 enp5s0
default         _gateway        0.0.0.0         UG    600    0        0 wlp3s0
10.0.0.0        0.0.0.0         255.255.255.0   U     100    0        0 enp5s0
10.0.0.0        0.0.0.0         255.255.255.0   U     600    0        0 wlp3s0
link-local      0.0.0.0         255.255.0.0     U     1000   0        0 enp5s0

It seems the gateway hasn't changed, I'm going to try to fix the default gateway, and also look into adding a permanent route

I might seek help on other forums because as you said this is a problem with my NetworkManager not with the driver

pmiss2, 2023-02-07:
I accidentally closed the thread sorry

closed 2023-02-07 by pmiss2