lwfinger/rtw89 #220: Direct firmware load for rtw89/rtw8852a_fw.bin failed with error -2

ArdityaWildan, 2023-02-10:
Hello,
I have a problem with this library on my debian 11 (Linux 6.1.11 x86_64) device
It doesn't seem to work, found below in kern.log :
`rtw89_8852ae 0000:01:00.0: Direct firmware load for rtw89/rtw8852a_fw.bin failed with error -2`

lspci :
`01:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device 8852
	DeviceName: WLAN
	Subsystem: Hewlett-Packard Company Device 88e1
	Flags: fast devsel, IRQ 24, IOMMU group 10
	I/O ports at 2000 [size=256]
	Memory at d0800000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: <access denied>
	Kernel modules: rtw_8852ae`

lsmod | grep rtw :
[image]

thanks for your attention

ArdityaWildan, 2023-02-10:
Sorry, I have found a solution for this problem by downloading that related bin manually

closed 2023-02-10 by ArdityaWildan

lwfinger, 2023-02-10:
Your distro should have provided it automatically.

Be aware that with compressed firmware, which is the normal form, the driver will get a failure when it tries to load the uncompressed one directly, and then succeeds when it loads the compressed form indirectly. As you only provided that one line from dmesg, I cannot tell you if that is what was happening.