lwfinger/rtw89 #221: Wifi used to work but now it doesn’t

an4s911, 2023-02-11:
I installed and setup Arch on a new Yoga Slim 7 Pro. 

And after installing the driver when I turn on the computer, it used to take a while but eventually connect. 

But now it just loads and loads and asks for the Wifi password a couple of times and just never gets connected. 

It happened after I installed virt-manager. I'm not sure if that has anything to do with it. It has some network setup involved as well.

an4s911, 2023-02-11:
When I try to connect using `nmtui` it says "Could not activate connection: Activation failed: The Wi-Fi network could not be found"

When I use Network Manager applet to connect, it keeps prompting me for the Wifi password and then just doesn’t connect.

I also tried download the firmware mentioned at the end of README.md. But seems like there file was already there. Yet I copied the file over to `/lib/firmware/rtw89`.

That didn't seem to fix it.

an4s911, 2023-02-11:
How I installed it:

`yay -S rtw89-dkms-git`

Made a file '/etc/modules-load.d/rtw89.conf`:
```
rtw_8852ae
```

And then ran:
`sudo modprobe -v rtw89_8852ae`

And then rebooted and it worked fine. But suddenly it stopped working.

I even uninstalled all of the libvirt and virt manager related packages. That didn't help either.

an4s911, 2023-02-11:
dmesg output while trying to connect to the network:
```
[  306.922717] audit: type=1111 audit(1676123171.015:101): pid=651 uid=0 auid=4294967295 ses=4294967295 msg='op=connection-activate uuid=0585134e-6762-4539-bcf9-718cb4674894 name=426173686565722057692D466920 pid=838 uid=1000 result=success exe="/usr/bin/NetworkManager" hostname=? addr=? terminal=? res=success'
[  306.923180] wlan0: deauthenticating from c8:9e:43:ba:16:f2 by local choice (Reason: 3=DEAUTH_LEAVING)
[  307.130645] wlan0: authenticate with c8:9e:43:ba:16:f3
[  307.130747] rtw89_8852ae 0000:01:00.0: MAC has already powered on
[  307.263104] wlan0: send auth to c8:9e:43:ba:16:f3 (try 1/3)
[  307.285155] wlan0: authenticated
[  307.331584] wlan0: associate with c8:9e:43:ba:16:f3 (try 1/3)
[  307.354341] wlan0: associate with c8:9e:43:ba:16:f3 (try 2/3)
[  307.425332] wlan0: associate with c8:9e:43:ba:16:f3 (try 3/3)
[  307.496280] wlan0: association with c8:9e:43:ba:16:f3 timed out
[  321.034277] wlan0: authenticate with c8:9e:43:ba:16:f3
[  321.034372] rtw89_8852ae 0000:01:00.0: MAC has already powered on
[  321.168243] wlan0: send auth to c8:9e:43:ba:16:f3 (try 1/3)
[  321.246108] wlan0: authenticate with ce:9e:43:ba:16:f4
[  321.381547] wlan0: send auth to ce:9e:43:ba:16:f4 (try 1/3)
[  321.389446] wlan0: authenticated
[  321.570477] wlan0: associate with c8:9e:43:ba:16:f3 (try 1/3)
[  321.573674] wlan0: RX AssocResp from c8:9e:43:ba:16:f3 (capab=0x1511 status=0 aid=2)
[  321.685567] wlan0: associated
[  321.751852] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  332.058186] audit: type=1334 audit(1676123196.151:102): prog-id=40 op=LOAD
[  332.058302] audit: type=1334 audit(1676123196.151:103): prog-id=41 op=LOAD
[  332.058336] audit: type=1334 audit(1676123196.151:104): prog-id=42 op=LOAD
[  332.116840] audit: type=1130 audit(1676123196.208:105): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-hostnamed comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[  347.778846] wlan0: disconnect from AP c8:9e:43:ba:16:f3 for new auth to ce:9e:43:ba:16:f4
[  347.878289] wlan0: authenticate with ce:9e:43:ba:16:f4
[  348.014616] wlan0: send auth to ce:9e:43:ba:16:f4 (try 1/3)
[  348.124608] wlan0: send auth to ce:9e:43:ba:16:f4 (try 2/3)
[  348.139988] wlan0: send auth to ce:9e:43:ba:16:f4 (try 3/3)
[  348.146416] wlan0: authenticated
[  348.204591] wlan0: associate with ce:9e:43:ba:16:f4 (try 1/3)
[  348.315010] wlan0: associate with ce:9e:43:ba:16:f4 (try 2/3)
[  349.169206] wlan0: associate with ce:9e:43:ba:16:f4 (try 3/3)
[  349.191301] wlan0: association with ce:9e:43:ba:16:f4 timed out
[  362.175357] audit: type=1131 audit(1676123226.268:106): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-hostnamed comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[  362.258281] audit: type=1334 audit(1676123226.351:107): prog-id=42 op=UNLOAD
[  362.258322] audit: type=1334 audit(1676123226.351:108): prog-id=41 op=UNLOAD
[  362.258334] audit: type=1334 audit(1676123226.351:109): prog-id=40 op=UNLOAD
[  368.083470] audit: type=1334 audit(1676123232.175:110): prog-id=43 op=LOAD
[  368.083632] audit: type=1334 audit(1676123232.175:111): prog-id=44 op=LOAD
[  368.083667] audit: type=1334 audit(1676123232.175:112): prog-id=45 op=LOAD
[  368.142599] audit: type=1130 audit(1676123232.235:113): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-hostnamed comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[  371.004201] wlan0: authenticate with ce:9e:43:ba:16:f4
[  371.004298] rtw89_8852ae 0000:01:00.0: MAC has already powered on
[  371.133704] wlan0: send auth to ce:9e:43:ba:16:f4 (try 1/3)
[  371.158249] wlan0: authenticated
[  371.161602] wlan0: associate with ce:9e:43:ba:16:f4 (try 1/3)
[  371.164984] wlan0: RX AssocResp from ce:9e:43:ba:16:f4 (capab=0x1511 status=0 aid=1)
[  371.275291] wlan0: associated
[  371.341487] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
```

lwfinger, 2023-02-11:
My first guess is that your signal is weak. It is taking several tries to associate or authenticate. Run the following command:

sudo iw dev wlan0 scan | egrep "signal|SSID|frreq:"

In the output, find the line with your AP and post it and the signal and freq lines above it. If you get multiple entries for that AP, post them all.

The output will tell me if you are trying to connect to a 5G or 2.4G AP. The latter propagates better, but lower performance.

an4s911, 2023-02-12:
I was about to reinstall it, so I moved the partition. And when I booted, the wifi is connected. Something funny. 
But anyways here is the output asked: 
```
freq: 2447
	signal: -62.00 dBm
	SSID: Basheer Wi-Fi\x20
		 * SSID List
		 * center freq segment 1: 10
		 * center freq segment 2: 0
	freq: 2447
	signal: -62.00 dBm
	SSID:
		 * SSID List
		 * center freq segment 1: 10
		 * center freq segment 2: 0
	freq: 5540
	signal: -62.00 dBm
	SSID:
		 * SSID List
		 * center freq segment 1: 106
		 * center freq segment 2: 0
	freq: 5200
	signal: -70.00 dBm
	SSID: Basheer Wi-Fi\x20
		 * SSID List
		 * center freq segment 1: 42
		 * center freq segment 2: 0
	freq: 2457
	signal: -69.00 dBm
	SSID: \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
	freq: 5540
	signal: -62.00 dBm
	SSID: Basheer Wi-Fi\x20
		 * SSID List
		 * center freq segment 1: 106
		 * center freq segment 2: 0
	freq: 2457
	signal: -69.00 dBm
	SSID: TP-Link_574A
```

Don't close the issue yet. I will try to open it a couple more times and see if the problem still persists.

an4s911, 2023-02-12:
I tried restarting and it still did persist, but it doesn't matter anymore. I installed ArcoLinuxD and the wifi works perfectly. 

I will close this issue as **Close as not planned**.

closed 2023-02-12 by an4s911

lwfinger, 2023-02-12:
The strongest signal in your list is -62 dBm, which is quite weak. That will explain the difficulty in connecting, and the connection drops. What surprises me is the same signal strength for frequency 2457 (channel 10) as for frequency 5540 (channel 106). My results are as follows:

        freq: 2437
        signal: -43.00 dBm
        SSID: Larry-ax
        freq: 5180
        signal: -47.00 dBm
        SSID: Larry-ax
                 * SSID List
                 * center freq segment 1: 42
                 * center freq segment 2: 0
                 * SSID List

        freq: 2437
        signal: -65.00 dBm
        SSID: Larry-ax
        freq: 5180
        signal: -76.00 dBm
        SSID: Larry-ax
                 * SSID List
                 * center freq segment 1: 42
                 * center freq segment 2: 0
                 * SSID List

The first 2.4/5 G pair are a Wifi6 AP located about 1 m above my laptop. The second pair are a mesh network  repeater located about 25 m away at the other end of my house. Slight changes in the position of that repeater will change the speed seen at that repeater by a factor of 10. Note that a change of 3 in dBm changes by the power by roughly a factor of 2. By that measure, your signals are weaker than mine by a factor of 50.