lwfinger/rtw89 #222: dkms error updating to Kernel 5.15.93-1

salimp2009, 2023-02-12:
Hi,
I am not sure if this is related to Make file here or not but when there was a Kernel update (on Manjaro) there was a dkms  error
(I have been using rtw89-dkms-git)  from the AUR package for this repo (almost a year)

The update (dkms) gave a bad return status and exited with return status 10. 
My wifi still works and looked at the pacman logs it installed it.

I also reported the error to the Aur package site as well (just wondering if this is specific to Make file here or the package because it was not happening in previous updates)

Here is the make.log that error was pointing out under

/var/lib/dkms/rtw89-dkms/git/build/make.log ;

DKMS make.log for rtw89-dkms-git for kernel 5.15.93-1-MANJARO (x86_64) Sun Feb 12 01:02:35 PM +03 2023 make: Entering directory '/usr/lib/modules/5.15.93-1-MANJARO/build' scripts/Makefile.build:44: /var/lib/dkms/@_PKGBASE@/@PKGVER@/build/Makefile: No such file or directory make[1]: ***No rule to make target '/var/lib/dkms/@_PKGBASE@/@PKGVER@/build/Makefile'. Stop. make:*** [Makefile:1903: /var/lib/dkms/@_PKGBASE@/@PKGVER@/build] Error 2 make: Leaving directory '/usr/lib/modules/5.15.93-1-MANJARO/build'

lwfinger, 2023-02-12:
I provide the dkms configuration written by others in the distribution, but I do not maintain it. You will need to debug this yourself.

closed 2023-02-12 by lwfinger

salimp2009, 2023-02-12:
> I provide the dkms configuration written by others in the distribution, but I do not maintain it. You will need to debug this yourself.

ok I will try . Thank I tought it  is not your deal. Thanks a lot tough . I might again start building from the repo ; not the package .