lwfinger/rtw89 #223: wifi breaks down down after suspending

kkkstra, 2023-02-27:
My laptop is Thinkbook 16+ i5/rtx-2050
Everytime I wake up the laptop from suspending, the wifi seems to break down
And when I try to reboot, the terminal prints `Call to reboot failed: There's already a shutdown or sleep in progress`

This is the syslogs:

```
2月 27 00:12:57 kkkstra-Thinkbook systemd-logind[607]: Delay lock is active (UID 1000/kkkstra, PID 1778/gsd-power) but inhibitor timeout is reached.
2月 27 00:12:57 kkkstra-Thinkbook systemd[1]: Reached target Sleep.
2月 27 00:12:57 kkkstra-Thinkbook systemd[1]: Starting System Suspend...
2月 27 00:12:57 kkkstra-Thinkbook systemd-sleep[58590]: [5] INFO: # Pre-suspend hook
2月 27 00:12:57 kkkstra-Thinkbook systemd-sleep[58590]: [6] INFO: Previous state was: {'type': 'done', 'switch_id': '20230226T210458', 'current_mode': 'integrated'}
2月 27 00:12:57 kkkstra-Thinkbook systemd-sleep[58590]: [6] INFO: Switching option: none
2月 27 00:12:57 kkkstra-Thinkbook systemd-sleep[58590]: [6] INFO: Turning Nvidia GPU back on
2月 27 00:12:57 kkkstra-Thinkbook systemd-sleep[58592]: modinfo: ERROR: Module bbswitch not found.
2月 27 00:12:57 kkkstra-Thinkbook systemd-sleep[58593]: modinfo: ERROR: Module acpi_call not found.
2月 27 00:12:57 kkkstra-Thinkbook systemd-sleep[58590]: [25] INFO: switching=none, nothing to do
2月 27 00:12:57 kkkstra-Thinkbook systemd-sleep[58590]: [25] INFO: Writing state {'type': 'pending_post_resume', 'switch_id': '20230226T210458', 'current_mode': 'integrated'}
2月 27 00:12:57 kkkstra-Thinkbook systemd-sleep[58590]: [25] INFO: Pre-suspend hook completed successfully.
2月 27 00:12:57 kkkstra-Thinkbook systemd-sleep[58587]: Entering sleep state 'suspend'...
2月 27 00:12:57 kkkstra-Thinkbook kernel: PM: suspend entry (s2idle)
2月 27 00:12:57 kkkstra-Thinkbook kernel: Filesystems sync: 0.003 seconds
2月 27 00:12:57 kkkstra-Thinkbook /usr/lib/gdm-x-session[1437]: (II) event24 - Xiaoxin Mouse 5.0: device removed
2月 27 09:13:14 kkkstra-Thinkbook kernel: Freezing user space processes ... (elapsed 0.004 seconds) done.
2月 27 09:13:14 kkkstra-Thinkbook kernel: OOM killer disabled.
2月 27 09:13:14 kkkstra-Thinkbook kernel: Freezing remaining freezable tasks ... (elapsed 0.002 seconds) done.
2月 27 09:13:14 kkkstra-Thinkbook kernel: printk: Suspending console(s) (use no_console_suspend to debug)
2月 27 09:13:14 kkkstra-Thinkbook kernel: e1000e: EEE TX LPI TIMER: 00000011
2月 27 09:13:14 kkkstra-Thinkbook kernel: ACPI: EC: interrupt blocked
2月 27 09:13:14 kkkstra-Thinkbook kernel: ACPI: EC: interrupt unblocked
2月 27 09:13:14 kkkstra-Thinkbook kernel: ACPI: button: The lid device is not compliant to SW_LID.
2月 27 09:13:14 kkkstra-Thinkbook kernel: i915 0000:00:02.0: [drm] GuC firmware i915/adlp_guc_70.bin version 70.5.1
2月 27 09:13:14 kkkstra-Thinkbook kernel: i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc.bin version 7.9.3
2月 27 09:13:14 kkkstra-Thinkbook kernel: i915 0000:00:02.0: [drm] HuC authenticated
2月 27 09:13:14 kkkstra-Thinkbook kernel: i915 0000:00:02.0: [drm] GuC submission enabled
2月 27 09:13:14 kkkstra-Thinkbook kernel: i915 0000:00:02.0: [drm] GuC SLPC enabled
2月 27 09:13:14 kkkstra-Thinkbook kernel: i915 0000:00:02.0: [drm] GuC RC: enabled
2月 27 09:13:14 kkkstra-Thinkbook kernel: nvme nvme0: 16/0/0 default/read/poll queues
2月 27 09:13:14 kkkstra-Thinkbook kernel: usb 3-10: reset full-speed USB device number 4 using xhci_hcd
2月 27 09:13:14 kkkstra-Thinkbook kernel: mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
2月 27 09:13:14 kkkstra-Thinkbook kernel: OOM killer enabled.
2月 27 09:13:14 kkkstra-Thinkbook kernel: Restarting tasks ... 
2月 27 09:13:14 kkkstra-Thinkbook kernel: mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
2月 27 09:13:14 kkkstra-Thinkbook kernel: Bluetooth: hci0: RTL: examining hci_ver=0b hci_rev=000b lmp_ver=0b lmp_subver=8852
2月 27 09:13:14 kkkstra-Thinkbook kernel: Bluetooth: hci0: RTL: rom_version status=0 version=1
2月 27 09:13:14 kkkstra-Thinkbook kernel: Bluetooth: hci0: RTL: loading rtl_bt/rtl8852bu_fw.bin
2月 27 09:13:14 kkkstra-Thinkbook kernel: Bluetooth: hci0: RTL: loading rtl_bt/rtl8852bu_config.bin
2月 27 09:13:14 kkkstra-Thinkbook kernel: Bluetooth: hci0: RTL: cfg_sz 6, total sz 38955
2月 27 09:13:14 kkkstra-Thinkbook kernel: done.
2月 27 09:13:14 kkkstra-Thinkbook kernel: random: crng reseeded on system resumption
2月 27 09:13:14 kkkstra-Thinkbook systemd-logind[607]: Lid opened.
2月 27 09:13:14 kkkstra-Thinkbook /usr/lib/gdm-x-session[1437]: (II) config/udev: removing device Xiaoxin Mouse 5.0
```

lwfinger, 2023-02-27:
Why are you posting this here? Your logs do not show any reference to any rtw89 device. You do have a Realtek Bluetooth device, but it is restarting with no errors.

Have you made any changes in you OS since installing? If not, then you need to explore this problem with your distro, or try installing a different one. The bottom line is that the code handling sleep and resume is totally screwed up.

closed 2023-02-27 by lwfinger

kkkstra, 2023-02-27:
Sorry, but I do have a rtw89 device, which is my Realtek wifi, but it seems to disappear after sleeping...