lwfinger/rtw89 #225: Realtek b852 stop working with new firmware

dshein-alt, 2023-03-10:
HW: Lenovo ThinkPad E15
```
03:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. Device [10ec:b852]
	Subsystem: Lenovo Device [17aa:4853]
	Physical Slot: 0
	Flags: bus master, fast devsel, latency 0, IRQ 75, IOMMU group 11
	I/O ports at 2000 [size=256]
	Memory at fd400000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: [40] Power Management version 3
	Capabilities: [50] MSI: Enable+ Count=1/1 Maskable- 64bit+
	Capabilities: [70] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [148] Device Serial Number 00-de-ad-be-ef-12-34-56
	Capabilities: [158] Latency Tolerance Reporting
	Capabilities: [160] L1 PM Substates
	Kernel driver in use: rtw89_8852be
```

Kernel: 6.1.15
Driver: kernel-modules-rtw89 (git.fce040c)

Wi-Fi module stopped to work with newest linux-firmware installed.

Issue with linux-firmware-20230301:
```
 kernel: rtw89_8852be 0000:03:00.0: enabling device (0000 -> 0003)
 kernel: rtw89_8852be 0000:03:00.0: no suitable firmware found
 kernel: rtw89_8852be 0000:03:00.0: failed to recognize firmware
 kernel: rtw89_8852be 0000:03:00.0: failed to setup chip information
 rtw89_8852be: probe of 0000:03:00.0 failed with error -2
```
Resolved by rolling back to firmware-linux-20230215.

alxwk, 2023-03-11:
No, the firmware is there, but the module refuses to load the newest 0.29 versions (I tried both v0.29.26.0 and v0.29.29.0 from git.kernel.org linux-firmware repository). Probably something changed in the firmware format.

lwfinger, 2023-03-11:
I get the following:
rtw89_8852be 0000:02:00.0: Firmware version 0.29.29.0, cmd version 0, type 5
rtw89_8852be 0000:02:00.0: Firmware version 0.29.29.0, cmd version 0, type 3
rtw89_8852be 0000:02:00.0: chip rfe_type is 1
rtw89_8852be 0000:02:00.0 wls1: renamed from wlan0
wls1: authenticate with a8:63:7d:63:4f:33
wls1: 80 MHz not supported, disabling VHT
wls1: send auth to a8:63:7d:63:4f:33 (try 1/3)
wls1: authenticated
wls1: associate with a8:63:7d:63:4f:33 (try 1/3)
wls1: RX AssocResp from a8:63:7d:63:4f:33 (capab=0x1431 status=0 aid=1)
wls1: associated

No problem with 0.29.29.0 firmware. I am using a 6.3.0-rc1 kernel. I do not have access to a 6.1.15, but I do not think that the rtw8852be driver appeared in the kernel until 6.2.

I then installed the driver from this repo. It reports the following:
rtw89_8852be 0000:02:00.0: Firmware version 0.29.29.0, cmd version 0, type 5
rtw89_8852be 0000:02:00.0: Firmware version 0.29.29.0, cmd version 0, type 3
rtw89_8852be 0000:02:00.0: chip rfe_type is 1
rtw89_8852be 0000:02:00.0 wls1: renamed from wlan0

Are you using the driver from this repo, or did you get it from some other source? If from this repo, you need to do a 'git pull'. If from dkms, it may be pulling the wrong revision.

lwfinger, 2023-03-11:
After I wrote the above, I booted kernel 6.2.2 and got the same problem. I will report the issue to Realtek to see if a patch can be backported to handle firmware with version 0.29.29.0. The solution for you is to clone this repo, make, and install it.

lwfinger, 2023-03-12:
I sent an E-mail message to Realtek notifying them of the problem. I also placed a file with version 0.27.32.1 of the 8852be firmware in my download site. Use the instructions at the end of  README.md to obtain a copy.

dshein-alt, 2023-03-13:
Thank you for the quick response!
I'm glad that my bug report helped to discover an actual issue.
I'm running ALT Linux distro (branch 'Sisyphus'), driver build from this repository (https://github.com/lwfinger/rtw89/commit/fce040c12fbf93bfd904ded48df60dea2c8d4423).
Firmware is automatically built from upstream.

Am I got you right that there is some changes that should be backported to make driver works with newest firmware from Realtek on current kernel?
That mentioned patch is for kernel itself or for kernel module?

lwfinger, 2023-03-13:
Firmware is not "built" from upstream. Firmware is not open source as it would contain considerable intellectual property. As firmware does not execute on the host - only in the CPU on the wifi chip - including it does not violate the open nature of Linux. Realtek submits any new firmware to the linux-firmware repository, which is where your distro obtains it for inclusion in the linux-firmware package.

In my mind, there is no difference between the kernel and the modules that are included in the kernel source. My contact at Realtek says that the code changes required to support version 0.29.x of the firmware are too extensive to backport to kernel 6.2.  BTW, your distro has backported the rtw8852be driver to kernel 6.1. That is already non-standard.

Your only three options are to continue using the firmware-linux-20230215 package, upgrade to the newer firmware but install the firmware from my download page, or to switch to the driver from this repo, which can handle the newest firmware.

closed 2023-03-13 by lwfinger

dshein-alt, 2023-03-14:
About firmware package I've meant it is build to distro's rpm package from upstream repository - my mistake.
Anyway thanks for detailed explanation.
I'll be fine using the current linux-firmware until kernel 6.2 will be available  in my distro repository.

RiderALT, 2023-03-14:
> Your only three options are to continue using the firmware-linux-20230215 package, upgrade to the newer firmware but install the firmware from my download page, or to switch to the driver from this repo, which can handle the newest firmware.

The driver of this repo, of course, does not work with the new Realtek firmware.

lwfinger, 2023-03-14:
On the contrary, the driver in this repo DOES work with either version 0.27.x or 0.29.x of the firmware. Do a 'git pull' and rebuild.

RiderALT, 2023-03-14:
> On the contrary, the driver in this repo DOES work with either version 0.27.x or 0.29.x of the firmware. Do a 'git pull' and rebuild.

thanks, after update it worked fine.