lwfinger/rtw89 #226: WiFi hangs periodically

dmitrmax, 2023-03-14:
Hi, I have HP ProBook 450 G9 running Linux kernel 6.1 with the latest rtw89 driver from this repo and firmware from linux-firmware repository. Approx. twice a day wifi on laptop hangs. I mean by that that it doesn't show up disconnect but there is no traffic between laptop and access point. This can be fixed if I tell NetworkManager to connect to another BSS (or disconnect from current one). After some timeout disconnect happens and then I can successfuly reconnect to my access point.

When this happens I see following in dmesg:
`[422282.703330] rtw89_8852be 0000:03:00.0: FW status = 0xef001100
[422282.703357] rtw89_8852be 0000:03:00.0: FW BADADDR = 0x9a
[422282.703364] rtw89_8852be 0000:03:00.0: FW EPC/RA = 0x0
[422282.703370] rtw89_8852be 0000:03:00.0: FW MISC = 0xb8989787
[422282.703376] rtw89_8852be 0000:03:00.0: R_AX_HALT_C2H = 0x1000
[422282.703381] rtw89_8852be 0000:03:00.0: R_AX_SER_DBG_INFO = 0xf200000c
[422282.703391] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a007
[422282.703422] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a0df
[422282.703450] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a023
[422282.703479] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a105
[422282.703508] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a0d7
[422282.703537] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a02d
[422282.703566] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a0f7
[422282.703595] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a011
[422282.703624] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a037
[422282.703641] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a025
[422282.703667] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a031
[422282.703695] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a105
[422282.703724] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a0eb
[422282.703753] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a0d5
[422282.703782] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a0f3
[422282.703796] rtw89_8852be 0000:03:00.0: --->
                err=0x1000
[422282.703811] rtw89_8852be 0000:03:00.0: R_AX_SER_DBG_INFO =0xf200000c
[422282.703819] rtw89_8852be 0000:03:00.0: R_AX_DMAC_ERR_ISR=0x00000040
[422282.703825] rtw89_8852be 0000:03:00.0: R_AX_DMAC_ERR_IMR=0x00000000
[422282.703830] rtw89_8852be 0000:03:00.0: R_AX_WDE_ERR_FLAG_CFG=0x00000000
[422282.703835] rtw89_8852be 0000:03:00.0: R_AX_PLE_ERR_FLAG_CFG=0x83190000
[422282.703841] rtw89_8852be 0000:03:00.0: R_AX_WDE_ERR_IMR=0xffffffff
[422282.703846] rtw89_8852be 0000:03:00.0: R_AX_WDE_ERR_ISR=0x00000000
[422282.703851] rtw89_8852be 0000:03:00.0: R_AX_PLE_ERR_IMR=0xffffffdf
[422282.703856] rtw89_8852be 0000:03:00.0: R_AX_PLE_ERR_FLAG_ISR=0x02000000
[422282.703861] rtw89_8852be 0000:03:00.0: R_AX_WD_CPUQ_OP_0=0x00000000
[422282.703866] rtw89_8852be 0000:03:00.0: R_AX_WD_CPUQ_OP_1=0x00000000
[422282.703871] rtw89_8852be 0000:03:00.0: R_AX_WD_CPUQ_OP_2=0x00000000
[422282.703875] rtw89_8852be 0000:03:00.0: R_AX_WD_CPUQ_OP_STATUS=0x00000000
[422282.703881] rtw89_8852be 0000:03:00.0: R_AX_PL_CPUQ_OP_0=0x04000000
[422282.703886] rtw89_8852be 0000:03:00.0: R_AX_PL_CPUQ_OP_1=0x000001c0
[422282.703891] rtw89_8852be 0000:03:00.0: R_AX_PL_CPUQ_OP_2=0x01e901e9
[422282.703896] rtw89_8852be 0000:03:00.0: R_AX_PL_CPUQ_OP_STATUS=0x80000000
[422282.703902] rtw89_8852be 0000:03:00.0: R_AX_RXDMA_PKT_INFO_0=0x800081b6
[422282.703907] rtw89_8852be 0000:03:00.0: R_AX_RXDMA_PKT_INFO_1=0x80008000
[422282.703912] rtw89_8852be 0000:03:00.0: R_AX_RXDMA_PKT_INFO_2=0x80018083
[422282.703922] rtw89_8852be 0000:03:00.0: R_AX_CMAC_ERR_ISR [0]=0x00000000
[422282.703929] rtw89_8852be 0000:03:00.0: R_AX_CMAC_FUNC_EN [0]=0xf000003f
[422282.703934] rtw89_8852be 0000:03:00.0: R_AX_CK_EN [0]=0xffffffff
[422282.703940] rtw89_8852be 0000:03:00.0: R_AX_CMAC_ERR_IMR [0]=0x00000000
[422282.703945] rtw89_8852be 0000:03:00.0: R_AX_RPQ_RXBD_IDX =0x00ae00ae
[422282.703951] rtw89_8852be 0000:03:00.0: R_AX_DBG_ERR_FLAG=0x00000000
[422282.703955] rtw89_8852be 0000:03:00.0: R_AX_LBC_WATCHDOG=0x00000081
[422282.703958] rtw89_8852be 0000:03:00.0: <---
[422282.703960] rtw89_8852be 0000:03:00.0: SER catches error: 0x1000
[422282.704366] rtw89_8852be 0000:03:00.0: FW status = 0xef001100
[422282.704389] rtw89_8852be 0000:03:00.0: FW BADADDR = 0x9a
[422282.704396] rtw89_8852be 0000:03:00.0: FW EPC/RA = 0x0
[422282.704402] rtw89_8852be 0000:03:00.0: FW MISC = 0xb8989787
[422282.704407] rtw89_8852be 0000:03:00.0: R_AX_HALT_C2H = 0x1001
[422282.704413] rtw89_8852be 0000:03:00.0: R_AX_SER_DBG_INFO = 0xf200000c
[422282.704423] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a03d
[422282.704452] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a02f
[422282.704480] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a0e1
[422282.704509] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a021
[422282.704538] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a0ed
[422282.704567] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a0d7
[422282.704596] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a027
[422282.704613] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a01f
[422282.704639] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a023
[... 470 lines cut ...]
[423254.834674] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 3
[423254.834886] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 8
[423255.042734] rtw89_8852be 0000:03:00.0: timed out to flush queues
[423255.378936] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 0
[423255.379167] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 3
[423255.379380] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 8
[423255.588707] rtw89_8852be 0000:03:00.0: timed out to flush queues
[423255.588973] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 0
[423255.589192] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 3
[423255.589404] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 8
[423255.790716] rtw89_8852be 0000:03:00.0: timed out to flush queues
[423261.335774] wlp3s0: deauthenticating from 78:45:58:c6:52:2c by local choice (Reason: 3=DEAUTH_LEAVING)
[423261.542855] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 0
[423261.543083] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 3
[423261.543291] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 8
[423261.748741] rtw89_8852be 0000:03:00.0: timed out to flush queues
[423261.830872] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 0
[423261.831104] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 3
[423261.831316] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 8
[423262.034833] rtw89_8852be 0000:03:00.0: timed out to flush queues
[423262.038467] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 0
[423262.038697] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 3
[423262.038899] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 8`

deauthinticating in the end of the log happens by my request.

lwfinger, 2023-03-14:
One problem with HP laptops is where the BIOS handling of PCI devices is wrong. To test if this is your problem, create (as root) a file named /usr/lib/modprobe.d/70-rtw89.conf. To this file, add the lines

options rtw89_pci  disable_aspm_l1-y disable_aspm_l1ss=y
options rtw89pci  disable_aspm_l1-y disable_aspm_l1ss=y

The first line handles the in-kernel version, and the second handles the driver from this repo. After this file is created, run the sequence
sudo modprobe -rv rtw_8852be
sudo modprobe -v rtw_8852be

If you have done it correctly, the second command should produce output like
insmod /lib/modules/<kernel version>/drivers/net/wireless/realtek/rtw89/rtw89pci.ko  disable_aspm_l1-y disable_aspm_l1ss=y 
insmod /lib/modules/<kernel version>/kernel/drivers/net/wireless/realtek/rtw89/rtw_8852b.ko 
insmod /lib/modules/<kernel version>/kernel/drivers/net/wireless/realtek/rtw89/rtw_8852be.ko

The critical parts are the disable stuff at the end of the first line.

dmitrmax, 2023-03-14:
Should be:
options rtw89pci disable_aspm_l1=y disable_aspm_l1ss=y
"=y" instead of "-y" in first parameter.

Thanks for the feedback. I have applied your advices. I will return in couple of days with the result.

By the way - after modules reload was done WiFi failed to reconnect to my access point though it could scan wireless networks. But after reboot it managed to connect. There was no relevant information in dmesg. May be this is an issue of NetworkManager and friends.

lwfinger, 2023-03-16:
You are getting unexpected interrupts. Edit the options lines we created earlier, and add 'disable_clkreq=y' to each of those lines. I do not know if it will help or hurt, but it is our last option available.

leiless, 2023-04-17:
FYI, I have the same problem with my HP Pro Mini 260 G9.
https://github.com/lwfinger/rtw89/issues/181

lwfinger, 2023-04-17:
@leiless: Have you applied the module parameters described in the March 14 postings? (I wish that GitHub would number the comments.)

leiless, 2023-04-18:
> @leiless: Have you applied the module parameters described in the March 14 postings? (I wish that GitHub would number the comments.)

```
$ cat /usr/lib/modprobe.d/70-rtw89.conf
options rtw89_pci disable_aspm_l1-y disable_aspm_l1ss=y disable_clkreq=y
options rtw89pci  disable_aspm_l1-y disable_aspm_l1ss=y disable_clkreq=y
```

I did the same config as of now, and I need some time to check if it'll hang periodically.

lwfinger, 2023-04-18:
Those two lines with ...l1-y were a typo as shown in the comment following the one that I wrote. Each line should have ....l1=y instead.

leiless, 2023-04-18:
Okay, conf updated!

```bash
$ cat /usr/lib/modprobe.d/70-rtw89.conf
options rtw89_pci disable_aspm_l1=y disable_aspm_l1ss=y disable_clkreq=y
options rtw89pci  disable_aspm_l1=y disable_aspm_l1ss=y disable_clkreq=y

$ sudo modprobe -rv rtw_8852be
rmmod rtw_8852be
rmmod rtw89pci
rmmod rtw_8852b
rmmod rtw89core
rmmod mac80211
rmmod libarc4
rmmod cfg80211

$ sudo modprobe -v rtw_8852be
insmod /lib/modules/5.19.0-38-generic/kernel/lib/crypto/libarc4.ko 
insmod /lib/modules/5.19.0-38-generic/kernel/net/wireless/cfg80211.ko 
insmod /lib/modules/5.19.0-38-generic/kernel/net/mac80211/mac80211.ko 
insmod /lib/modules/5.19.0-38-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw89core.ko 
insmod /lib/modules/5.19.0-38-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw89pci.ko  disable_aspm_l1=y disable_aspm_l1ss=y disable_clkreq=y 
insmod /lib/modules/5.19.0-38-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw_8852b.ko 
insmod /lib/modules/5.19.0-38-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw_8852be.ko
```

dmitrmax, 2023-04-20:
Hey! It is been a while since I opened this issue. I've applied disable_clkreq=y to the config but it seems that nothing is changed. Still I get this logs in dmesg. Though the FW status has changed to an another value

 ```
[ 3691.369128] rtw89_8852be 0000:03:00.0: FW status = 0xea001100
[ 3691.369149] rtw89_8852be 0000:03:00.0: FW BADADDR = 0x99
[ 3691.369157] rtw89_8852be 0000:03:00.0: FW EPC/RA = 0x0
[ 3691.369163] rtw89_8852be 0000:03:00.0: FW MISC = 0xb8989787
[ 3691.369170] rtw89_8852be 0000:03:00.0: R_AX_HALT_C2H = 0x1000
[ 3691.369177] rtw89_8852be 0000:03:00.0: R_AX_SER_DBG_INFO = 0xf1000008
[ 3691.369188] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a015
[ 3691.369214] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a0d5
[ 3691.369233] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a0df
[ 3691.369251] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a0e7
[ 3691.369270] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a02b
[ 3691.369288] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a003
[ 3691.369306] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a0e5
[ 3691.369324] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a00f
[ 3691.369342] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a0ef
[ 3691.369360] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a035
[ 3691.369379] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a007
[ 3691.369397] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a0e7
[ 3691.369415] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a025
[ 3691.369433] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a003
[ 3691.369451] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a0e5
[ 3691.369466] rtw89_8852be 0000:03:00.0: --->
               err=0x1000
[ 3691.369473] rtw89_8852be 0000:03:00.0: R_AX_SER_DBG_INFO =0xf1000008
[ 3691.369483] rtw89_8852be 0000:03:00.0: R_AX_DMAC_ERR_ISR=0x00000040
[ 3691.369489] rtw89_8852be 0000:03:00.0: R_AX_DMAC_ERR_IMR=0x00000000
[ 3691.369495] rtw89_8852be 0000:03:00.0: R_AX_WDE_ERR_FLAG_CFG=0x00000000
[ 3691.369501] rtw89_8852be 0000:03:00.0: R_AX_PLE_ERR_FLAG_CFG=0x83190000
[ 3691.369508] rtw89_8852be 0000:03:00.0: R_AX_WDE_ERR_IMR=0xffffffff
[ 3691.369514] rtw89_8852be 0000:03:00.0: R_AX_WDE_ERR_ISR=0x00000000
[ 3691.369520] rtw89_8852be 0000:03:00.0: R_AX_PLE_ERR_IMR=0xffffffdf
[ 3691.369526] rtw89_8852be 0000:03:00.0: R_AX_PLE_ERR_FLAG_ISR=0x02000000
[ 3691.369532] rtw89_8852be 0000:03:00.0: R_AX_WD_CPUQ_OP_0=0x00000000
[ 3691.369538] rtw89_8852be 0000:03:00.0: R_AX_WD_CPUQ_OP_1=0x00000000
[ 3691.369543] rtw89_8852be 0000:03:00.0: R_AX_WD_CPUQ_OP_2=0x00000000
[ 3691.369549] rtw89_8852be 0000:03:00.0: R_AX_WD_CPUQ_OP_STATUS=0x00000000
[ 3691.369555] rtw89_8852be 0000:03:00.0: R_AX_PL_CPUQ_OP_0=0x04000000
[ 3691.369561] rtw89_8852be 0000:03:00.0: R_AX_PL_CPUQ_OP_1=0x000001c0
[ 3691.369567] rtw89_8852be 0000:03:00.0: R_AX_PL_CPUQ_OP_2=0x00d600d6
[ 3691.369573] rtw89_8852be 0000:03:00.0: R_AX_PL_CPUQ_OP_STATUS=0x80000000
[ 3691.369580] rtw89_8852be 0000:03:00.0: R_AX_RXDMA_PKT_INFO_0=0x80008019
[ 3691.369586] rtw89_8852be 0000:03:00.0: R_AX_RXDMA_PKT_INFO_1=0x80008000
[ 3691.369592] rtw89_8852be 0000:03:00.0: R_AX_RXDMA_PKT_INFO_2=0x80f380e2
[ 3691.369604] rtw89_8852be 0000:03:00.0: R_AX_CMAC_ERR_ISR [0]=0x00000000
[ 3691.369612] rtw89_8852be 0000:03:00.0: R_AX_CMAC_FUNC_EN [0]=0xf000003f
[ 3691.369618] rtw89_8852be 0000:03:00.0: R_AX_CK_EN [0]=0xffffffff
[ 3691.369625] rtw89_8852be 0000:03:00.0: R_AX_CMAC_ERR_IMR [0]=0x00000000
[ 3691.369632] rtw89_8852be 0000:03:00.0: R_AX_RPQ_RXBD_IDX =0x00480048
[ 3691.369638] rtw89_8852be 0000:03:00.0: R_AX_DBG_ERR_FLAG=0x00000000
[ 3691.369644] rtw89_8852be 0000:03:00.0: R_AX_LBC_WATCHDOG=0x00000081
[ 3691.369647] rtw89_8852be 0000:03:00.0: <---
[ 3691.369650] rtw89_8852be 0000:03:00.0: SER catches error: 0x1000
[ 3691.369909] rtw89_8852be 0000:03:00.0: FW status = 0xea001100
[ 3691.369915] rtw89_8852be 0000:03:00.0: FW BADADDR = 0x99
[ 3691.369921] rtw89_8852be 0000:03:00.0: FW EPC/RA = 0x0
[ 3691.369927] rtw89_8852be 0000:03:00.0: FW MISC = 0xb8989787
[ 3691.369933] rtw89_8852be 0000:03:00.0: R_AX_HALT_C2H = 0x1001
[ 3691.369939] rtw89_8852be 0000:03:00.0: R_AX_SER_DBG_INFO = 0xf1000008
[ 3691.369950] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a01f
[ 3691.369969] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a0d3
[ 3691.369987] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a0e7
[ 3691.370005] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a0e3
[ 3691.370024] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a0f1
[ 3691.370042] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a0d9
[ 3691.370060] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a02f
[ 3691.370078] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a013
[ 3691.370097] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb898a10d
[... 6 lines cut ...]
[ 3691.370220] rtw89_8852be 0000:03:00.0: SER catches error: 0x1001
[ 3691.370404] rtw89_8852be 0000:03:00.0: FW status = 0xea008100
[ 3691.370410] rtw89_8852be 0000:03:00.0: FW BADADDR = 0x99
[ 3691.370416] rtw89_8852be 0000:03:00.0: FW EPC/RA = 0x0
[ 3691.370422] rtw89_8852be 0000:03:00.0: FW MISC = 0xb898828b
[ 3691.370428] rtw89_8852be 0000:03:00.0: R_AX_HALT_C2H = 0x1002
[ 3691.370434] rtw89_8852be 0000:03:00.0: R_AX_SER_DBG_INFO = 0xf1000008
[ 3691.370444] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb89a37cd
[ 3691.370463] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb89aca9d
[ 3691.370481] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb89a8dfb
[ 3691.370499] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb8930380
[... cut ...]
[ 3691.370609] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb89acd0f
[ 3691.370627] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb89a96e5
[ 3691.370646] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb897d195
[ 3691.370664] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb8934443
[ 3691.370686] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb8935dc7
[ 3691.370704] rtw89_8852be 0000:03:00.0: [ERR]fw PC = 0xb89360af
[ 3691.370719] rtw89_8852be 0000:03:00.0: SER catches error: 0x1002
[ 3691.387653] rtw89_8852be 0000:03:00.0: c2h class 1 func 3 not support
```

lwfinger, 2023-04-20:
Did you add the other two options? Please do the following:

sudo modprobe -rv rtw_8852be
sudo modprobe -v rtw_8852be
lsmod | grep rtw

Post the output of all 3 commands.

TArvela, 2023-04-21:
I also have the same problem with a hp 645 g9 (8852be)
`barnabe@pop-os:~$ sudo modprobe -rv rtw_8852be
rmmod rtw_8852be
rmmod rtw89pci
rmmod rtw_8852b
rmmod rtw89core
rmmod mac80211
rmmod libarc4
rmmod cfg80211
barnabe@pop-os:~$ sudo modprobe -v rtw_8852be
insmod /lib/modules/6.2.6-76060206-generic/kernel/lib/crypto/libarc4.ko 
insmod /lib/modules/6.2.6-76060206-generic/kernel/net/wireless/cfg80211.ko 
insmod /lib/modules/6.2.6-76060206-generic/kernel/net/mac80211/mac80211.ko 
insmod /lib/modules/6.2.6-76060206-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw89core.ko 
insmod /lib/modules/6.2.6-76060206-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw89pci.ko disable_aspm_l1=y disable_aspm_l1ss=y disable_clkreq=y 
insmod /lib/modules/6.2.6-76060206-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw_8852b.ko 
insmod /lib/modules/6.2.6-76060206-generic/kernel/
[logs_wifi_8852be.txt](https://github.com/lwfinger/rtw89/files/11299505/logs_wifi_8852be.txt)
drivers/net/wireless/realtek/rtw89/rtw_8852be.ko 
barnabe@pop-os:~$ lsmod | grep rtw
rtw_8852be             16384  0
rtw_8852b             380928  1 rtw_8852be
rtw89pci               73728  1 rtw_8852be
rtw89core             569344  2 rtw89pci,rtw_8852b
mac80211             1617920  2 rtw89pci,rtw89core
cfg80211             1241088  3 rtw89core,mac80211,rtw_8852b`

You can find attached a full dmesg dump.

I also just tried disabling a pcie power saving feature in the BIOS. I'll post here again if it works.

lwfinger, 2023-04-21:
That was NOT a complete dmesg - you selected only a portion. What I did see is that you seem to be missing the firmware. Please run the following command 'md5sum /lib/firmware/rtw89' and post the output.

TArvela, 2023-04-22:
Sorry about the dmesg, I thought everything was there ...

`md5sum /lib/firmware/rtw89`
returns me 
`md5sum: /lib/firmware/rtw89: Is a directory`

But disabling the pcie power saving feature in the BIOS seems to have worked. My wifi hasn't dropped once since I applied it (approx 3h usage). (It used to drop after 15 minutes in average).

lwfinger, 2023-04-22:
@dmitrmax: Does your BIOS have a similar setting? I know you have an Intel CPU and @TArvela's has an AMD CPU, and may have different options.

leiless, 2023-04-23:
> Those two lines with ...l1-y were a typo as shown in the comment following the one that I wrote. Each line should have ....l1=y instead.

@lwfinger, after I updated the config, the Wi-Fi could still hang sometimes.

Logs: https://pastebin.com/ehZHyNTF

The 8852BE driver was built upon the commit `d6ca1625d5b4b32255c5b2d0d6f9d56ce3474fc2`
```bash
$ cat /usr/lib/modprobe.d/70-rtw89.conf
options rtw89_pci disable_aspm_l1=y disable_aspm_l1ss=y disable_clkreq=y
options rtw89pci  disable_aspm_l1=y disable_aspm_l1ss=y disable_clkreq=y

$ lscpu | grep -w Model
Model name:                      12th Gen Intel(R) Core(TM) i5-1235U

$ md5sum /lib/firmware/rtw89/rt*
49951f548b87ea0258d128195ef6e0cf  /lib/firmware/rtw89/rtl8852au_config.bin
2dc64f98c75ec454a1b18962eefe7b2f  /lib/firmware/rtw89/rtl8852au_fw.bin
6edd77ccced6eda33850d302853fe665  /lib/firmware/rtw89/rtw8852a_fw.bin
a3317e9646c8e669c7c2d1fc19a6109d  /lib/firmware/rtw89/rtw8852b_fw.bin
ace8c45cf09d44fe99e2480a94abc2ff  /lib/firmware/rtw89/rtw8852c_fw.bin
```

lwfinger, 2023-04-23:
The firmware you have for rtw8852a and rtw8852b are the latest,but you are missing rtw8852b_fw_1.bin. That is a later version with some improvements. Your firmware for rtw8852c is not the latest, but that should not make any difference. My md5sum values are as follows:

6edd77ccced6eda33850d302853fe665  rtw89/rtw8852a_fw.bin
97cc27ddbaacfcfd9f4b4ad425d114d3  rtw89/rtw8852b_fw-1.bin
a3317e9646c8e669c7c2d1fc19a6109d  rtw89/rtw8852b_fw.bin
ceed5cd7b2066d7a8b99e73e9056d3c1  rtw89/rtw8852c_fw.bin

lwfinger, 2023-04-23:
I just pushed commit 6169c9bf50cb153f10319e. I contains all the changes that will be in kernel 6.4.

leiless, 2023-04-24:
> The firmware you have for rtw8852a and rtw8852b are the latest,but you are missing rtw8852b_fw_1.bin. That is a later version with some improvements. Your firmware for rtw8852c is not the latest, [...]

Can I just download the latest firmware from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/rtw89 and replace those `bin` files under the `/lib/firmware/rtw89/`?

Will it work?

lwfinger, 2023-04-24:
Yes. That is where I get mine.

leiless, 2023-04-24:
@lwfinger, I updated the firmware.

BTW, still using the driver built from commit `d6ca1625d5b4b32255c5b2d0d6f9d56ce3474fc2`.

See if it'll still hang.

```bash
$ md5sum /lib/firmware/rtw89/*
49951f548b87ea0258d128195ef6e0cf  /lib/firmware/rtw89/rtl8852au_config.bin
2dc64f98c75ec454a1b18962eefe7b2f  /lib/firmware/rtw89/rtl8852au_fw.bin
6edd77ccced6eda33850d302853fe665  /lib/firmware/rtw89/rtw8852a_fw.bin
97cc27ddbaacfcfd9f4b4ad425d114d3  /lib/firmware/rtw89/rtw8852b_fw-1.bin
a3317e9646c8e669c7c2d1fc19a6109d  /lib/firmware/rtw89/rtw8852b_fw.bin
ceed5cd7b2066d7a8b99e73e9056d3c1  /lib/firmware/rtw89/rtw8852c_fw.bin
```

lwfinger, 2023-04-24:
I contacted Realtek and was pointed to a patch that has been submitted but not yet merged. It was incorporated into this repo and a new version was pushed. Please pull and test.

TArvela, 2023-04-24:
I pulled the changes, reinstalled, removed the BIOS changes I did before (now running again with enabled pcie power savings) and it seems to work fine. No drop in 2-3 hours.

I also never thanked you for maintaining this driver, so thank you very much :smiley: :tada: !!!!

lwfinger, 2023-04-24:
Glad to hear. I am going to close the issue. If there are any changes, re-open it.

closed 2023-04-24 by lwfinger