lwfinger/rtw89 #228: Low performance after upgrade to kernel 6.2

erotallgeese, 2023-03-23:
Hi,

After upgrading from kernel 5.15 to 6.2, the throughput about RTL8852BE/RTL8852CE turn from 900Mbps/1.5Gbps to 500Mbps (5G, ax).
I use iperf3 to check its throughput.
I manually build same latest rtw89 for both kernel.

During testing, i notice one difference is that its ksoftirqd would be very high in kernel 6.2, following is the top snap:
```
top - 10:24:51 up  1:27,  2 users,  load average: 0.83, 0.41, 0.25
Tasks: 216 total,   2 running, 214 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.4 us,  3.6 sy,  0.0 ni, 70.3 id,  0.7 wa,  0.0 hi, 24.9 si,  0.0 st
MiB Mem :  13814.2 total,  10664.3 free,   1475.2 used,   1674.8 buff/cache
MiB Swap:  16384.0 total,  16384.0 free,      0.0 used.  12047.8 avail Mem

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
     19 root      20   0       0      0      0 R  69.1   0.0   1:20.90 ksoftirqd/1
   3755 root     -51   0       0      0      0 S  29.9   0.0   5:42.41 irq/137-rtw89_pci
   4019 wind      20   0    7196   4096   2560 S   8.6   0.0   0:04.96 iperf3
   2814 wind      20   0 4512384 250020 105716 S   1.6   1.8   0:19.59 gnome-shell
   2651 wind      20   0  487508  75136  46228 S   1.3   0.5   0:11.31 Xorg
   4021 root      20   0    6420   3584   2816 R   1.3   0.0   0:00.74 top
   3911 root       0 -20       0      0      0 I   1.0   0.0   0:01.06 kworker/u9:2-rtw89_tx_wq
   1611 root       0 -20       0      0      0 I   0.7   0.0   0:00.99 kworker/u9:0-rtw89_tx_wq
   4022 root       0 -20       0      0      0 I   0.7   0.0   0:00.34 kworker/u9:3-rtw89_tx_wq

```

I still checking the kernel difference.
Any advice would be appreciated.
Thanks

lwfinger, 2023-03-23:
What kernel configuration parameter did you get wrong? Documenting that might be useful.

erotallgeese, 2023-03-24:
It's about KASAN, i just forget it...