lwfinger/rtw89 #230: 8852BE - NULL pointer 8852BE

a1ks, 2023-03-25:
Hello, 

I am experiencing a null pointer error when another device attempts to connect to 8852BE hotspot. This issue is persistent on the latest commits. However, when I use this specific commit (https://github.com/lwfinger/rtw89/commit/4f3464d85d070039085c6d948d9f418f860babf4), everything works correctly.

I have included logs from /var/log/messages (Linux version 6.2.8) for your reference.

```
Mar 25 17:13:56 localhost vmunix: [ 5256.998248] rtw89_8852be 0000:03:00.0: timed out to flush queues
Mar 25 17:13:57 localhost vmunix: [ 5257.164032] BUG: kernel NULL pointer dereference, address: 00000000000000d4
Mar 25 17:13:57 localhost vmunix: [ 5257.164041] #PF: supervisor read access in kernel mode
Mar 25 17:13:57 localhost vmunix: [ 5257.164044] #PF: error_code(0x0000) - not-present page
Mar 25 17:13:57 localhost vmunix: [ 5257.164047] PGD 0 P4D 0
Mar 25 17:13:57 localhost vmunix: [ 5257.164051] Oops: 0000 [#1] PREEMPT SMP NOPTI
Mar 25 17:13:57 localhost vmunix: [ 5257.164055] CPU: 1 PID: 152 Comm: kworker/u33:0 Tainted: G           O       6.2.8 #1 a92d919946d09b6345d397ca9e031c63ca992f2d
Mar 25 17:13:57 localhost vmunix: [ 5257.164063] Workqueue: rtw89_tx_wq rtw89_core_txq_work [rtw89core]
Mar 25 17:13:57 localhost vmunix: [ 5257.164102] RIP: 0010:rtw89_core_tx_update_desc_info+0x271/0x8e0 [rtw89core]
Mar 25 17:13:57 localhost vmunix: [ 5257.164124] Code: 8b 81 d8 04 00 00 80 b9 e2 03 00 00 00 48 8d 04 c0 41 8b 94 c4 ec 00 00 00 b8 04 00 00 00 75 0a 31 c0 85 d2 0f 95 c0 c1 e0 02 <41> 8b 94 96 d4 00 00 00 85 d2 74 07 f3 48 0f bc d2 01 d0 66 89 43
Mar 25 17:13:57 localhost vmunix: [ 5257.164127] RSP: 0018:ffffb9d140cebcf8 EFLAGS: 00010256
Mar 25 17:13:57 localhost vmunix: [ 5257.164130] RAX: 0000000000000000 RBX: ffffb9d140cebd50 RCX: ffff9ecb5e851c08
Mar 25 17:13:57 localhost vmunix: [ 5257.164133] RDX: 0000000000000000 RSI: ffff9ecc21204a30 RDI: ffff9ecd37ecc100
Mar 25 17:13:57 localhost vmunix: [ 5257.164135] RBP: ffffb9d140cebd38 R08: ffff9ecc3adcce00 R09: 0000000000000001
Mar 25 17:13:57 localhost vmunix: [ 5257.164137] R10: 0000000000000008 R11: 0000000000000002 R12: ffff9ecb60282080
Mar 25 17:13:57 localhost vmunix: [ 5257.164139] R13: ffff9ecd37ecc100 R14: 0000000000000000 R15: 0000000000000000
Mar 25 17:13:57 localhost vmunix: [ 5257.164141] FS:  0000000000000000(0000) GS:ffff9ece0fc40000(0000) knlGS:0000000000000000
Mar 25 17:13:57 localhost vmunix: [ 5257.164143] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Mar 25 17:13:57 localhost vmunix: [ 5257.164146] CR2: 00000000000000d4 CR3: 0000000034a10000 CR4: 0000000000750ee0
Mar 25 17:13:57 localhost vmunix: [ 5257.164148] PKRU: 55555554
Mar 25 17:13:57 localhost vmunix: [ 5257.164150] Call Trace:
Mar 25 17:13:57 localhost vmunix: [ 5257.164152]  <TASK>
Mar 25 17:13:57 localhost vmunix: [ 5257.164155]  ? ieee80211_tx_dequeue+0x5ba/0x7e0 [mac80211 24f2988cb8c1c2c328e5bcefd139c9b4b7c7dc0a]
Mar 25 17:13:57 localhost vmunix: [ 5257.164232]  rtw89_core_tx_write+0x80/0x120 [rtw89core b21a30f59e0895b36b855e2913194e3b369899ec]
Mar 25 17:13:57 localhost vmunix: [ 5257.164255]  rtw89_core_txq_work+0x1ac/0x5a0 [rtw89core b21a30f59e0895b36b855e2913194e3b369899ec]
Mar 25 17:13:57 localhost vmunix: [ 5257.164276]  process_one_work+0x21d/0x430
Mar 25 17:13:57 localhost vmunix: [ 5257.164289]  worker_thread+0x54/0x3e0
Mar 25 17:13:57 localhost vmunix: [ 5257.164293]  ? __pfx_worker_thread+0x10/0x10
Mar 25 17:13:57 localhost vmunix: [ 5257.164297]  kthread+0xf4/0x120
Mar 25 17:13:57 localhost vmunix: [ 5257.164306]  ? __pfx_kthread+0x10/0x10
Mar 25 17:13:57 localhost vmunix: [ 5257.164311]  ret_from_fork+0x2c/0x50
Mar 25 17:13:57 localhost vmunix: [ 5257.164318]  </TASK>
Mar 25 17:13:57 localhost vmunix: [ 5257.164476]  pata_acpi nls_iso8859_1 wp512 serpent_generic xts dm_crypt hid_apple hid_generic usbhid hid uas usb_storage ahci libahci
Mar 25 17:13:57 localhost vmunix: [ 5257.164491] CR2: 00000000000000d4
Mar 25 17:13:57 localhost vmunix: [ 5257.164496] ---[ end trace 0000000000000000 ]---
Mar 25 17:13:57 localhost vmunix: [ 5257.575646] pstore: backend (efi_pstore) writing error (-5)
Mar 25 17:13:57 localhost vmunix: [ 5257.575651] RIP: 0010:rtw89_core_tx_update_desc_info+0x271/0x8e0 [rtw89core]
Mar 25 17:13:57 localhost vmunix: [ 5257.575671] Code: 8b 81 d8 04 00 00 80 b9 e2 03 00 00 00 48 8d 04 c0 41 8b 94 c4 ec 00 00 00 b8 04 00 00 00 75 0a 31 c0 85 d2 0f 95 c0 c1 e0 02 <41> 8b 94 96 d4 00 00 00 85 d2 74 07 f3 48 0f bc d2 01 d0 66 89 43
Mar 25 17:13:57 localhost vmunix: [ 5257.575674] RSP: 0018:ffffb9d140cebcf8 EFLAGS: 00010256
Mar 25 17:13:57 localhost vmunix: [ 5257.575677] RAX: 0000000000000000 RBX: ffffb9d140cebd50 RCX: ffff9ecb5e851c08
Mar 25 17:13:57 localhost vmunix: [ 5257.575679] RDX: 0000000000000000 RSI: ffff9ecc21204a30 RDI: ffff9ecd37ecc100
Mar 25 17:13:57 localhost vmunix: [ 5257.575680] RBP: ffffb9d140cebd38 R08: ffff9ecc3adcce00 R09: 0000000000000001
Mar 25 17:13:57 localhost vmunix: [ 5257.575682] R10: 0000000000000008 R11: 0000000000000002 R12: ffff9ecb60282080
Mar 25 17:13:57 localhost vmunix: [ 5257.575683] R13: ffff9ecd37ecc100 R14: 0000000000000000 R15: 0000000000000000
Mar 25 17:13:57 localhost vmunix: [ 5257.575685] FS:  0000000000000000(0000) GS:ffff9ece0fc40000(0000) knlGS:0000000000000000
Mar 25 17:13:57 localhost vmunix: [ 5257.575688] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Mar 25 17:13:57 localhost vmunix: [ 5257.575691] CR2: 00000000000000d4 CR3: 000000028c6c8000 CR4: 0000000000750ee0
Mar 25 17:13:57 localhost vmunix: [ 5257.575694] PKRU: 55555554
Mar 25 17:13:57 localhost vmunix: [ 5257.575696] note: kworker/u33:0[152] exited with irqs disabled
```

lwfinger, 2023-03-25:
I think I see where the NULL dereference is happening, and I have added a test with a printout. Please do a pull and rebuild the driver. The new version will likely log a warning, but it should not crash. Once we know that I am in the right place, then we can attack the cause.

a1ks, 2023-03-25:
After pull and rebuild same crash and same /var/log/message ("skb NULL in ..." does not appear here)

lwfinger, 2023-03-27:
I removed the error messge and added some new debugging statements. Please run again and report back.

a1ks, 2023-03-27:
No crash this time! But hotspot don't work

```
Mar 27 09:55:59 localhost vmunix: [    1.810644] rtw89_8852be: disagrees about version of symbol rtw89_pci_fill_txaddr_info
Mar 27 09:55:59 localhost vmunix: [    1.810646] rtw89_8852be: Unknown symbol rtw89_pci_fill_txaddr_info (err -22)
Mar 27 09:55:59 localhost vmunix: [    1.810649] rtw89_8852be: disagrees about version of symbol rtw89_pci_recognize_intrs
Mar 27 09:55:59 localhost vmunix: [    1.810649] rtw89_8852be: Unknown symbol rtw89_pci_recognize_intrs (err -22)
Mar 27 09:55:59 localhost vmunix: [    1.810651] rtw89_8852be: disagrees about version of symbol rtw89_pci_config_intr_mask
Mar 27 09:55:59 localhost vmunix: [    1.810652] rtw89_8852be: Unknown symbol rtw89_pci_config_intr_mask (err -22)
Mar 27 09:55:59 localhost vmunix: [    1.810653] rtw89_8852be: disagrees about version of symbol rtw89_pci_ltr_set
Mar 27 09:55:59 localhost vmunix: [    1.810653] rtw89_8852be: Unknown symbol rtw89_pci_ltr_set (err -22)
Mar 27 09:55:59 localhost vmunix: [    1.810655] rtw89_8852be: disagrees about version of symbol rtw8852b_chip_info
Mar 27 09:55:59 localhost vmunix: [    1.810656] rtw89_8852be: Unknown symbol rtw8852b_chip_info (err -22)
Mar 27 09:55:59 localhost vmunix: [    1.810658] rtw89_8852be: disagrees about version of symbol rtw89_pci_enable_intr
Mar 27 09:55:59 localhost vmunix: [    1.810658] rtw89_8852be: Unknown symbol rtw89_pci_enable_intr (err -22)
Mar 27 09:55:59 localhost vmunix: [    1.810662] rtw89_8852be: disagrees about version of symbol rtw89_pci_disable_intr
Mar 27 09:55:59 localhost vmunix: [    1.810662] rtw89_8852be: Unknown symbol rtw89_pci_disable_intr (err -22)
...
Mar 27 09:56:03 localhost vmunix: [   11.961334] desc_info->data_rate is NULL in rtw89_core_tx_update_mgmt_info!
...
```

lwfinger, 2023-03-27:
Your kernel and your headers do not agree, but you did get the output I needed, namely "desc_info->data_rate is NULL".

How are you setting up your AP. I put an 8852be in my machine, and I would lite to try duplicaring your results here.

A1Eon, 2023-03-27:
I turn on AP using the default gui of Gnome NetworkManager. Here is autogenerated  configuration file "/etc/NetworkManager/system-connections/Hotspot.nmconnection" :
```
id=Hotspot
uuid=<some uuid here>
type=wifi
autoconnect=false
interface-name=wlo1
timestamp=1676817840

[wifi]
mode=ap
ssid=hotspot_name

[wifi-security]
group=ccmp;
key-mgmt=wpa-psk
pairwise=ccmp;
proto=rsn;
psk=1234

[ipv4]
method=shared

[ipv6]
addr-gen-mode=default
method=auto

[proxy]
```
To activate this AP, I used the command 
```
nmcli con up Hotspot
```

lwfinger, 2023-04-14:
There have been some changes in wireless-next, which I have pushed to this repo. Please try again.

a1ks, 2023-04-24:
I tested the new version at the last commit https://github.com/lwfinger/rtw89/commit/83f33c87e716a9cdb1c52e70a51d91e12fa7e1b0, now the system doesn't crash anymore. Thanks!

lwfinger, 2023-04-24:
Good. I will close this isssue, Re-open it if things change.

closed 2023-10-02 by lwfinger