lwfinger/rtw89 #233: [ERR]invalid fw dynamic header len

helperShang, 2023-04-02:
- Linux rock-5b 5.10.110-rockchip-rk3588 aarch64 GNU/Linux
- lspci
```
0002:21:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device b852
	Subsystem: Hewlett-Packard Company Device 88e3
	Flags: fast devsel, IRQ 132
	I/O ports at 1000 [size=256]
	Memory at f2200000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: [40] Power Management version 3
	Capabilities: [50] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [70] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [148] Device Serial Number 00-e0-4c-ff-fe-88-52-01
	Capabilities: [158] Latency Tolerance Reporting
	Capabilities: [160] L1 PM Substates
	Kernel modules: rtw_8852be
```
### log
```
[    6.470298] rtw89_8852be 0002:21:00.0: Firmware version 67.84.89.80, cmd version 62, type 1
[    6.477440] rtw89_8852be 0002:21:00.0: [ERR]invalid fw dynamic header len
[    6.477445] rtw89_8852be 0002:21:00.0: parse fw header fail
[    6.477450] rtw89_8852be 0002:21:00.0: [ERR]fwdl 0x1E0 = 0x1
[    6.477455] rtw89_8852be 0002:21:00.0: [ERR]fwdl 0x83F2 = 0x0
[    6.477464] rtw89_8852be 0002:21:00.0: [ERR]fw PC = 0xb89002e8
[    6.477479] rtw89_8852be 0002:21:00.0: [ERR]fw PC = 0xb89002fc
[    6.477493] rtw89_8852be 0002:21:00.0: [ERR]fw PC = 0xb8900384
[    6.477508] rtw89_8852be 0002:21:00.0: [ERR]fw PC = 0xb8900388
[    6.477522] rtw89_8852be 0002:21:00.0: [ERR]fw PC = 0xb8900384
[    6.477536] rtw89_8852be 0002:21:00.0: [ERR]fw PC = 0xb8900384
[    6.477551] rtw89_8852be 0002:21:00.0: [ERR]fw PC = 0xb8900394
[    6.477565] rtw89_8852be 0002:21:00.0: [ERR]fw PC = 0xb8900384
[    6.477580] rtw89_8852be 0002:21:00.0: [ERR]fw PC = 0xb8900384
[    6.477594] rtw89_8852be 0002:21:00.0: [ERR]fw PC = 0xb8900384
[    6.477608] rtw89_8852be 0002:21:00.0: [ERR]fw PC = 0xb8900384
[    6.477623] rtw89_8852be 0002:21:00.0: [ERR]fw PC = 0xb8900384
[    6.477637] rtw89_8852be 0002:21:00.0: [ERR]fw PC = 0xb8900384
[    6.477652] rtw89_8852be 0002:21:00.0: [ERR]fw PC = 0xb8900384
[    6.477666] rtw89_8852be 0002:21:00.0: [ERR]fw PC = 0xb8900394
[    6.477680] rtw89_8852be 0002:21:00.0: failed to setup chip information
[    6.485530] rtw89_8852be: probe of 0002:21:00.0 failed with error -22
```

lwfinger, 2023-04-02:
Your firmware version is completely wrong. Mine shows 0.27.32.1. Reload the firmware-realtek package. or whatever your distro calls it. Otherwise clone git://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git and copy the files from the rtw89 directory to /usr/lib/firmware/rtw89/.

helperShang, 2023-04-03:
working , thinks

closed 2023-04-03 by helperShang