lwfinger/rtw89 #235: 8852be error

satishwagh, 2023-04-11:
Following error came when compiled against 6.1.x LTS kernel.

```
rtw89_8852be 0000:57:00.0: MAC has already powered on
rtw89_8852be 0000:57:00.0: [ERR]pci config read 719
rtw89_8852be 0000:57:00.0: [ERR] pcie autok fail -22
rtw89_8852be 0000:57:00.0: failed to setup chip information
rtw89_8852be: probe of 0000:57:00.0 failed with error -22
```

I even replaced the 6.2.10 stable kernel's rtw89  directory into 6.1.23 kernel directory. Compiled it successfully but it throws same error as mentioned above.

lwfinger, 2023-04-11:
What version of the firmware do you have? You did not include that part of the dmesg output.

lwfinger, 2023-04-13:
Those xz files are supplied by my distro. They are available for openSUSE users, the way they should for all distros. Unfortunately, some are brain-dead.

You can get the nonn-compressed versions by cloning the git repo at git://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git. That will get all of the kernel firmware.

satishwagh, 2023-04-13:
I downloaded said firmware but still same error.
I had tried various firmware.

Firmware versions of rtw8852b_fw-1.bin as:
0.29.29.1
0.29.29.0
0.29.26.0

Firmware versions of rtw8852b_fw.bin as:
0.27.32.1
0.27.32.0

No matter which firmware I used, it throws same error every time.

satishwagh, 2023-04-13:
> Attach your complete dmesg output.

[dmesg.txt](https://github.com/lwfinger/rtw89/files/11223316/dmesg.txt)

lwfinger, 2023-04-13:
Thanks. Unfortunately, I see nothing wrong there.

You say that Ubuntu 23.04 works. What version are you running?

satishwagh, 2023-04-13:
I see bunch of errors in dmesg. Those messages are mentioned in first post itself.

I am not sure about the version you are asking. 
I checked with Ubuntu 23.04 desktop edition. I even copied the Ubuntu 23.04's firmware onto my machine but got the same result.

lwfinger, 2023-04-13:
What I meant is that I saw nothing new.

Two days ago, you said that "Ubuntu 23.04 is working flawlessly." What version/distro is failing? I am confused!!

satishwagh, 2023-04-13:
Oh Sorry! My bad.

I am having my own custom Linux distribution based upon [LFS](https://www.linuxfromscratch.org/lfs/).
It does have custom compiled kernel. Kernel version I tried so far for this issue are from 6.1.15 till 6.1.23 and 6.2.10.

For cross checking, I booted with Ubuntu-23.04 on same hardware and found that 8852be device works perfectly fine with Ubuntu. So I copied the /lib/firmware/rtw89 directory from Ubuntu and replaced it with my custom OS at /lib/firmware/rtw89. But still the 8852be device fails to work.

lwfinger, 2023-04-13:
I now understand more. I have never seen that kind of problem with any kernel using an openSUSE distro. Obviously, Ubuntu works as well as dose Fedora.

My only conclusion is that you have some problem with the PCI subsystem with your version of LFS. If testing here could be accomplished with downloading and installing a standard distro, I could do that, but I am not prepared to build my complete system as would be required. I do not think there is anything else I can do. You have found one of the huge downsides of building your own. When one of the major distros builds a system, it gets lots of testing before it is released. It may take some time before the latest software gets released, but I find that a reasonable tradeoff. Besides, I run openSUSE Tumbleweed, which is a rolling release. It has a 6.2.x kernel, gcc v13, and a lot of other cutting-edge software. The only thing I build is a kernel, because I want to test the latest kernel HEAD - 6.3.0-rc6 at the moment.

closed 2023-04-13 by lwfinger

satishwagh, 2023-04-13:
One more thing: on the same hardware, with the same custom OS, if I switch to customly build 5.15.73 LTS kernel then this 8852be works smooth. Just because I had upgraded kernel to 6.x, this problem occurs.
Please note that I used the same .config file for both 5.x and 6.x kernel

satishwagh, 2023-04-13:
Yes I've taken care of that. There is no single module in kernel which starts with rtw.
All related kernel module are compiled from rtw89 git repo only.
For 6.2.10 kernel, I had tried two approach.
First: I deselected rtw89 from kernel tree and compiled rtw89 git repo.
Second: I selected the rtw89 modules from kernel tree. No external module involved.

In both approach, I got the same result as stated in first post.
Is there any specific congratulation in kernel 6.x which might be affecting and causing this error?

lwfinger, 2023-04-14:
New updates were merged into wireless-next today. I added them to this repo and pushed the new code.

lwfinger, 2023-04-16:
I just booted kernel 6.2.10 fron openSUSE today, and the in-kernel 8852be driver works fine. If you were missing a kernel feature, I would expect a buidl or module loading problem. For guidance, I have attached my kernel configuration file. Depending on your hardware, it may not boot as I have trimmed a lot of features and drivers.
[config.txt](https://github.com/lwfinger/rtw89/files/11243292/config.txt)

aghand0ur, 2023-11-22:
```
sudo dmesg | grep rtw
[    3.157597] rtw89_8852be 0000:09:00.0: loaded firmware rtw89/rtw8852b_fw-1.bin
[    3.157852] rtw89_8852be 0000:09:00.0: enabling device (0000 -> 0003)
[    3.159296] rtw89_8852be 0000:09:00.0: Firmware version 0.29.29.3 (91835e0f), cmd version 0, type 5
[    3.159299] rtw89_8852be 0000:09:00.0: Firmware version 0.29.29.3 (91835e0f), cmd version 0, type 3
[    3.159305] rtw89_8852be 0000:09:00.0: MAC has already powered on
[    3.165011] rtw89_8852be 0000:09:00.0: [ERR]pci config read 719
[    3.165016] rtw89_8852be 0000:09:00.0: [ERR] pcie autok fail -22
[    3.165020] rtw89_8852be 0000:09:00.0: failed to setup chip information
[    3.166179] rtw89_8852be: probe of 0000:09:00.0 failed with error -22
```

aghand0ur, 2023-11-22:
> Following error came when compiled against 6.1.x LTS kernel.
>
> ```
> rtw89_8852be 0000:57:00.0: MAC has already powered on
> rtw89_8852be 0000:57:00.0: [ERR]pci config read 719
> rtw89_8852be 0000:57:00.0: [...]

I was having the same error. I downloaded the latest firmware from here:
https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/rtw89
but I got this error:
```
[    3.210042] rtw89_8852be 0000:09:00.0: loaded firmware rtw89/rtw8852b_fw-1.bin
[    3.210309] rtw89_8852be 0000:09:00.0: enabling device (0000 -> 0003)
[    3.211636] rtw89_8852be 0000:09:00.0: Unknown firmware header version 10
[    3.211642] rtw89_8852be 0000:09:00.0: failed to recognize firmware
[    3.211643] rtw89_8852be 0000:09:00.0: failed to setup chip information
[    3.212687] rtw89_8852be: probe of 0000:09:00.0 failed with error -2

```

I am using a Lenovo LOQ with Ubuntu 22.04 kernel `6.2.0-37-generic`

lwfinger, 2023-11-22:
You have a faulty version of the firmware.Run the command 'md5sum /lib/firmware/rtw89/*'. The correct result should be 
4df8d6e2459b9409837f64a4be879288  /lib/firmware/rtw89/rtw8852b_fw-1.bin

The firmware is reported as
[    6.935970] rtw89_8852be 0000:02:00.0: Firmware version 0.29.29.5 (da87cccd), cmd version 0, type 5
[    6.935977] rtw89_8852be 0000:02:00.0: Firmware version 0.29.29.5 (da87cccd), cmd version 0, type 3

aghand0ur, 2023-11-25:
Where can I locate the right firmware version?

lwfinger, 2023-11-25:
You should read the last sentence of README.md.

aghand0ur, 2023-11-25:
I re-downloaded the faulty firmware. 
dmesg shows that the latest firmware is detected but still receiving an error:

```
[    3.208217] rtw89_8852be 0000:09:00.0: loaded firmware rtw89/rtw8852b_fw-1.bin
[    3.208456] rtw89_8852be 0000:09:00.0: enabling device (0000 -> 0003)
[    3.210270] rtw89_8852be 0000:09:00.0: Firmware version 0.29.29.5 (da87cccd), cmd version 0, type 5
[    3.210275] rtw89_8852be 0000:09:00.0: Firmware version 0.29.29.5 (da87cccd), cmd version 0, type 3
[    3.215040] rtw89_8852be 0000:09:00.0: [ERR]pci config read 719
[    3.215054] rtw89_8852be 0000:09:00.0: [ERR] pcie autok fail -22
[    3.215058] rtw89_8852be 0000:09:00.0: failed to setup chip information
[    3.216711] rtw89_8852be: probe of 0000:09:00.0 failed with error -22
```

lwfinger, 2023-11-25:
I am suspiciousof the Lenovo BIOS . Create an options file using the following command:

sudo nano /etc/modprobe.d/70-rtw8852be.conf

Add the following 2 lines to that file:

options rtw89pci disable_aspm_l1=y disable_aspm_l1ss=y
options rtw89_pci disable_aspm_l1=y disable_aspm_l1ss=y

You can test whenther you got the file correctly entered with

sudo modprobe -rv rtw_8852be
sudo modprobe -v rtw_8852be

The output of the second command should show the disable_aspm... options appended to the rtw89pci load line.

If this does not help, you will need to post your problem on the linux-wireless mailing list <linux-wireless@vger.kernel.org>.

aghand0ur, 2023-11-25:
right, this is the output when adding the module:

```
insmod /lib/modules/6.2.0-37-generic/kernel/lib/crypto/libarc4.ko 
insmod /lib/modules/6.2.0-37-generic/kernel/net/mac80211/mac80211.ko 
insmod /lib/modules/6.2.0-37-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw89core.ko 
insmod /lib/modules/6.2.0-37-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw89pci.ko disable_aspm_l1=y disable_aspm_l1ss=y 
insmod /lib/modules/6.2.0-37-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw_8852b.ko 
insmod /lib/modules/6.2.0-37-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw_8852be.ko
```
but dmseg output did not change:

```
[  188.333979] rtw89_8852be 0000:09:00.0: loaded firmware rtw89/rtw8852b_fw-1.bin
[  188.335858] rtw89_8852be 0000:09:00.0: Firmware version 0.29.29.5 (da87cccd), cmd version 0, type 5
[  188.335861] rtw89_8852be 0000:09:00.0: Firmware version 0.29.29.5 (da87cccd), cmd version 0, type 3
[  188.335867] rtw89_8852be 0000:09:00.0: MAC has already powered on
[  188.341597] rtw89_8852be 0000:09:00.0: [ERR]pci config read 719
[  188.341599] rtw89_8852be 0000:09:00.0: [ERR] pcie autok fail -22
[  188.341600] rtw89_8852be 0000:09:00.0: failed to setup chip information
[  188.342425] rtw89_8852be: probe of 0000:09:00.0 failed with error -22

```