lwfinger/rtw89 #238: dmesg errors for rtl 8852be

boxerab, 2023-04-19:
Thanks so much for maintaining these drivers - wouldn't have wifi connectivity otherwise.

I installed the driver following instructions in this link :
https://discussion.fedoraproject.org/t/how-to-install-the-driver-for-wifi-chip-rtl8852be/68307

I have errors in dmesg, but wifi works. Is there a way of fixing these errors ?

This is for a Lenovo Thinkbook 16p gen3 laptop.

Thanks!

lwfinger, 2023-04-19:
Your user-space code is starting the driver twice. That is not the driver's fault. If you cannot find the reason, it is safe to ignore everything here. Do you see the same thing if you do a reboot?

boxerab, 2023-04-19:
Thanks, I think the reason is that I ran the install script twice,  since I got an error the first time after 
`sudo systemctl restart systemd-modules-load.service`

closed 2023-04-23 by boxerab