lwfinger/rtw89 #239: rtw89_8852be no longer working after kernel update: 6.2.11-200

boxerab, 2023-04-23:
After update, I ran

```
sudo dnf --refresh upgrade
sudo reboot
```

Here is the log from dmesg :

```
[    1.463697] rtw89_8852be 0000:02:00.0: Direct firmware load for rtw89/rtw8852b_fw-1.bin failed with error -2
[    1.463707] rtw89_8852be 0000:02:00.0: Direct firmware load for rtw89/rtw8852b_fw.bin failed with error -2
[    1.463708] rtw89_8852be 0000:02:00.0: failed to early request firmware: -2
[    1.463754] rtw89_8852be 0000:02:00.0: enabling device (0000 -> 0003)
[    1.463781] rtw89_8852be 0000:02:00.0: Direct firmware load for rtw89/rtw8852b_fw.bin failed with error -2
[    1.464766] rtw89_8852be 0000:02:00.0: failed to wait firmware completion
[    1.464768] rtw89_8852be 0000:02:00.0: failed to setup chip information
[    1.465052] rtw89_8852be: probe of 0000:02:00.0 failed with error -22

```

lwfinger, 2023-04-23:
I think your firmware is wrong. Please post the output of
md5sum /lib/firmware/rtw89/*

boxerab, 2023-04-23:
thanks, @lwfinger that was probably the issue. I ended up re-installing and then updating firmware, and it appears that 6.2.11 has drivers for my card.

closed 2023-04-23 by boxerab

Jirubizu, 2023-05-12:
@boxerab did you reinstall the entire os? I am trying to not do that

lwfinger, 2023-05-12:
You should not need to reinstall everything. That is one of the reasons for maintining this repo. Note that 6.2.11 may have drivers for your card, there are significant improvements in 6.3.0 and 6.4-rc1, and other improvements will be in 6.5 in about 6 months. ALL of those changes are in this repo!