lwfinger/rtw89 #240: rtw89_8852be network card ejects randomly

dgloukhman, 2023-04-23:
my wifi randomly stops working and  probably the network card gets ejected (don't see the wifi in my settings anymore).
this started occurring after the recent kernel updates and subsequent repo updates.
frequency: this happens some times only once a day but sometimes 5-6 times in a row (shortly after reboot)

System:
```
OS: Pop!_OS 22.04 LTS x86_64 
Host: 21ECS00000 ThinkPad E14 Gen 4 
Kernel: 6.2.6-76060206-generic 
```

md5sum /lib/firmware/rtw89/* :

```
49951f548b87ea0258d128195ef6e0cf  /lib/firmware/rtw89/rtl8852au_config.bin
2dc64f98c75ec454a1b18962eefe7b2f  /lib/firmware/rtw89/rtl8852au_fw.bin
6edd77ccced6eda33850d302853fe665  /lib/firmware/rtw89/rtw8852a_fw.bin
a3317e9646c8e669c7c2d1fc19a6109d  /lib/firmware/rtw89/rtw8852b_fw.bin
5f7c159627f2ecd6a8411960f0758c07  /lib/firmware/rtw89/rtw8852c_fw.bin
```

dmesg output:
```
[38071.404065] pcieport 0000:00:02.3: pciehp: Slot(0): Link Down
[38071.404073] pcieport 0000:00:02.3: pciehp: Slot(0): Card not present
[38071.439959] rtw89_8852be 0000:03:00.0: failed to pre-release fwcmd
[38071.442045] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.442081] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.463967] wlp3s0: deauthenticating from 3c:a6:2f:c6:e1:7c by local choice (Reason: 3=DEAUTH_LEAVING)
[38071.464964] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 0
[38071.465098] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 1
[38071.465214] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 2
[38071.465330] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 3
[38071.465446] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 8
[38071.465562] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 9
[38071.465739] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 0
[38071.465855] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 1
[38071.465971] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 2
[38071.466089] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 3
[38071.466206] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 8
[38071.466324] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 9
[38071.471949] rtw89_8852be 0000:03:00.0: FW does not process h2c registers
[38071.472124] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472159] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472191] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.472223] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.472255] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472288] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472321] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472355] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472388] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472421] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472454] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472487] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472521] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472554] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472588] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472623] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472658] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472692] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472725] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472758] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472791] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472824] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472857] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472890] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472923] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472956] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.472989] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.473022] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.473055] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.473088] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.473121] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.473154] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.473186] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.473219] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.473251] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.473282] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.473314] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.473345] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.473376] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.473407] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.473438] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.473469] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.473500] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.473531] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.473562] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.473595] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.473626] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.473660] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.473691] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.473723] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.473754] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.473790] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.473821] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.473852] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.473883] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.473914] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.473945] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.473977] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.474008] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.474039] rtw89_8852be 0000:03:00.0: read rf busy swsi
[38071.474071] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.479932] rtw89_8852be 0000:03:00.0: FW does not process h2c registers
[38071.480053] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 0
[38071.480176] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 1
[38071.480292] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 2
[38071.480408] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 3
[38071.480524] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 8
[38071.480640] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 9
[38071.480689] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.480722] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.480754] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.480786] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.480818] rtw89_8852be 0000:03:00.0: write rf busy swsi
[38071.533623] rtw89_8852be 0000:03:00.0: xtal si not ready(W): offset=90 val=10 mask=10
[38071.533842] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 0
[38071.533974] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 1
[38071.534105] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 2
[38071.534229] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 3
[38071.534348] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 8
[38071.534466] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 9
[38071.534470] rtw89_8852be 0000:03:00.0: failed to pre-release fwcmd
[38071.534628] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 0
[38071.534746] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 1
[38071.534863] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 2
[38071.534974] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 3
[38071.535083] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 8
[38071.535194] rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 9
[38071.677356] pci 0000:03:00.0: Removing from iommu group 11
[38071.677774] pcieport 0000:00:02.3: pciehp: Slot(0): Card present
[38073.795912] pcieport 0000:00:02.3: pciehp: Slot(0): No link
[38073.795921] pcieport 0000:00:02.3: pciehp: Slot(0): Card present
[38075.907896] pcieport 0000:00:02.3: pciehp: Slot(0): No link
```

lwfinger, 2023-04-23:
The PCIe card does not eject. These are the kinds of errors we see from some HP and Lenovo BIOSs. In the BIOS setup, is there an option to disable PCI power saving? That has helped with some HP laptops. I do not know if it is available for Lenovo.

The other thing to do is to invoke the module options to disable lspm and PCI clockreq.
As root, create a file named  /usr/lib/modprobe.d/70-rtw89.conf and add the following lines to that file:

options rtw89_pci disable_clkreq=y disable_aspm_l1=y disable_aspm_l1ss=y
options rtw89pci disable_clkreq=y disable_aspm_l1=y disable_aspm_l1ss=y

The first line handles the in-kernel version, and the second handles the driver from this repo. After this file is created, run the sequence
sudo modprobe -rv rtw_8852be
sudo modprobe -v rtw_8852be

If you have done it correctly, the second command should produce output like
insmod /lib/modules//drivers/net/wireless/realtek/rtw89/rtw89pci.ko disable_clkreq=y disable_aspm_l1=y disable_aspm_l1ss=y
insmod /lib/modules//kernel/drivers/net/wireless/realtek/rtw89/rtw_8852b.ko
insmod /lib/modules//kernel/drivers/net/wireless/realtek/rtw89/rtw_8852be.ko

The critical parts are the disable stuff at the end of the first line.

lwfinger, 2023-04-24:
Over the weekend, I contacted Realtek about these problems. They pointed me to a submitted, but not yet incorporated patch, which has been merged into this repo. Do a 'git pull' and rebuiild. Report if this helps.

dgloukhman, 2023-04-24:
thanks, I did both the disable stuff and repo update. So far it works. will report back in a week

lwfinger, 2023-04-24:
Sounds good.

dgloukhman, 2023-07-01:
> The PCIe card does not eject. These are the kinds of errors we see from some HP and Lenovo BIOSs. In the BIOS setup, is there an option to disable PCI power saving? That has helped with some HP [...]

btw. this definitely fixes the issue. haven't had any issues yet

closed 2023-07-01 by dgloukhman

DiviyamPathak, 2023-08-18:
> Over the weekend, I contacted Realtek about these problems. They pointed me to a submitted, but not yet incorporated patch, which has been merged into this repo. Do a 'git pull' and rebuiild. Report if this helps.

I did this , it worked for 2 startups and then same problem occurred but after making  /usr/lib/modprobe.d/70-rtw89.conf  it has started working. Also after adding files shutdown the laptop and then turn it on again. I tried to reboot it didnt work. 
also my dmesg was same as OP

scott-parkhill, 2023-09-09:
@lwfinger I'm having the same problem. Wifi is connecting on reboot for about 30 seconds to 2 minutes before dropping. My `dmesg` output is below, I'm on Ubuntu 22.04.3 on a Lenovo IdeaPad 3. I have created that `70-rtw89.conf` file with the options in it and did the `modprobe` to no success. After building an trying to `modprobe` as well, I usually get an error saying `Exec format error`, not sure if this is helpful at all. Only sometimes will `sudo modprobe rtw89pci` work.

Also thanks for this repo because it's saved my butt, bought this laptop then couldn't use linux on it I was so sad lol. But now I can thanks to this!

```
[  113.754333] rtw89_8852be 0000:02:00.0: FW does not process h2c registers
[  113.754519] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.754557] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.754591] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  113.754626] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  113.754660] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.754697] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.754733] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.754768] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.754806] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.754842] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.754879] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.754915] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.754953] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.754992] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.755028] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.755065] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.755101] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.755136] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.755173] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.755208] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.755244] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.755282] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.755320] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.755356] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.755393] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.755430] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.755466] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.755502] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.755539] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.755576] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.755611] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.755646] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.755682] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.755717] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  113.755750] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  113.755786] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  113.755821] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.755855] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.755888] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  113.755922] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  113.755957] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  113.755992] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.756029] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  113.756063] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  113.756100] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  113.756134] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  113.756168] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.756202] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.756242] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  113.756276] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  113.756310] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  113.756344] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.756376] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  113.756409] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  113.756443] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  113.756476] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  113.756510] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  113.756543] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  113.762147] rtw89_8852be 0000:02:00.0: FW does not process h2c registers
[  113.762304] rtw89_8852be 0000:02:00.0: timed out to flush pci txch: 0
[  113.762427] rtw89_8852be 0000:02:00.0: timed out to flush pci txch: 1
[  113.762549] rtw89_8852be 0000:02:00.0: timed out to flush pci txch: 2
[  113.762671] rtw89_8852be 0000:02:00.0: timed out to flush pci txch: 3
[  113.762793] rtw89_8852be 0000:02:00.0: timed out to flush pci txch: 8
[  113.762911] rtw89_8852be 0000:02:00.0: timed out to flush pci txch: 9
```

Tambalance, 2024-02-11:
> The PCIe card does not eject. These are the kinds of errors we see from some HP and Lenovo BIOSs. In the BIOS setup, is there an option to disable PCI power saving? That has helped with some HP [...]

Hi,
I am starting out with Linux (Ubuntu 22.04 LTS), a total beginner, and I am experiencing similar issues with my WiFi on a HP laptop with Realtek 8852be.
Could you please assist me with the instructions above and clarify for me the following:

_> As root, create a file named /usr/lib/modprobe.d/70-rtw89.conf and add the following lines to that file:_

Question 1: How do I become root? Is it sudo in terminal?
Question 2: How do I create the file and where should I save it?

Thank you for your help. It is much appreciated.

conflictboy, 2024-02-11:
> > The PCIe card does not eject. These are the kinds of errors we see from some HP and Lenovo BIOSs. In the BIOS setup, is there an option to disable PCI power saving? That has helped with some HP laptops. I do not know if it is available for Lenovo.
> > The other thing to do is to invoke the module options to disable lspm and PCI clockreq. As root, create a file named /usr/lib/modprobe.d/70-rtw89.conf and add the following lines to that file:
> > options rtw89_pci disable_clkreq=y disable_aspm_l1=y disable_aspm_l1ss=y options rtw89pci disable_clkreq=y disable_aspm_l1=y disable_aspm_l1ss=y
> > The first line handles the in-kernel version, and the second handles the driver from this repo. After this file is created, run the sequence sudo modprobe -rv rtw_8852be sudo modprobe -v rtw_8852be
> > If you have done it correctly, the second command should produce output like insmod /lib/modules//drivers/net/wireless/realtek/rtw89/rtw89pci.ko disable_clkreq=y disable_aspm_l1=y disable_aspm_l1ss=y insmod /lib/modules//kernel/drivers/net/wireless/realtek/rtw89/rtw_8852b.ko insmod /lib/modules//kernel/drivers/net/wireless/realtek/rtw89/rtw_8852be.ko
> > The critical parts are the disable stuff at the end of the first line.
> 
> Hi, I am starting out with Linux (OS: Ubuntu 22.04.3 LTS x86_64, Kernel: 6.5.0-18-generic, DE: GNOME 42.9), a total beginner, and I am experiencing similar issues with my WiFi on a HP laptop with Realtek 8852be. Could you please assist me with the instructions above and clarify for me the following:
> 
> _> As root, create a file named /usr/lib/modprobe.d/70-rtw89.conf and add the following lines to that file:_
> 
> Question 1: How do I become root? Is it sudo in terminal? Question 2: How do I create the file and where should I save it?
> 
> Thank you for your help. It is much appreciated.

Hi,
I had a similar problem but now it should be fixed - so you should do:
```
git clone https://github.com/lwfinger/rtw89.git
cd rtw89
make
sudo make install
```

Your questions:
Question 1: yes, it's sudo in terminal
Question 2: /usr/lib/modprobe.d/70-rtw89.conf  is path of file. You can create it by `sudo touch /usr/lib/modprobe.d/70-rtw89.conf `. But you should do it only if previous install with `sudo make install` won't working (you have to restart PC)

Tambalance, 2024-02-12:
Hello @conflictboy
Thank you a lot for your help. I was able to successfully install rtw89.
Time will tell if it solved the problem, but I am hopeful for now :).

If some other newbie is reading this and experiencing the same issues:
After I typed _make_ in my terminal and hit enter, I got an error saying something about missing gcc-12.
I was able to fix that with:
`sudo add-apt-repository ppa:ubuntu-toolchain-r/test
sudo apt update
sudo apt install gcc-12 g++-12`
After this I typed in _make_ again and then sudo make install. Rtw89 successfully installed.

Tambalance, 2024-02-12:
Hmm.. unfortunately now it turns out that after reboot Ubuntu is not able to find any WiFi at all. Probably have to reinstall the OS, as I dont have a ethernet port on my laptop.

Thank you anyway for trying to help.

If there are any other suggestions I would be happy to try, thanks.

lwfinger, 2024-02-12:
Post the output of 'lsmod | grep rtw'

Tambalance, 2024-02-12:
[image]

Tambalance, 2024-02-16:
I just noticed, that if I start the computer and via the advanced boot menu, and go to an older version of the kernel, then the wifi works and here's the output of
`lsmod | grep rtw`
[image]

Tambalance, 2024-02-17:
Hi,

Thank you for your assistance, it is much appreciated. To be honest I am not quite sure I understand what you mean with building and installing the drivers here.

Here's the WiFi not working at all in kernel 6.5.0-18-generic:
[dmesg_bad.txt](https://github.com/lwfinger/rtw89/files/14318221/dmesg_bad.txt)

And here's the WiFi dropping in 6.5.0-17-generic:
[dmesg_drops.txt](https://github.com/lwfinger/rtw89/files/14318222/dmesg_drops.txt)

lwfinger, 2024-02-17:
Whomever built the 6.5.0-18  kernel apparently did not include the drivers in rtw89 in the build. The dmesg_bad file sees the BT part of the RTW8852BE, but not the wifi.

Your dmesg_drops file had the kernel buffer wrap around, thus I cannot see what caused the problem to start. Only run that kernel for 10 minutes gefore generating dmesg_drops.

You are posting in the issues section for a set of drivers at GitHub.com. The README.md file here describes how to download the code and build the drivers locally. That is what I meant, but with a 6.5 kernel, you should not need to do that step. The repo is mostly for users with older kernels.

Tambalance, 2024-02-18:
There was a new dropout of WiFi and I managed to save the output of sudo dmesg just a couple of seconds after the dropout.

Some lines that seems odd to me are:
[ 7133.207089] rtw89_8852be 0000:01:00.0: FW does not process h2c registers
[ 7133.207095] rtw89_8852be 0000:01:00.0: HW scan failed with status: -110
[ 7133.207164] warning: `iwconfig' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211
[ 7134.052056] rtw89_8852be 0000:01:00.0: Update probe request failed
[ 7134.166090] rtw89_8852be 0000:01:00.0: FW does not process h2c registers

Maybe there's some clues in those lines, or should I maybe start looking for some other network adapter?

[dmesg_drops3.txt](https://github.com/lwfinger/rtw89/files/14324462/dmesg_drops3.txt)

Thank you again for your guidance.

lwfinger, 2024-02-19:
You quite likely have either a Lenovo or an HP laptop. You should read the section of README.md that covers how to to overcome the defective BIOS code in those models. That should fix your problems.

bgarcial, 2024-03-07:
@lwfinger First at all, thanks for the repo.
I have a Lenovo laptop. I've followed the Readme process, and I ended up here in [this part](https://github.com/lwfinger/rtw89?tab=readme-ov-file#option-configuration)

>IMPORTANT: If you have an HP or Lenovo laptop, Their BIOS does not handle the PCIe interface correctly. To compensate, run the following command: sudo cp 70-rtw89.conf /etc/modprobe.d/. Then unload the drivers and reload. You should see the options appended to the end of the rtw89_pci or rtw89pci load line.

I did the above part this is the content of the `/etc/modprobe.d/70-rtw89.conf` file:
```
 set options for faulty HP and Lenovo BIOS code
options rtw89_pci disable_aspm_l1=y disable_aspm_l1ss
options rtw89pci disable_aspm_l1=y disable_aspm_l1ss
```
And when uploading the driver and reloading it, I got this output:

```
➜  rtw89 git:(main) sudo cp 70-rtw89.conf /etc/modprobe.d 
➜  rtw89 git:(main) sudo modprobe -rv rtw_8852ae
➜  rtw89 git:(main) sudo modprobe -rv rtw89core  
rmmod rtw89core
➜  rtw89 git:(main) sudo modprobe -v rtw_8852ae
insmod /lib/modules/6.7.7-200.fc39.x86_64/kernel/drivers/net/wireless/realtek/rtw89/rtw89core.ko.xz 
insmod /lib/modules/6.7.7-200.fc39.x86_64/kernel/drivers/net/wireless/realtek/rtw89/rtw89pci.ko.xz disable_aspm_l1=y disable_aspm_l1ss 
insmod /lib/modules/6.7.7-200.fc39.x86_64/kernel/drivers/net/wireless/realtek/rtw89/rtw_8852a.ko.xz 
insmod /lib/modules/6.7.7-200.fc39.x86_64/kernel/drivers/net/wireless/realtek/rtw89/rtw_8852ae.ko.xz 
```

However when reading the part where the process says:
>If it turns out that your system needs one of the other configuration options, then do the following:
> sudo nano /etc/modprobe.d/<dev_name>.conf

>There, enter the line below:

>options <driver_name> <<driver_option_name>>=<value>
 
I am confused about what should be the value of the `dev_name.conf`.. Should it be `rtw89pci`?
and what should I enter as a content of that file, I mean instead of this:
```
options <driver_name> <<driver_option_name>>=<value>
```

By the way I am using Fedora 39 with this kernel version `6.7.7-200.fc39.x86_64` and I have disabled UEFI SecureBoot in the machine in order to be able to install Fedora alongside Windows. Not sure if this is an important detail, (about having UEFI Secureboot off from the BIOS), since my installation/compilation process was form the all distro option in the readme (and not with module signing for secureboot)

I will appreciate your input on that as I am trying to overcome this same issue on my workstation.

lwfinger, 2024-03-07:
Those names withing <> are symbolic names. Substitute the actual names. For <dev_name.conf>, you already have a name!

Is your workstation a Lenovo or HP?

bgarcial, 2024-03-07:
@lwfinger MY workstation is Lenovo ThinkPad P16s Gen 2

>For <dev_name.conf>, you already have a name!
Do you mean I should use `rtw89pci` and its values. 

Other thing my wifi card is not realtek it is an Intel Corporation Raptor Lake PCH CNVi WiFi (rev 01), not sure if that applies

lwfinger, 2024-03-08:
With an Intel card, why are you posting here? Intel issues need to go to linux-wireless@vger.kernel.org. We only work with Realtek devices here!!!

bgarcial, 2024-03-09:
@lwfinger yes, sorry, it was that I got confused because I don't know why but in somehow my kernel got updates about realtek devices too. That is why I had to verify the real chipset my laptop has. 
Thanks for your time after all. :)

admmq, 2025-01-12:
Hello everyone!
For someone who has the same issue but the solution above didn't work.
In my case (Lenovo ThinkPad E14 Gen5) I needed to add options for  rtw89_core (and I also added options for rtw89core, which is not really needed because I don't have such module, but anyway). So, in my case the end config file looks this:
```
options rtw89_pci disable_clkreq=y disable_aspm_l1=y disable_aspm_l1ss=y
options rtw89pci disable_clkreq=y disable_aspm_l1=y disable_aspm_l1ss=y
options rtw89_core disable_ps_mode=y
options rtw89core disable_ps_mode=y
```