lwfinger/rtw89 #241: WiFi packet drop issue on RTL8852AE

alou-S, 2023-04-27:
Device : RTL8852AE
Laptop : HP Omen 16-c0xxx

So I am in a university hostel that has tons of wireless interference. There are some weird packet drop issues that I experience when using WiFi that maybe caused due to interference(?)

Interference as in whenever somebody walks through the hostel corridor or enters my room the WiFi just breaks for a second or two (basically if I run a constant ping test with a 0.05s interval between pings are like lots of dropped packets and erratic ping at the exact moment people walk through the corridor). This issue happens both when I use my Mobile Hotspot and use the University Provided WiFi network so I don't think its related to their routers.

Additionally, I have observed that a similar pattern of packet drops can be triggered by opening the WiFi menu and making the card scan for networks, suggesting that it may be related to this action.

There are also some instances of this issue where there are absolutely no packets going through for 5 to 7 seconds.

I have also noticed that this problem does not occur on Windows. While Windows does have random packet drops, they are not as severe as the ones I experience on Linux, lasting only for one or two packets instead of an entire 2 seconds.

 Please let me know if you need any further information to troubleshoot this issue.

lwfinger, 2023-04-27:
I have quite a bit of experience with the 8852be, but none with the 8852ae. The performance of the 8852be is not very good in the 2.4 GHz band, but much better in the 5 GHz band. I expect the same for the ae variety. Which are you using? The command 'iw dev <wifi name> info' will tell.

alou-S, 2023-04-28:
Actually I forgot to mention earlier but the issue occurs on both the 2.4GHz and 5GHz bands.

alou-S, 2023-04-28:
Okay after enabling debug logs and `wpa_supplicant` I could fully confirm that the lag was caused due to the `bgscan` module in `wpa_supplicant` and is probably not related to the driver itself.

lwfinger, 2023-04-28:
I am glad to hear this news. A scan does cause interruptions. I am not sure what triggers a scan other than opening the NetworkManager applet on my KDE desktop.

alou-S, 2023-05-01:
@lwfinger Seems like the scan was triggered due to the interference. Disabling the scan just makes the issue extremely more bearable and doesn't fix it.

So now instead of 5 seconds of terrible lag during the scans I still have minor amounts of stuttering during said "interference". So well the route cause of the issue is not fixed. Any ideas?

lwfinger, 2023-05-01:
This is not an issue I can fix. Read the last paragraph of README.md and submit it to the Linux wireless mailing list.