lwfinger/rtw89 #242: rtl8852be Higole Gole1 Pro mini PC

gdan342, 2023-05-01:
Myself and a few others amongst the community who have this device are having issues getting wireless to work on linux. I have tried multiple distros, installing from source, installing from AUR with no luck. There is no technical documentation for this device for linux use, no forums or anything yet. I am no expert, novice at best. Its been a few days and many hours trying to get this to work. If anyone could point me in the right direction, I would be forever thankful. I'm wanting to make a guide so others don't run into this issue with this device, but for the life of me cannot figure it out on my own.

lwfinger, 2023-05-01:
The 8852be driver was not included in the kernel until v6.3. That kernel will soon be appearing in some distros, but not yet.

You obviously found this repo, which backports the kernel driver from kernel 6.4+ to older kernels. All you need to do is install git and do a 'git clone https://github.com/lwfinger/rtw89.git' and 'cd rtw89'. Then install the pre-requisites as described in README.md. Finally, do 'make' and 'sudo make install', then reboot and wireless should be working.

If you have already done this, and not gotten anywhere, then run 'lspci -nn' and post the output.

gdan342, 2023-05-01:
Yes have tried the steps on Debian and Arch based distros. With arch have also tried with the DKMS from AUR with no luck. I have various adapters I install drivers for all the time with no issue, this is why im so stumped. Here is output of lspci -nn. `00:00.0 Host bridge [0600]: Intel Corporation Gemini Lake Host Bridge [8086:31f0] (rev 06)
00:00.1 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor Dynamic Platform and Thermal Framework Processor Participant [8086:318c] (rev 06)
00:02.0 VGA compatible controller [0300]: Intel Corporation GeminiLake [UHD Graphics 600] [8086:3185] (rev 06)
00:0e.0 Audio device [0403]: Intel Corporation Celeron/Pentium Silver Processor High Definition Audio [8086:3198] (rev 06)
00:0f.0 Communication controller [0780]: Intel Corporation Celeron/Pentium Silver Processor Trusted Execution Engine Interface [8086:319a] (rev 06)
00:12.0 SATA controller [0106]: Intel Corporation Celeron/Pentium Silver Processor SATA Controller [8086:31e3] (rev 06)
00:13.0 PCI bridge [0604]: Intel Corporation Gemini Lake PCI Express Root Port [8086:31da] (rev f6)
00:14.0 PCI bridge [0604]: Intel Corporation Gemini Lake PCI Express Root Port [8086:31d7] (rev f6)
00:15.0 USB controller [0c03]: Intel Corporation Celeron/Pentium Silver Processor USB 3.0 xHCI Controller [8086:31a8] (rev 06)
00:16.0 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor I2C 0 [8086:31ac] (rev 06)
00:16.1 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor I2C 1 [8086:31ae] (rev 06)
00:16.2 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor I2C 2 [8086:31b0] (rev 06)
00:16.3 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor I2C 3 [8086:31b2] (rev 06)
00:17.0 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor I2C 4 [8086:31b4] (rev 06)
00:17.1 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor I2C 5 [8086:31b6] (rev 06)
00:17.2 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor I2C 6 [8086:31b8] (rev 06)
00:17.3 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor I2C 7 [8086:31ba] (rev 06)
00:18.0 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor Serial IO UART Host Controller [8086:31bc] (rev 06)
00:18.1 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor Serial IO UART Host Controller [8086:31be] (rev 06)
00:18.2 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor Serial IO UART Host Controller [8086:31c0] (rev 06)
00:18.3 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor Serial IO UART Host Controller [8086:31ee] (rev 06)
00:19.0 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor Serial IO SPI Host Controller [8086:31c2] (rev 06)
00:19.1 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor Serial IO SPI Host Controller [8086:31c4] (rev 06)
00:19.2 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor Serial IO SPI Host Controller [8086:31c6] (rev 06)
00:1c.0 SD Host controller [0805]: Intel Corporation Celeron/Pentium Silver Processor SDA Standard Compliant SD Host Controller [8086:31cc] (rev 06)
00:1e.0 SD Host controller [0805]: Intel Corporation Device [8086:31d0] (rev 06)
00:1f.0 ISA bridge [0601]: Intel Corporation Celeron/Pentium Silver Processor LPC Controller [8086:31e8] (rev 06)
00:1f.1 SMBus [0c05]: Intel Corporation Celeron/Pentium Silver Processor Gaussian Mixture Model [8086:31d4] (rev 06)
01:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 15)
02:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. Device [10ec:b852]
`

lwfinger, 2023-05-01:
The 10ec:b852 device is handled by this driver.

Do the following from the rtw89 directory:
make clean
git pull
make > build.txt 2>&1
sudo make install >> build.txt 2>&1
sudo modprobe -v rtw_8852be >> build.txt 2>&1

Attach the file build.txt.

gdan342, 2023-05-01:
Heres the build.txt. Very confused why it says "device or resource busy"

`  CC [M]  /home/tj/rtw89/core.o
  CC [M]  /home/tj/rtw89/chan.o
  CC [M]  /home/tj/rtw89/mac80211.o
  CC [M]  /home/tj/rtw89/mac.o
  CC [M]  /home/tj/rtw89/phy.o
  CC [M]  /home/tj/rtw89/fw.o
  CC [M]  /home/tj/rtw89/cam.o
  CC [M]  /home/tj/rtw89/efuse.o
  CC [M]  /home/tj/rtw89/regd.o
  CC [M]  /home/tj/rtw89/sar.o
  CC [M]  /home/tj/rtw89/coex.o
  CC [M]  /home/tj/rtw89/ps.o
  CC [M]  /home/tj/rtw89/debug.o
  CC [M]  /home/tj/rtw89/ser.o
  CC [M]  /home/tj/rtw89/wow.o
  LD [M]  /home/tj/rtw89/rtw89core.o
  CC [M]  /home/tj/rtw89/rtw8852a.o
  CC [M]  /home/tj/rtw89/rtw8852a_table.o
  CC [M]  /home/tj/rtw89/rtw8852a_rfk.o
  CC [M]  /home/tj/rtw89/rtw8852a_rfk_table.o
  LD [M]  /home/tj/rtw89/rtw_8852a.o
  CC [M]  /home/tj/rtw89/rtw8852ae.o
  LD [M]  /home/tj/rtw89/rtw_8852ae.o
  CC [M]  /home/tj/rtw89/rtw8852b.o
  CC [M]  /home/tj/rtw89/rtw8852b_table.o
  CC [M]  /home/tj/rtw89/rtw8852b_rfk.o
  CC [M]  /home/tj/rtw89/rtw8852b_rfk_table.o
  LD [M]  /home/tj/rtw89/rtw_8852b.o
  CC [M]  /home/tj/rtw89/rtw8852be.o
  LD [M]  /home/tj/rtw89/rtw_8852be.o
  CC [M]  /home/tj/rtw89/rtw8852c.o
  CC [M]  /home/tj/rtw89/rtw8852c_table.o
  CC [M]  /home/tj/rtw89/rtw8852c_rfk.o
  CC [M]  /home/tj/rtw89/rtw8852c_rfk_table.o
  LD [M]  /home/tj/rtw89/rtw_8852c.o
  CC [M]  /home/tj/rtw89/rtw8852ce.o
  LD [M]  /home/tj/rtw89/rtw_8852ce.o
  CC [M]  /home/tj/rtw89/pci.o
  LD [M]  /home/tj/rtw89/rtw89pci.o
  MODPOST /home/tj/rtw89/Module.symvers
  CC [M]  /home/tj/rtw89/rtw89core.mod.o
  LD [M]  /home/tj/rtw89/rtw89core.ko
  BTF [M] /home/tj/rtw89/rtw89core.ko
  CC [M]  /home/tj/rtw89/rtw_8852a.mod.o
  LD [M]  /home/tj/rtw89/rtw_8852a.ko
  BTF [M] /home/tj/rtw89/rtw_8852a.ko
  CC [M]  /home/tj/rtw89/rtw_8852ae.mod.o
  LD [M]  /home/tj/rtw89/rtw_8852ae.ko
  BTF [M] /home/tj/rtw89/rtw_8852ae.ko
  CC [M]  /home/tj/rtw89/rtw_8852b.mod.o
  LD [M]  /home/tj/rtw89/rtw_8852b.ko
  BTF [M] /home/tj/rtw89/rtw_8852b.ko
  CC [M]  /home/tj/rtw89/rtw_8852be.mod.o
  LD [M]  /home/tj/rtw89/rtw_8852be.ko
  BTF [M] /home/tj/rtw89/rtw_8852be.ko
  CC [M]  /home/tj/rtw89/rtw_8852c.mod.o
  LD [M]  /home/tj/rtw89/rtw_8852c.ko
  BTF [M] /home/tj/rtw89/rtw_8852c.ko
  CC [M]  /home/tj/rtw89/rtw_8852ce.mod.o
  LD [M]  /home/tj/rtw89/rtw_8852ce.ko
  BTF [M] /home/tj/rtw89/rtw_8852ce.ko
  CC [M]  /home/tj/rtw89/rtw89pci.mod.o
  LD [M]  /home/tj/rtw89/rtw89pci.ko
  BTF [M] /home/tj/rtw89/rtw89pci.ko
make -C /lib/modules/6.2.13-arch1-1/build M=/home/tj/rtw89 modules
Install rtw89 SUCCESS
modprobe: ERROR: could not insert 'rtw_8852be': Device or resource busy
insmod /lib/modules/6.2.13-arch1-1/kernel/drivers/net/wireless/realtek/rtw89/rtw_8852be.ko 
`

lwfinger, 2023-05-02:
The driver was already loaded. I forgot the unload step. Do a reboot, then run the command 'lsmod | grep rtw' and post the output.

gdan342, 2023-05-02:
`rtw89_8852be           16384  0
rtw_8852b             360448  1 rtw89_8852be
rtw89pci               86016  1 rtw89_8852be
rtw89core             626688  2 rtw89pci,rtw_8852b
mac80211             1490944  2 rtw89pci,rtw89core
cfg80211             1277952  4 rtw89core,mac80211,88XXau,rtw_8852b
`

lwfinger, 2023-05-02:
All of your posts are wonky. Are you posting them from Windows?

You have a mixture of modules from the kernel, and those from this repo. That is guaranteed to fail, as I state in the first paragraph of README.md. Blacklist rtw89_8852be.

gdan342, 2023-05-02:
I'm posting from my main machine that runs Artix, connected to Higole machine over ssh. There are a bunch of different ways of blacklisting modules I'm finding. Some in grub file, some in a blacklist.conf in modprobe.d. How should I blacklist rtw89_8852be?

lwfinger, 2023-05-02:
The best way is in modules.d. I do not know how one would do it with grub.

gdan342, 2023-05-02:
i have made a blacklist.conf with `blacklist rtw89_8852be`

here is updated lsmod
`rtw_8852be             16384  0
rtw_8852b             360448  1 rtw_8852be
rtw89pci               86016  1 rtw_8852be
rtw89core             626688  2 rtw89pci,rtw_8852b
mac80211             1490944  2 rtw89pci,rtw89core
cfg80211             1277952  4 rtw89core,mac80211,88XXau,rtw_8852b
`

lwfinger, 2023-05-03:
That is the right list. Does it work? If not, there is something funny about you user-space wireless stuff. Are you using NetworkManager or something else?

One thing to try:
Run the command 'dmesg -t | grep wlan'. You will see a line ending in something line "wlp4s0: renamed from wlan0". Use the first name (without the colon) in the following command:

sudo iw dev <name> scan | egrep "signal|SSID"

Post the output.

gdan342, 2023-05-14:
No luck. Updated today to Kernel 6.3.1 and thought something it would work if that kernel does include this driver. Still no luck. going to install OS again from scratch, if still dosent work will try installing this driver again.

gdan342, 2023-05-14:
Fresh install of Endeavor OS with up to date kernel. no wireless. I blacklisted rtw89_8852be in modprobe.d,  then installed rtw89 using the arch method. Still no wireless. When doing a 'dmesg -t | grep wlan' nothing returns, and nothing shows with ifconfig. Will comment here again if I figure anything out so others can find it. For now, totally lost. Using an external adapter is okay, but is a pain

lwfinger, 2023-05-14:
What does the output of 'sudo dmesg | less' tell you? Are you missing firmware?

gdan342, 2023-05-14:
Yes, the bottom of this page details how to install firmware for this specific chipset. Tried, still no wireless. 

https://easylinuxtipsproject.blogspot.com/p/realtek.html

lwfinger, 2023-05-14:
Please post the output of 'lspci -nn'.

gdan342, 2023-05-14:
`00:00.0 Host bridge [0600]: Intel Corporation Gemini Lake Host Bridge [8086:31f0] (rev 06)
00:00.1 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor Dynamic Platform and Thermal Framework Processor Participant [8086:318c] (rev 06)
00:02.0 VGA compatible controller [0300]: Intel Corporation GeminiLake [UHD Graphics 600] [8086:3185] (rev 06)
00:0e.0 Audio device [0403]: Intel Corporation Celeron/Pentium Silver Processor High Definition Audio [8086:3198] (rev 06)
00:0f.0 Communication controller [0780]: Intel Corporation Celeron/Pentium Silver Processor Trusted Execution Engine Interface [8086:319a] (rev 06)
00:12.0 SATA controller [0106]: Intel Corporation Celeron/Pentium Silver Processor SATA Controller [8086:31e3] (rev 06)
00:13.0 PCI bridge [0604]: Intel Corporation Gemini Lake PCI Express Root Port [8086:31da] (rev f6)
00:14.0 PCI bridge [0604]: Intel Corporation Gemini Lake PCI Express Root Port [8086:31d7] (rev f6)
00:15.0 USB controller [0c03]: Intel Corporation Celeron/Pentium Silver Processor USB 3.0 xHCI Controller [8086:31a8] (rev 06)
00:16.0 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor I2C 0 [8086:31ac] (rev 06)
00:16.1 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor I2C 1 [8086:31ae] (rev 06)
00:16.2 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor I2C 2 [8086:31b0] (rev 06)
00:16.3 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor I2C 3 [8086:31b2] (rev 06)
00:17.0 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor I2C 4 [8086:31b4] (rev 06)
00:17.1 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor I2C 5 [8086:31b6] (rev 06)
00:17.2 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor I2C 6 [8086:31b8] (rev 06)
00:17.3 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor I2C 7 [8086:31ba] (rev 06)
00:18.0 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor Serial IO UART Host Controller [8086:31bc] (rev 06)
00:18.1 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor Serial IO UART Host Controller [8086:31be] (rev 06)
00:18.2 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor Serial IO UART Host Controller [8086:31c0] (rev 06)
00:18.3 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor Serial IO UART Host Controller [8086:31ee] (rev 06)
00:19.0 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor Serial IO SPI Host Controller [8086:31c2] (rev 06)
00:19.1 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor Serial IO SPI Host Controller [8086:31c4] (rev 06)
00:19.2 Signal processing controller [1180]: Intel Corporation Celeron/Pentium Silver Processor Serial IO SPI Host Controller [8086:31c6] (rev 06)
00:1c.0 SD Host controller [0805]: Intel Corporation Celeron/Pentium Silver Processor SDA Standard Compliant SD Host Controller [8086:31cc] (rev 06)
00:1e.0 SD Host controller [0805]: Intel Corporation Device [8086:31d0] (rev 06)
00:1f.0 ISA bridge [0601]: Intel Corporation Celeron/Pentium Silver Processor LPC Controller [8086:31e8] (rev 06)
00:1f.1 SMBus [0c05]: Intel Corporation Celeron/Pentium Silver Processor Gaussian Mixture Model [8086:31d4] (rev 06)
01:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 15)
02:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. Device [10ec:b852]
`

lwfinger, 2023-05-14:
OK, you do have an RTW8852BE device. Run the command 'sudo dmesg > dmesg.txt' and attach dmesg.txt to this issue.

gdan342, 2023-05-14:
`[    0.000000] microcode: updated early: 0xc -> 0x22, date = 2022-09-20
[    0.000000] Linux version 6.3.1-arch2-1 (linux@archlinux) (gcc (GCC) 13.1.1 20230429, GNU ld (GNU Binutils) 2.40.0) #1 SMP PREEMPT_DYNAMIC Wed, 10 May 2023 08:54:47 +0000
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-linux root=UUID=c9068df8-c7c8-4f94-acbd-a9d4c28706ac rw nowatchdog nvme_load=YES loglevel=3
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[3]:  576, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  640, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1b, context size is 704 bytes, using 'compacted' format.
[    0.000000] signal: max sigframe size: 2032
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000003efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000003f000-0x000000000003ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000040000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000010000000-0x0000000012150fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000012151000-0x0000000076476fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000076477000-0x0000000079d43fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000079d44000-0x0000000079d67fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x0000000079d68000-0x0000000079dd1fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000079dd2000-0x000000007a139fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007a13a000-0x000000007a1dffff] type 20
[    0.000000] BIOS-e820: [mem 0x000000007a1e0000-0x000000007aa58fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007aa59000-0x000000007aa59fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007aa5a000-0x000000007aa5bfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007aa5c000-0x000000007ab14fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ab15000-0x000000007abc4fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007abc5000-0x000000007affffff] usable
[    0.000000] BIOS-e820: [mem 0x000000007b000000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d0000000-0x00000000d0ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d3709000-0x00000000d3709fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe042000-0x00000000fe044fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe900000-0x00000000fe902fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff800000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000027fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.7 by TECLAST
[    0.000000] efi: ACPI 2.0=0x79d4b000 ACPI=0x79d4b000 TPMFinalLog=0x79da0000 SMBIOS=0x79ff2000 SMBIOS 3.0=0x79ff1000 MEMATTR=0x73293418 ESRT=0x73b7ba18 
[    0.000000] efi: Remove mem52: MMIO range=[0xd0000000-0xd0ffffff] (16MB) from e820 map
[    0.000000] e820: remove [mem 0xd0000000-0xd0ffffff] reserved
[    0.000000] efi: Not removing mem53: MMIO range=[0xd3709000-0xd3709fff] (4KB) from e820 map
[    0.000000] efi: Remove mem54: MMIO range=[0xe0000000-0xefffffff] (256MB) from e820 map
[    0.000000] e820: remove [mem 0xe0000000-0xefffffff] reserved
[    0.000000] efi: Not removing mem55: MMIO range=[0xfe042000-0xfe042fff] (4KB) from e820 map
[    0.000000] efi: Not removing mem56: MMIO range=[0xfe043000-0xfe043fff] (4KB) from e820 map
[    0.000000] efi: Not removing mem57: MMIO range=[0xfe044000-0xfe044fff] (4KB) from e820 map
[    0.000000] efi: Not removing mem58: MMIO range=[0xfe900000-0xfe902fff] (12KB) from e820 map
[    0.000000] efi: Not removing mem59: MMIO range=[0xfec00000-0xfec00fff] (4KB) from e820 map
[    0.000000] efi: Not removing mem60: MMIO range=[0xfed01000-0xfed01fff] (4KB) from e820 map
[    0.000000] efi: Not removing mem61: MMIO range=[0xfee00000-0xfee00fff] (4KB) from e820 map
[    0.000000] efi: Remove mem62: MMIO range=[0xff800000-0xffffffff] (8MB) from e820 map
[    0.000000] e820: remove [mem 0xff800000-0xffffffff] reserved
[    0.000000] SMBIOS 3.2.0 present.
[    0.000000] DMI: Default string To be filled by O.E.M./To be filled by O.E.M., BIOS 1.18 02/24/2023
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 1996.800 MHz processor
[    0.000030] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000036] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000058] last_pfn = 0x280000 max_arch_pfn = 0x400000000
[    0.000068] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000154] last_pfn = 0x7b000 max_arch_pfn = 0x400000000
[    0.003333] esrt: Reserving ESRT space from 0x0000000073b7ba18 to 0x0000000073b7ba50.
[... cut ...]
[ 7701.615262] usb 1-1: USB disconnect, device number 2
[ 7701.819973] usb 1-4: USB disconnect, device number 4
[ 7710.705604] i915 0000:00:02.0: [drm] *ERROR* GPIO index 3 request failed (-2)
[ 7710.711074] i915 0000:00:02.0: [drm] *ERROR* GPIO index 6 request failed (-2)
[ 7710.731536] i915 0000:00:02.0: [drm] *ERROR* GPIO index 6 request failed (-2)
[ 7710.761969] i915 0000:00:02.0: [drm] *ERROR* GPIO index 6 request failed (-2)
[ 7711.037456] i915 0000:00:02.0: [drm] *ERROR* GPIO index 4 request failed (-2)
`

gdan342, 2023-05-14:
`[    0.000000] microcode: updated early: 0xc -> 0x22, date = 2022-09-20
[    0.000000] Linux version 6.3.1-arch2-1 (linux@archlinux) (gcc (GCC) 13.1.1 20230429, GNU ld (GNU Binutils) 2.40.0) #1 SMP PREEMPT_DYNAMIC Wed, 10 May 2023 08:54:47 +0000
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-linux root=UUID=c9068df8-c7c8-4f94-acbd-a9d4c28706ac rw nowatchdog nvme_load=YES loglevel=3
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[3]:  576, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  640, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1b, context size is 704 bytes, using 'compacted' format.
[    0.000000] signal: max sigframe size: 2032
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000003efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000003f000-0x000000000003ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000040000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000010000000-0x0000000012150fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000012151000-0x0000000076476fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000076477000-0x0000000079d43fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000079d44000-0x0000000079d67fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x0000000079d68000-0x0000000079dd1fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000079dd2000-0x000000007a139fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007a13a000-0x000000007a1dffff] type 20
[    0.000000] BIOS-e820: [mem 0x000000007a1e0000-0x000000007aa58fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007aa59000-0x000000007aa59fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007aa5a000-0x000000007aa5bfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007aa5c000-0x000000007ab14fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ab15000-0x000000007abc4fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007abc5000-0x000000007affffff] usable
[    0.000000] BIOS-e820: [mem 0x000000007b000000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d0000000-0x00000000d0ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d3709000-0x00000000d3709fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe042000-0x00000000fe044fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe900000-0x00000000fe902fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff800000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000027fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.7 by TECLAST
[    0.000000] efi: ACPI 2.0=0x79d4b000 ACPI=0x79d4b000 TPMFinalLog=0x79da0000 SMBIOS=0x79ff2000 SMBIOS 3.0=0x79ff1000 MEMATTR=0x73293418 ESRT=0x73b7ba18 
[    0.000000] efi: Remove mem52: MMIO range=[0xd0000000-0xd0ffffff] (16MB) from e820 map
[    0.000000] e820: remove [mem 0xd0000000-0xd0ffffff] reserved
[    0.000000] efi: Not removing mem53: MMIO range=[0xd3709000-0xd3709fff] (4KB) from e820 map
[    0.000000] efi: Remove mem54: MMIO range=[0xe0000000-0xefffffff] (256MB) from e820 map
[    0.000000] e820: remove [mem 0xe0000000-0xefffffff] reserved
[    0.000000] efi: Not removing mem55: MMIO range=[0xfe042000-0xfe042fff] (4KB) from e820 map
[    0.000000] efi: Not removing mem56: MMIO range=[0xfe043000-0xfe043fff] (4KB) from e820 map
[    0.000000] efi: Not removing mem57: MMIO range=[0xfe044000-0xfe044fff] (4KB) from e820 map
[    0.000000] efi: Not removing mem58: MMIO range=[0xfe900000-0xfe902fff] (12KB) from e820 map
[    0.000000] efi: Not removing mem59: MMIO range=[0xfec00000-0xfec00fff] (4KB) from e820 map
[    0.000000] efi: Not removing mem60: MMIO range=[0xfed01000-0xfed01fff] (4KB) from e820 map
[    0.000000] efi: Not removing mem61: MMIO range=[0xfee00000-0xfee00fff] (4KB) from e820 map
[    0.000000] efi: Remove mem62: MMIO range=[0xff800000-0xffffffff] (8MB) from e820 map
[    0.000000] e820: remove [mem 0xff800000-0xffffffff] reserved
[    0.000000] SMBIOS 3.2.0 present.
[    0.000000] DMI: Default string To be filled by O.E.M./To be filled by O.E.M., BIOS 1.18 02/24/2023
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 1996.800 MHz processor
[    0.000030] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000036] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000058] last_pfn = 0x280000 max_arch_pfn = 0x400000000
[    0.000068] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000154] last_pfn = 0x7b000 max_arch_pfn = 0x400000000
[    0.003333] esrt: Reserving ESRT space from 0x0000000073b7ba18 to 0x0000000073b7ba50.
[... cut ...]
[ 7701.615262] usb 1-1: USB disconnect, device number 2
[ 7701.819973] usb 1-4: USB disconnect, device number 4
[ 7710.705604] i915 0000:00:02.0: [drm] *ERROR* GPIO index 3 request failed (-2)
[ 7710.711074] i915 0000:00:02.0: [drm] *ERROR* GPIO index 6 request failed (-2)
[ 7710.731536] i915 0000:00:02.0: [drm] *ERROR* GPIO index 6 request failed (-2)
[ 7710.761969] i915 0000:00:02.0: [drm] *ERROR* GPIO index 6 request failed (-2)
[ 7711.037456] i915 0000:00:02.0: [drm] *ERROR* GPIO index 4 request failed (-2)
`

lwfinger, 2023-05-14:
Why cannot you follow directions. It is one thing to try to look past the missing line breaks for something short like the lspci output, but much more effort for the entire dmesg output. If you want help, USE the cammand I gave you. It will produce a file. ATTACH the file.

gdan342, 2023-05-14:
[dmesg.txt](https://github.com/lwfinger/rtw89/files/11472143/dmesg.txt)

gdan342, 2023-05-14:
[dmesg.txt](https://github.com/lwfinger/rtw89/files/11472261/dmesg.txt)

lwfinger, 2023-05-14:
Thank you.

Your problem is here:
[    6.306763] rtw89_8852be 0000:02:00.0: loaded firmware rtw89/rtw8852b_fw-1.bin
[    6.307637] rtw89_8852be 0000:02:00.0: enabling device (0000 -> 0003)
[    6.350288] Bluetooth: hci0: RTL: fw version 0xdfb791cb
[    6.369101] rtw89_8852be 0000:02:00.0: xtal si not ready(R): offset=41
[    6.369115] rtw89_8852be 0000:02:00.0: no suitable firmware found
[    6.369117] rtw89_8852be 0000:02:00.0: failed to recognize firmware
[    6.369119] rtw89_8852be 0000:02:00.0: failed to setup chip information
[    6.372705] rtw89_8852be: probe of 0000:02:00.0 failed with error -2

What you should see is:
[    6.488336] rtw89_8852be 0000:02:00.0: loaded firmware rtw89/rtw8852b_fw-1.bin
[    6.506786] rtw89_8852be 0000:02:00.0: Firmware version 0.29.29.1, cmd version 0, type 5
[    6.506794] rtw89_8852be 0000:02:00.0: Firmware version 0.29.29.1, cmd version 0, type 3

On my system, the command 'md5sum /lib/firmware/rtw89/*' shows (in part):

97cc27ddbaacfcfd9f4b4ad425d114d3  /lib/firmware/rtw89/rtw8852b_fw-1.bin
66764a3447732e2b5d563a2d7c69d39f  /lib/firmware/rtw89/rtw8852b_fw-1.bin.xz
a3317e9646c8e669c7c2d1fc19a6109d  /lib/firmware/rtw89/rtw8852b_fw.bin
6c91e2112e09b7313cee17f884fd4f70  /lib/firmware/rtw89/rtw8852b_fw.bin.xz

The files that end in .xz are compressed by my distro (openSUSE). The others come from the linux-firmware repo directly.

gdan342, 2023-05-14:
Okay, great. What is the solution? How to I get it to work properly?

lwfinger, 2023-05-14:
Do your firmware files match mine?

lwfinger, 2023-05-14:
I have no idea what the "Arch" method of installing the driver is. Do the following:
git clone https://github.com/lwfinger/rtw89.git
cd rtw89
make -j4
sudo make install

Then reboot. After the reboot, do the 'sudo dmesg > dmesg.txt' and attach dmesg.txt.

gdan342, 2023-05-14:
I did it this way. From your guide. 

For Arch: After installing the necessary kernel headers and base-devel,

git clone https://aur.archlinux.org/rtw89-dkms-git.git
cd rtw89-dkms-git
makepkg -sri

gdan342, 2023-05-14:
Awesome. I download all of them? or just the rtw8852b ones? And then i move them to /lib/firmware/rtw89 correct?

gdan342, 2023-05-14:
downloaded and replaced them and rebooted. Here is output of md5sum /lib/firmware/rtw89/*

`97cc27ddbaacfcfd9f4b4ad425d114d3  /lib/firmware/rtw89/rtw8852b_fw-1.bin
66764a3447732e2b5d563a2d7c69d39f  /lib/firmware/rtw89/rtw8852b_fw-1.bin.xz
a3317e9646c8e669c7c2d1fc19a6109d  /lib/firmware/rtw89/rtw8852b_fw.bin
6c91e2112e09b7313cee17f884fd4f70  /lib/firmware/rtw89/rtw8852b_fw.bin.xz
`
Still no wireless. I ran dmesg ag

lwfinger, 2023-05-15:
I am not quite sure why you closed this issue. Obviously, your problem was not with the firmware.

One thing I do not understand is the "rtw89_8852be 0000:02:00.0: xtal si not ready(R): offset=41" message. It arises because the code is unable to read one of the registers. I sent a question to my contact at Realtek, but it will likely take him a while.

One thing we can do is to try a couple of module parameters that control the actions on the PCIe bus. These parameters have helped users of newer HP and Lenovo laptops overcome some of the problems with their faulty BIOSs. It is possible that this unit has the same kind of problem.

Run the command 'sudo nano /usr/lib/modprobe.d/70-rtw8852be.conf'
In the open window, enter the following 2 lines:
options rtw89pci disable_clkreq=y  disable_aspm_l1=y  disable_aspm_l1ss=y
options rtw89i_pci disable_clkreq=y  disable_aspm_l1=y  disable_aspm_l1ss=y

Once you have created this file, run the following commands:
sudo modprobe -rv rtw_8852be
sudo modprobe -v rtw_8852be

If you did it right, the line that shows the insmod for rtw89pci should end with the parameters in the two lines in the files.

Make a new copy of dmesg.txt and attach it here.

gdan342, 2023-05-15:
I did not close it on purpose I'm not sure how it happened. Not easy working on that tiny screen ha must've accidently touched something. I will try this when I have some time again to mess with it. Re-opening, sorry about that

Subsentient, 2023-06-14:
@lwfinger Hi, I've got one of these little guys as well. It seems that I can't get any firmware version, old or new, to load in the rtw89_8852be driver. I tried with this repo, that didn't work, so I wondered if the in-tree module in 6.3 would work. Nope, same issue, but I can see they're very similar codebases. This is what I get in both versions, with any version of rtw8852b_fw.bin from any of many git revisions in the linux-firmware tree I've tried. I've tried the modprobe.d method of passing those args to the rtw89pci drivers, and I did also see the `rtw89_8852be 0000:02:00.0: xtal si not ready(R): offset=41` log in dmesg when using your module, I think the mainline version just eats the bug. dmesg attached.
[dmesg.txt](https://github.com/lwfinger/rtw89/files/11741587/dmesg.txt)

lwfinger, 2023-06-14:
This repo does not have any firmware in it. Usually, you would get that from your distro. Please run the following commend and post the results:

md5sum /lib/firmware/rtw89/*

On my system, for the RTW8852BE, I get the following:
97cc27ddbaacfcfd9f4b4ad425d114d3  /lib/firmware/rtw89/rtw8852b_fw-1.bin
66764a3447732e2b5d563a2d7c69d39f  /lib/firmware/rtw89/rtw8852b_fw-1.bin.xz
a3317e9646c8e669c7c2d1fc19a6109d  /lib/firmware/rtw89/rtw8852b_fw.bin
6c91e2112e09b7313cee17f884fd4f70  /lib/firmware/rtw89/rtw8852b_fw.bin.xz

The .xz versions come from my distro, and the non-xz varieties come from the linux-firmware repo. If your files differ, you can download them at https://lwfinger.com/download/, and then use 'sudo cp' to copy them to /usr/lib/firmware/rtw89/.

Subsentient, 2023-06-15:
> This repo does not have any firmware in it. Usually, you would get that from your distro. Please run the following commend and post the results:
> 
> md5sum /lib/firmware/rtw89/*

Yes, I know the rtw89 driver (yours and mainline) contains no firmware. I have only one firmware file in that directory at the moment, because I deleted all the others to rule out interference, I can reinstall the linux-firmware package if for whatever reason I need them. I think this is a similar issue with the hardware as gdan342, because we have the same device with the same wireless chip.
`
a3317e9646c8e669c7c2d1fc19a6109d  /lib/firmware/rtw89/rtw8852b_fw.bin`

lwfinger, 2023-06-15:
It could be that there is something funky about this device; however, my budget does not allow speinding $199 for a test device.

Who knows what the Windows driver does?

I created a new branch (test) with a number of printouts. Please check it out and build with

git checkout origin/test -b test
make
sudo make install

Then reboot. After the firmware load fails, run the command 'sudo dmesg -t > dmesg.txt' and attach dmesg.txt to this issue so that I can see the results of my debugging messages.

Thanks.

HeXis-YS, 2023-10-30:
Same problem here.
Actually, the mini PC uses a wireless module (6252C-PUB) made by Fn-Link. The RTL8852BE chips used in them are likely customized or modified.
And here is the source code of the firmware for Linux provided by the manufacturer. I hope that helps.
[rtl8852BE_WiFi_linux_v1.15.10.0.5-0-gfa2af07cf.20220503_PC.tar.gz](https://github.com/lwfinger/rtw89/files/13210660/rtl8852BE_WiFi_linux_v1.15.10.0.5-0-gfa2af07cf.20220503_PC.tar.gz)

lwfinger, 2023-10-31:
What am I supposed to do with that? You can untar that and build it yourself.

Tomblarom, 2023-11-30:
I also bought 6 of them. They were supposed to support Linux out-of-the-box. Wifi only works under Windows 11. It's a huge bummer.

@HeXis-YS and @gdan342 the devices come without serial number written to the BIOS and no auto-start function. I posted about this and several other issues here: https://4pda.to/forum/index.php?showtopic=1072665

Tomblarom, 2023-11-30:
@lwfinger I read through your answers and highly appreciate your support so far! I'm using IGEL OS12 on a [Higole X133](https://www.cnx-software.com/2023/04/07/higole-pc-stick-j4125-wifi-6-review-specs-unboxing-teardown-first-boot/) which is based on Ubuntu 20.04 Focal and has Kernel 6.1.42. I reproduced all steps and was not able to make it work. Additionally to that I'm working with IGEL and have them figure this out, since we have enterprise support.

My [dmesg.log](https://github.com/lwfinger/rtw89/files/13514522/dmesg.log) is exactly the same as with gdan342. 

I tried placing the `70-rtw8852be.conf` in `/usr/lib/modprobe.d` and `/etc/modprobe.d`. I used [your latest .bin](https://lwfinger.com/download/) files:
```
root@TVS006:/lib# md5sum /lib/firmware/rtw89/*
5f2b81e0f3f3e72d0b4564b44e075701  /lib/firmware/rtw89/rtw8851b_fw.bin
6edd77ccced6eda33850d302853fe665  /lib/firmware/rtw89/rtw8852a_fw.bin
4df8d6e2459b9409837f64a4be879288  /lib/firmware/rtw89/rtw8852b_fw-1.bin
a3317e9646c8e669c7c2d1fc19a6109d  /lib/firmware/rtw89/rtw8852b_fw.bin
ceed5cd7b2066d7a8b99e73e9056d3c1  /lib/firmware/rtw89/rtw8852c_fw.bin
```

Let me know if you are willing to receive a test-device. I can surely supply one!

Tomblarom, 2023-11-30:
> And here is the source code of the firmware for Linux provided by the manufacturer.

They send me a similar one: [rtl8852BE_WiFi_linux_v1.19.7.0-18-gee24b6aac.20230614_PC.tar.gz](https://github.com/lwfinger/rtw89/files/13515636/rtl8852BE_WiFi_linux_v1.19.7.0-18-gee24b6aac.20230614_PC.tar.gz)

lwfinger, 2023-12-01:
@Tomblarom - That is not firmware. It is the vendor driver. If you build and install it, does it work?

If that does not compile, you might also checkout https://github.com/rtw8852be.git. That is from the same source, but will build on kernels through 6.7. See what you geet with it.

Usually, I woulld be happy to receive a test device, but I am not sure how I will be feeling the next 4 weeks as I will be undergoing a new medical procedure.

apraviah, 2024-05-04:
> who has higole 1 pro please send me the touch screen driver

https://higole-my.sharepoint.com/personal/support_higole_onmicrosoft_com/_layouts/15/onedrive.aspx?id=%2Fpersonal%2Fsupport%5Fhigole%5Fonmicrosoft%5Fcom%2FDocuments%2FGole1%20Pro%20Touch%20Driver%2FSigned%5FGoodixdriver4%2E7%2E2%2E43156%5F20220902%2E7z&parent=%2Fpersonal%2Fsupport%5Fhigole%5Fonmicrosoft%5Fcom%2FDocuments%2FGole1%20Pro%20Touch%20Driver&ga=1

christal87, 2025-02-05:
I'm using Higole F10 mini PCs to run a single fullscreen app I'm developing. That has a PCIe rtl8821ce soldered on. My wireless only works 10/10 after reboot, but only 3/10 if I boot the PC cold. Other times the whole device doesn't even show up in the lspci output. I've spent a week or so trying to solve the issue without success. No kind of driver (debian shipped/lwfinger/tomaspinho) would make it more stable. I gave up, because I have a tight deadline and I have to focus on my app. My very dirty workaround was to create a script that looks for a wifi capable interface on boot using nmcli and restart if none found. Also bought a working USB wifi dongle to be sure if this solution turns out to be unstable. If it's not a power management issue, one other solution might be to buy an Intel AX200 M.2 LGA type 1216 form factor wifi module and replace it (soldering). I suspect the issue is around the broken BIOS/EFI, broken power savings, PNP<-->Wifi PCIe device resource conflict remap or a combination of these. Solving resource overlaps are out of my league. The BIOS is so broken or unfinished that it has a lot of (demo/unlocked?) options and most of them don't work. It's an Apollo Lake platform and it shouldn't have Intel's CNVi wifi config option in the BIOS yet. It's a mess. I've noticed I can't even disable the d3cold support correctly. Changing some other options cause the thing to not boot, needing an RTC/BIOS config reset. Unfortunately these no name chinese stuff with weak or non-existent tech support are basically e-waste so be aware.
<details>
  <summary>Filtered PCIe dmesg output suggesting a resource conflict</summary>

```
[    1.121915] pci 0000:02:00.0: [10ec:c821] type 00 class 0x028000
[    1.121941] pci 0000:02:00.0: reg 0x10: [io  0x0000-0x00ff]
[    1.121970] pci 0000:02:00.0: reg 0x18: [mem 0x00000000-0x0000ffff 64bit]
[    1.122006] pci 0000:02:00.0: Upstream bridge's Max Payload Size set to 128 (was 256, max 256)
[    1.122017] pci 0000:02:00.0: Max Payload Size set to 128 (was 128, max 128)
[    1.122125] pci 0000:02:00.0: supports D1 D2
[    1.122127] pci 0000:02:00.0: PME# supported from D0 D1 D2 D3hot D3cold
[    1.189239] pnp 00:01: disabling [io  0x002e-0x002f] because it overlaps 0000:02:00.0 BAR 0 [io  0x0000-0x00ff]
[    1.189244] pnp 00:01: disabling [io  0x004e-0x004f] because it overlaps 0000:02:00.0 BAR 0 [io  0x0000-0x00ff]
[    1.189248] pnp 00:01: disabling [io  0x0061] because it overlaps 0000:02:00.0 BAR 0 [io  0x0000-0x00ff]
[    1.189251] pnp 00:01: disabling [io  0x0063] because it overlaps 0000:02:00.0 BAR 0 [io  0x0000-0x00ff]
[    1.189254] pnp 00:01: disabling [io  0x0065] because it overlaps 0000:02:00.0 BAR 0 [io  0x0000-0x00ff]
[    1.189257] pnp 00:01: disabling [io  0x0067] because it overlaps 0000:02:00.0 BAR 0 [io  0x0000-0x00ff]
[    1.189260] pnp 00:01: disabling [io  0x0070] because it overlaps 0000:02:00.0 BAR 0 [io  0x0000-0x00ff]
[    1.189263] pnp 00:01: disabling [io  0x0080-0x008f] because it overlaps 0000:02:00.0 BAR 0 [io  0x0000-0x00ff]
[    1.189266] pnp 00:01: disabling [io  0x0092] because it overlaps 0000:02:00.0 BAR 0 [io  0x0000-0x00ff]
[    1.189269] pnp 00:01: disabling [io  0x00b2-0x00b3] because it overlaps 0000:02:00.0 BAR 0 [io  0x0000-0x00ff]
[    1.201823] pci 0000:02:00.0: BAR 2: assigned [mem 0x7c100000-0x7c10ffff 64bit]
[    1.201838] pci 0000:02:00.0: BAR 0: assigned [io  0x2000-0x20ff]
[    1.203578] pci 0000:02:00.0: Adding to iommu group 17
[    7.999663] rtw_8821ce 0000:02:00.0: enabling device (0000 -> 0003)
[    8.007184] rtw_8821ce 0000:02:00.0: firmware: direct-loading firmware rtw88/rtw8821c_fw.bin
[    8.007201] rtw_8821ce 0000:02:00.0: Firmware version 24.11.0, H2C version 12
[    8.007220] rtw_8821ce 0000:02:00.0: mac power on failed
[    8.007222] rtw_8821ce 0000:02:00.0: failed to power on mac
[    8.007224] rtw_8821ce 0000:02:00.0: failed to setup chip efuse info
[    8.007226] rtw_8821ce 0000:02:00.0: failed to setup chip information
[    8.038066] rtw_8821ce: probe of 0000:02:00.0 failed with error -114
```

</details>