lwfinger/rtw89 #244: no .bin file created after make

tal-grossman, 2023-05-15:
## machine:
**[ubuntu 20.04, Lenovo, legion, linux-headers-5.15.0-71-generic]**

steps:
```
git clone https://github.com/lwfinger/rtw89.git
cd rtw89
make
sudo make install
sudo mkdir /usr/lib/firmware/rtw89
sudo cp  rtw8852a_fw.bin  /usr/lib/firmware/rtw89/
sudo modprobe rtw89pci
```

when:

```
> tal@tal-Legion-5-Pro-16ACH6H:~/rtw89$ sudo cp  rtw8852a_fw.bin  /usr/lib/firmware/rtw89/
> cp: cannot stat 'rtw8852a_fw.bin': No such file or directory
```

make log:

> ltal@tal-Legion-5-Pro-16ACH6H:~/rtw89$ make
make -C /lib/modules/5.15.0-71-generic/build M=/home/tal/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-71-generic'
  CC [M]  /home/tal/rtw89/core.o
  CC [M]  /home/tal/rtw89/chan.o
  CC [M]  /home/tal/rtw89/mac80211.o
  CC [M]  /home/tal/rtw89/mac.o
  CC [M]  /home/tal/rtw89/phy.o
  CC [M]  /home/tal/rtw89/fw.o
  CC [M]  /home/tal/rtw89/cam.o
  CC [M]  /home/tal/rtw89/efuse.o
  CC [M]  /home/tal/rtw89/regd.o
  CC [M]  /home/tal/rtw89/sar.o
  CC [M]  /home/tal/rtw89/coex.o
  CC [M]  /home/tal/rtw89/ps.o
  CC [M]  /home/tal/rtw89/debug.o
  CC [M]  /home/tal/rtw89/ser.o
  CC [M]  /home/tal/rtw89/wow.o
  LD [M]  /home/tal/rtw89/rtw89core.o
  CC [M]  /home/tal/rtw89/rtw8852a.o
  CC [M]  /home/tal/rtw89/rtw8852a_table.o
  CC [M]  /home/tal/rtw89/rtw8852a_rfk.o
  CC [M]  /home/tal/rtw89/rtw8852a_rfk_table.o
  LD [M]  /home/tal/rtw89/rtw_8852a.o
  CC [M]  /home/tal/rtw89/rtw8852ae.o
  LD [M]  /home/tal/rtw89/rtw_8852ae.o
  CC [M]  /home/tal/rtw89/rtw8852b.o
  CC [M]  /home/tal/rtw89/rtw8852b_table.o
  CC [M]  /home/tal/rtw89/rtw8852b_rfk.o
  CC [M]  /home/tal/rtw89/rtw8852b_rfk_table.o
  LD [M]  /home/tal/rtw89/rtw_8852b.o
  CC [M]  /home/tal/rtw89/rtw8852be.o
  LD [M]  /home/tal/rtw89/rtw_8852be.o
  CC [M]  /home/tal/rtw89/rtw8852c.o
  CC [M]  /home/tal/rtw89/rtw8852c_table.o
  CC [M]  /home/tal/rtw89/rtw8852c_rfk.o
  CC [M]  /home/tal/rtw89/rtw8852c_rfk_table.o
  LD [M]  /home/tal/rtw89/rtw_8852c.o
  CC [M]  /home/tal/rtw89/rtw8852ce.o
  LD [M]  /home/tal/rtw89/rtw_8852ce.o
  CC [M]  /home/tal/rtw89/pci.o
  LD [M]  /home/tal/rtw89/rtw89pci.o
  MODPOST /home/tal/rtw89/Module.symvers
  CC [M]  /home/tal/rtw89/rtw89core.mod.o
  LD [M]  /home/tal/rtw89/rtw89core.ko
  BTF [M] /home/tal/rtw89/rtw89core.ko
Skipping BTF generation for /home/tal/rtw89/rtw89core.ko due to unavailability of vmlinux
  CC [M]  /home/tal/rtw89/rtw89pci.mod.o
  LD [M]  /home/tal/rtw89/rtw89pci.ko
  BTF [M] /home/tal/rtw89/rtw89pci.ko
Skipping BTF generation for /home/tal/rtw89/rtw89pci.ko due to unavailability of vmlinux
  CC [M]  /home/tal/rtw89/rtw_8852a.mod.o
  LD [M]  /home/tal/rtw89/rtw_8852a.ko
  BTF [M] /home/tal/rtw89/rtw_8852a.ko
Skipping BTF generation for /home/tal/rtw89/rtw_8852a.ko due to unavailability of vmlinux
  CC [M]  /home/tal/rtw89/rtw_8852ae.mod.o
  LD [M]  /home/tal/rtw89/rtw_8852ae.ko
  BTF [M] /home/tal/rtw89/rtw_8852ae.ko
Skipping BTF generation for /home/tal/rtw89/rtw_8852ae.ko due to unavailability of vmlinux
  CC [M]  /home/tal/rtw89/rtw_8852b.mod.o
  LD [M]  /home/tal/rtw89/rtw_8852b.ko
  BTF [M] /home/tal/rtw89/rtw_8852b.ko
Skipping BTF generation for /home/tal/rtw89/rtw_8852b.ko due to unavailability of vmlinux
  CC [M]  /home/tal/rtw89/rtw_8852be.mod.o
  LD [M]  /home/tal/rtw89/rtw_8852be.ko
  BTF [M] /home/tal/rtw89/rtw_8852be.ko
Skipping BTF generation for /home/tal/rtw89/rtw_8852be.ko due to unavailability of vmlinux
  CC [M]  /home/tal/rtw89/rtw_8852c.mod.o
  LD [M]  /home/tal/rtw89/rtw_8852c.ko
  BTF [M] /home/tal/rtw89/rtw_8852c.ko
Skipping BTF generation for /home/tal/rtw89/rtw_8852c.ko due to unavailability of vmlinux
  CC [M]  /home/tal/rtw89/rtw_8852ce.mod.o
  LD [M]  /home/tal/rtw89/rtw_8852ce.ko
  BTF [M] /home/tal/rtw89/rtw_8852ce.ko
Skipping BTF generation for /home/tal/rtw89/rtw_8852ce.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-71-generic

searched .bin file: `find . -name '*.bin'
`
when 

* tried pulling and make clean over and over no .bin file created to copy

lwfinger, 2023-05-15:
The make never creates a .bin file - only .ko files that are kernel modules.

Usually, .bin files are firmware. This code is not open source, and I have no idea on how to build it. It is provided by Realtek to the linix-firmware repo. Your distro should take it from there and provide a firmware package - mine does. If not, then you must get the firmware from some other source, and copy the files to /lib/firmware/rtw89/. One place is from https://www.lwfinger.com/download/.

tal-grossman, 2023-05-15:
I downloaded the `rtw8852a_fw.bin ` executed the` sudo modprobe rtw89pci` and still It does not recognize wifi adaptor, 
Any ideas?

lwfinger, 2023-05-15:
Please run the command 'lspci -nn > lspci.txt' and attach lspci.txt to this issue.

tal-grossman, 2023-05-15:
[lspci.txt](https://github.com/lwfinger/rtw89/files/11480740/lspci.txt)

attached here

lwfinger, 2023-05-15:
Thanks. The lsmod stuff was fine, but dmesg.txt shows 
[    6.013203] rtw89_8852ae 0000:04:00.0: Direct firmware load for rtw89/rtw8852a_fw.bin failed with error -2
[    6.013207] rtw89_8852ae 0000:04:00.0: failed to early request firmware: -2
[    6.013266] rtw89_8852ae 0000:04:00.0: enabling device (0000 -> 0003)
[    6.013284] rtw89_8852ae 0000:04:00.0: Direct firmware load for rtw89/rtw8852a_fw.bin failed with error -2
[    6.014389] rtw89_8852ae 0000:04:00.0: failed to wait firmware completion
[    6.014407] rtw89_8852ae 0000:04:00.0: failed to setup chip information
[    6.014756] rtw89_8852ae: probe of 0000:04:00.0 failed with error -22

Did you copy rtw8852a_fw.bin to /lib/firmware/rtw89/rtw8852a_fw.bin?
Look at 
ls -l /lib/firmware/rtw89/
 to see for sure.

tal-grossman, 2023-05-16:
Yes I did it. also, I think I had this file after ubutnu installation and didn't have wifi adapter so I moved the original one to backup directory.

Any thing else? Maybe downgrading the Linux headers kernel?

lwfinger, 2023-05-16:
Well, the previous dmesg says it is not there, or it cannot be read. Run the command

md5sum /lib/firmware/rtw89/* > md5sum.txt

and post md5sum.txt.

tal-grossman, 2023-05-16:
[md5sum.txt](https://github.com/lwfinger/rtw89/files/11484975/md5sum.txt)

lwfinger, 2023-05-16:
So it was an Ubuntu problem. Good to know.

closed 2023-05-16 by lwfinger