lwfinger/rtw89 #246: Building failed when using latest commit

nishikinocurtis, 2023-05-27:
I have not been updating my rtw89 for a while and I recently first tried to update it through AUR, it automatically fetched the latest commit. But my WiFi was removed then, and I was reported the dkms install exited code 10.

I did some searching and found the same issue as in #229, so I checked out bea0ed7 as is mentioned by @salimp2009. And it works. Later version than that not tested.

I am using kernel version 5.15.112-1-MANJARO.

lwfinger, 2023-05-28:
I just pushed a fix to allow builds on kernel 5.15

salimp2009, 2023-05-28:
> I just pushed a fix to allow builds on kernel 5.15

commit #9b69cd7 removed the wireless . I just opened another issue for that #248 . I am on on Kernel 5.15.112-1 (Manjaro)

closed 2023-10-02 by lwfinger