lwfinger/rtw89 #247: netif_napi_add() function in core.c seems to have fewer arguments given than required when called in line 2078

Captainkk75, 2023-05-28:
```
...
/home/kushik/rtw89/core.c: In function ‘rtw89_core_napi_init’:
/home/kushik/rtw89/core.c:2078:9: error: too few arguments to function ‘netif_napi_add’
 2078 |         netif_napi_add(&rtwdev->netdev, &rtwdev->napi,
      |         ^~~~~~~~~~~~~~
In file included from ./include/net/inet_sock.h:19,
                 from ./include/linux/udp.h:16,
                 from /home/kushik/rtw89/core.c:5:
./include/linux/netdevice.h:2567:1: note: declared here
 2567 | netif_napi_add(struct net_device *dev, struct napi_struct *napi,
      | ^~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:257: /home/kushik/rtw89/core.o] Error 1
make[1]: *** [Makefile:1850: /home/kushik/rtw89] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.19.0-42-generic'
make: *** [Makefile:97: all] Error 2
```

lwfinger, 2023-05-28:
I pushed a fix for this problem. Do a pull and make.

Captainkk75, 2023-05-29:
Thanks very much!

closed 2023-05-29 by Captainkk75