lwfinger/rtw89 #248: commit #9b69cd7 removed wireless on Kernel 5.15.112-1

salimp2009, 2023-05-28:
Just got the new update via package manager . I am on Kernel 5.15.112-1 (on Manjaro) and it removed the wireless. it did not give any error. reboot my comp and wireless is gone 
Salim

salimp2009, 2023-05-28:
I tried to install from the repo. When I do make I get this error
```
CC [M]  /usr/src/rtw89/rtw89/pci.o
  LD [M]  /usr/src/rtw89/rtw89/rtw89pci.o
  MODPOST /usr/src/rtw89/rtw89/Module.symvers
ERROR: modpost: "rtw8852b_chip_info" [/usr/src/rtw89/rtw89/rtw_8852be.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:133: /usr/src/rtw89/rtw89/Module.symvers] Error 1
make[2]: *** Deleting file '/usr/src/rtw89/rtw89/Module.symvers'
make[1]: *** [Makefile:1826: modules] Error 2
make: *** [Makefile:97: all] Error 2
```

salimp2009, 2023-05-28:
I also tried the previous commit and get this error  commit #0d763459  
```
make -C /lib/modules/5.15.112-1-MANJARO/build M=/usr/src/rtw89/rtw89 modules
  CC [M]  /usr/src/rtw89/rtw89/core.o
/usr/src/rtw89/rtw89/core.c: In function ‘rtw89_core_tx_update_ampdu_info’:
/usr/src/rtw89/rtw89/core.c:440:35: error: ‘struct ieee80211_sta’ has no member named ‘deflink’
  440 |                           4 << sta->deflink.ht_cap.ampdu_factor) - 1);
      |                                   ^~
/usr/src/rtw89/rtw89/core.c:443:39: error: ‘struct ieee80211_sta’ has no member named ‘deflink’
  443 |         desc_info->ampdu_density = sta->deflink.ht_cap.ampdu_density;
      |                                       ^~
/usr/src/rtw89/rtw89/core.c: In function ‘__rtw89_core_tx_check_he_qos_htc’:
/usr/src/rtw89/rtw89/core.c:654:25: error: ‘struct ieee80211_sta’ has no member named ‘deflink’
  654 |         if (!sta || !sta->deflink.he_cap.has_he)
      |                         ^~
/usr/src/rtw89/rtw89/core.c: In function ‘rtw89_core_get_data_rate’:
/usr/src/rtw89/rtw89/core.c:744:25: error: ‘struct ieee80211_sta’ has no member named ‘deflink’
  744 |         if (!sta || !sta->deflink.supp_rates[chan->band_type])
      |                         ^~
/usr/src/rtw89/rtw89/core.c:747:25: error: ‘struct ieee80211_sta’ has no member named ‘deflink’
  747 |         return __ffs(sta->deflink.supp_rates[chan->band_type]) + lowest_rate;
      |                         ^~
/usr/src/rtw89/rtw89/core.c: In function ‘rtw89_stats_trigger_frame’:
/usr/src/rtw89/rtw89/core.c:1501:54: error: invalid use of undefined type ‘struct ieee80211_trigger’
 1501 |         if (!ether_addr_equal(vif->bss_conf.bssid, tf->ta) ||
      |                                                      ^~
/usr/src/rtw89/rtw89/core.c:1506:32: error: invalid use of undefined type ‘struct ieee80211_trigger’
 1506 |         type = le64_get_bits(tf->common_info, IEEE80211_TRIGGER_TYPE_MASK);
      |                                ^~
/usr/src/rtw89/rtw89/core.c:1506:47: error: ‘IEEE80211_TRIGGER_TYPE_MASK’ undeclared (first use in this function); did you mean ‘IEEE80211_BSS_TYPE_MBSS’?
 1506 |         type = le64_get_bits(tf->common_info, IEEE80211_TRIGGER_TYPE_MASK);
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                               IEEE80211_BSS_TYPE_MBSS
/usr/src/rtw89/rtw89/core.c:1506:47: note: each undeclared identifier is reported only once for each function it appears in
/usr/src/rtw89/rtw89/core.c:1507:21: error: ‘IEEE80211_TRIGGER_TYPE_BASIC’ undeclared (first use in this function)
 1507 |         if (type != IEEE80211_TRIGGER_TYPE_BASIC)
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/src/rtw89/rtw89/core.c:1511:17: error: invalid use of undefined type ‘struct ieee80211_trigger’
 1511 |         pos = tf->variable;
      |                 ^~
/usr/src/rtw89/rtw89/core.c:1523:31: error: ‘struct ieee80211_vif’ has no member named ‘cfg’
 1523 |                 if (aid == vif->cfg.aid) {
      |                               ^~
/usr/src/rtw89/rtw89/core.c: In function ‘rtw89_vif_rx_stats_iter’:
/usr/src/rtw89/rtw89/core.c:1617:13: error: implicit declaration of function ‘ieee80211_is_trigger’; did you mean ‘ieee80211_is_frag’? [-Werror=implicit-function-declaration]
 1617 |         if (ieee80211_is_trigger(hdr->frame_control)) {
      |             ^~~~~~~~~~~~~~~~~~~~
      |             ieee80211_is_frag
/usr/src/rtw89/rtw89/core.c: In function ‘rtw89_core_napi_init’:
/usr/src/rtw89/rtw89/core.c:2053:9: error: too few arguments to function ‘netif_napi_add’
 2053 |         netif_napi_add(&rtwdev->netdev, &rtwdev->napi,
      |         ^~~~~~~~~~~~~~
In file included from ./include/net/inet_sock.h:19,
                 from ./include/linux/udp.h:16,
                 from /usr/src/rtw89/rtw89/core.c:5:
./include/linux/netdevice.h:2501:6: note: declared here
 2501 | void netif_napi_add(struct net_device *dev, struct napi_struct *napi,
      |      ^~~~~~~~~~~~~~
/usr/src/rtw89/rtw89/core.c: In function ‘rtw89_core_send_nullfunc’:
/usr/src/rtw89/rtw89/core.c:2416:56: error: ‘struct ieee80211_vif’ has no member named ‘cfg’
 2416 |         if (vif->type != NL80211_IFTYPE_STATION || !vif->cfg.assoc)
      |                                                        ^~
/usr/src/rtw89/rtw89/core.c:2426:15: error: too many arguments to function ‘ieee80211_nullfunc_get’
 2426 |         skb = ieee80211_nullfunc_get(rtwdev->hw, vif, -1, qos);
      |               ^~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/src/rtw89/rtw89/core.h:13,
                 from /usr/src/rtw89/rtw89/cam.h:8,
                 from /usr/src/rtw89/rtw89/core.c:7:
./include/net/mac80211.h:5090:17: note: declared here
 5090 | struct sk_buff *ieee80211_nullfunc_get(struct ieee80211_hw *hw,
      |                 ^~~~~~~~~~~~~~~~~~~~~~
/usr/src/rtw89/rtw89/core.c: In function ‘rtw89_init_he_cap’:
/usr/src/rtw89/rtw89/core.c:3303:50: error: ‘IEEE80211_HE_PHY_CAP9_NOMINAL_PKT_PADDING_16US’ undeclared (first use in this function); did you mean ‘IEEE80211_HE_PHY_CAP9_NOMIMAL_PKT_PADDING_16US’?
 3303 |                                   u8_encode_bits(IEEE80211_HE_PHY_CAP9_NOMINAL_PKT_PADDING_16US,
      |                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                  IEEE80211_HE_PHY_CAP9_NOMIMAL_PKT_PADDING_16US
[... cut ...]
      |                                                  IEEE80211_HE_PHY_CAP9_NOMIMAL_PKT_PADDING_MASK
/usr/src/rtw89/rtw89/core.c: In function ‘rtw89_core_get_data_rate’:
/usr/src/rtw89/rtw89/core.c:748:1: error: control reaches end of non-void function [-Werror=return-type]
  748 | }
      | ^
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:289: /usr/src/rtw89/rtw89/core.o] Error 1
make[1]: *** [Makefile:1905: /usr/src/rtw89/rtw89] Error 2
make: *** [Makefile:97: all] Error 2
```

salimp2009, 2023-05-28:
I checkedout older commit #83f33c87e and that works

lwfinger, 2023-05-28:
There was a problem with kernel 5.19, but that is now fixed. Try another pull and make again.

salimp2009, 2023-05-28:
> There was a problem with kernel 5.19, but that is now fixed. Try another pull and make again.

it still is not working I get this error (commit #084f63d)
```
ERROR: modpost: "rtw8852b_chip_info" [/home/salim/developer/rtw89/rtw_8852be.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:133: /home/salim/developer/rtw89/Module.symvers] Error 1
make[2]: *** Deleting file '/home/salim/developer/rtw89/Module.symvers'
make[1]: *** [Makefile:1826: modules] Error 2
make: *** [Makefile:97: all] Error 2
    ~/dev/rtw89    main    ERROR: modpost: "rtw8852b_chip_info" [/home/salim/developer/rtw89/rtw_8852be.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:133: /home/salim/developer/rtw89/Module.symvers] Error 1
make[2]: *** Deleting file '/home/salim/developer/rtw89/Module.symvers'
make[1]: *** [Makefile:1826: modules] Error 2
make: *** [Makefile:97: all] Error 2
```

vizv, 2023-05-28:
I have the same issue:

```
...
  MODPOST /root/rtw89/Module.symvers
ERROR: modpost: "rtw8852b_chip_info" [/root/rtw89/rtw_8852be.ko] undefined!
...
```

And `/root/rtw89/rtw_8852b.ko` is not listed in `modules.order` for some reason:

```
$ cat modules.order
/root/rtw89/rtw89core.ko
/root/rtw89/rtw_8852a.ko
/root/rtw89/rtw_8852ae.ko
/root/rtw89/rtw_8851b.ko
/root/rtw89/rtw_8851be.ko
/root/rtw89/rtw_8852be.ko
/root/rtw89/rtw_8852c.ko
/root/rtw89/rtw_8852ce.ko
/root/rtw89/rtw89pci.ko
```

vizv, 2023-05-28:
Here are full output running with root:

```
❯ make clean
❯ make -j20
make -C /lib/modules/6.1.27-gentoo-dist/build M=/root/rtw89 modules
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-pc-linux-gnu-gcc (Gentoo 12.2.1_p20230428-r1 p2) 12.2.1 20230428
  You are using:           gcc (Gentoo 12.2.1_p20230428-r1 p2) 12.2.1 20230428
  CC [M]  /root/rtw89/core.o
  CC [M]  /root/rtw89/chan.o
  CC [M]  /root/rtw89/mac80211.o
  CC [M]  /root/rtw89/mac.o
  CC [M]  /root/rtw89/phy.o
  CC [M]  /root/rtw89/fw.o
  CC [M]  /root/rtw89/cam.o
  CC [M]  /root/rtw89/efuse.o
  CC [M]  /root/rtw89/regd.o
  CC [M]  /root/rtw89/sar.o
  CC [M]  /root/rtw89/acpi.o
  CC [M]  /root/rtw89/coex.o
  CC [M]  /root/rtw89/ps.o
  CC [M]  /root/rtw89/debug.o
  CC [M]  /root/rtw89/ser.o
  CC [M]  /root/rtw89/wow.o
  CC [M]  /root/rtw89/rtw8852a.o
  CC [M]  /root/rtw89/rtw8852a_table.o
  CC [M]  /root/rtw89/rtw8852a_rfk.o
  CC [M]  /root/rtw89/rtw8852a_rfk_table.o
  CC [M]  /root/rtw89/rtw8852ae.o
  CC [M]  /root/rtw89/rtw8851b.o
  CC [M]  /root/rtw89/rtw8851b_table.o
  CC [M]  /root/rtw89/rtw8851b_rfk.o
  CC [M]  /root/rtw89/rtw8851b_rfk_table.o
  CC [M]  /root/rtw89/rtw8851be.o
  CC [M]  /root/rtw89/rtw8852be.o
  CC [M]  /root/rtw89/rtw8852c.o
  CC [M]  /root/rtw89/rtw8852c_table.o
  CC [M]  /root/rtw89/rtw8852c_rfk.o
  CC [M]  /root/rtw89/rtw8852c_rfk_table.o
  CC [M]  /root/rtw89/rtw8852ce.o
  CC [M]  /root/rtw89/pci.o
  LD [M]  /root/rtw89/rtw_8852ae.o
  LD [M]  /root/rtw89/rtw_8851be.o
  LD [M]  /root/rtw89/rtw_8852be.o
  LD [M]  /root/rtw89/rtw_8852ce.o
  LD [M]  /root/rtw89/rtw_8852a.o
  LD [M]  /root/rtw89/rtw_8851b.o
  LD [M]  /root/rtw89/rtw89core.o
  LD [M]  /root/rtw89/rtw_8852c.o
  LD [M]  /root/rtw89/rtw89pci.o
  MODPOST /root/rtw89/Module.symvers
ERROR: modpost: "rtw8852b_chip_info" [/root/rtw89/rtw_8852be.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:126: /root/rtw89/Module.symvers] Error 1
make[1]: *** [Makefile:1964: modpost] Error 2
make: *** [Makefile:97: all] Error 2
```

As you can see, `rtw8852b.o` is not even get compiled :thinking:

vizv, 2023-05-28:
@lwfinger maybe:

```diff
diff --git a/Makefile b/Makefile
index d8765bf..b2a567c 100644
--- a/Makefile
+++ b/Makefile
@@ -66,6 +66,7 @@ obj-m += rtw_8851b.o
 obj-m += rtw_8851be.o
 rtw_8851be-y := rtw8851be.o
 
+obj-m += rtw_8852b.o
 rtw_8852b-y := rtw8852b.o \
                    rtw8852b_table.o \
                    rtw8852b_rfk.o \
```

lwfinger, 2023-05-28:
Yes, that is the correct fix. I read through the same error several times until you pointed it out.

One suggestion: You should NEVER build anything as root. You are too vulneralbe to have an errant or malicious Makefile destroying your system. The only time you need to be root is doins]g the 'sudo make install', and the sudo does that for you.

salimp2009, 2023-05-29:
> Yes, that is the correct fix. I read through the same error several times until you pointed it out.
>
> One suggestion: You should NEVER build anything as root. You are too vulneralbe to have an errant [...]

> That does not make any sense. That symbol is devined in rtw89_8852b.ko. Perhaps you have some stale code. Do a 'make clean', 'make', and 'sudo make install'.

that is how I do make clean make and sudo make install. I always do the same way. older commit works tough. even the package manager version is not workiing after these update but it was working before.

salimp2009, 2023-05-29:
after the commit #8705056 now it is working again. Thank you for the fix :+1: