lwfinger/rtw89 #250: rtw89_8852be - bluetooth doesn't work/scan

gonciarz, 2023-06-17:
WIFI works fine, but Bluetooth doesn't (at least I cannot find my devices).
The problem was originally created by @MulaGabriel 
[on Archlinux forum](https://bbs.archlinux.org/viewtopic.php?pid=2105334)

It seems the problem exist in those boards:
```
- ASUS PRIME X670-P WIFI
- ASUS PRIME B650M-A WIFI (90MB1C00-M0EAY0).
```

and specifically:
```
> lspci -d 10ec:b852 -k
08:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device b852
	Subsystem: AzureWave Device 5471
	Kernel driver in use: rtw89_8852be
	Kernel modules: rtw89_8852be
```

```
> lsmod | grep rtw
rtw89_8852be           16384  0
rtw89_8852b           360448  1 rtw89_8852be
rtw89_pci              86016  1 rtw89_8852be
rtw89_core            622592  2 rtw89_8852b,rtw89_pci
mac80211             1589248  2 rtw89_core,rtw89_pci
cfg80211             1327104  3 rtw89_8852b,rtw89_core,mac80211

```

```
> lsusb -d 13d3:3571
Bus 001 Device 002: ID 13d3:3571 IMC Networks Bluetooth Radio
```

```
dmesg | grep -i bluetooth
[    2.385961] usb 1-2: Product: Bluetooth Radio
[    3.540819] Bluetooth: Core ver 2.22
[    3.540831] NET: Registered PF_BLUETOOTH protocol family
[    3.540832] Bluetooth: HCI device and connection manager initialized
[    3.540834] Bluetooth: HCI socket layer initialized
[    3.540836] Bluetooth: L2CAP socket layer initialized
[    3.540838] Bluetooth: SCO socket layer initialized
[    3.719915] Bluetooth: hci0: Failed to read codec capabilities (-22)
[    4.342696] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    4.342697] Bluetooth: BNEP filters: protocol multicast
[    4.342700] Bluetooth: BNEP socket layer initialized
[    4.343045] Bluetooth: MGMT ver 1.22
[    9.269172] Bluetooth: RFCOMM TTY layer initialized
[    9.269181] Bluetooth: RFCOMM socket layer initialized
[    9.269182] Bluetooth: RFCOMM ver 1.11

```

```
rfkill list
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
```

```
bluetoothctl 
Agent registered
[bluetooth]# 
[bluetooth]# show
Controller XX:XX:XX:XX:XX:XX (public)
	Name: red
	Alias: red
	Class: 0x007c0104
	Powered: yes
	Discoverable: yes
	DiscoverableTimeout: 0x000000b4
	Pairable: yes
	UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
	UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00005005-0000-1000-8000-0002ee000001)
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v1D6Bp0246d0542
	Discovering: yes
	Roles: central
	Roles: peripheral
Advertising Features:
	ActiveInstances: 0x00 (0)
	SupportedInstances: 0x0a (10)
	SupportedIncludes: tx-power
	SupportedIncludes: appearance
	SupportedIncludes: local-name
	SupportedSecondaryChannels: 1M
	SupportedSecondaryChannels: 2M
	SupportedSecondaryChannels: Coded
[bluetooth]# power on
Changing power on succeeded
[bluetooth]# scan on
Discovery started
```

```
journalctl -u bluetooth.service
Jun 18 04:12:33 red systemd[1]: Starting Bluetooth service...
Jun 18 04:12:33 red bluetoothd[30147]: Bluetooth daemon 5.66
Jun 18 04:12:33 red systemd[1]: Started Bluetooth service.
Jun 18 04:12:33 red bluetoothd[30147]: Starting SDP server
Jun 18 04:12:33 red bluetoothd[30147]: profiles/audio/vcp.c:vcp_init() D-Bus experimental not enabled
Jun 18 04:12:33 red bluetoothd[30147]: src/plugin.c:plugin_init() Failed to init vcp plugin
Jun 18 04:12:33 red bluetoothd[30147]: profiles/audio/mcp.c:mcp_init() D-Bus experimental not enabled
Jun 18 04:12:33 red bluetoothd[30147]: src/plugin.c:plugin_init() Failed to init mcp plugin
Jun 18 04:12:33 red bluetoothd[30147]: profiles/audio/bap.c:bap_init() D-Bus experimental not enabled
Jun 18 04:12:33 red bluetoothd[30147]: src/plugin.c:plugin_init() Failed to init bap plugin
Jun 18 04:12:33 red bluetoothd[30147]: Bluetooth management interface 1.22 initialized
Jun 18 04:12:33 red bluetoothd[30147]: src/profile.c:register_profile() :1.254 tried to register 0000111f-0000-1000-8000-00805f9b34fb which is already registered
Jun 18 04:12:33 red bluetoothd[30147]: Endpoint registered: sender=:1.264 path=/MediaEndpoint/A2DPSource/ldac
Jun 18 04:12:33 red bluetoothd[30147]: Endpoint registered: sender=:1.264 path=/MediaEndpoint/A2DPSink/aptx_hd
Jun 18 04:12:33 red bluetoothd[30147]: Endpoint registered: sender=:1.264 path=/MediaEndpoint/A2DPSource/aptx_hd
Jun 18 04:12:33 red bluetoothd[30147]: Endpoint registered: sender=:1.264 path=/MediaEndpoint/A2DPSink/aptx
Jun 18 04:12:33 red bluetoothd[30147]: Endpoint registered: sender=:1.264 path=/MediaEndpoint/A2DPSource/aptx
(...)
```

Please let me know how I may help you to fix the issue.

lwfinger, 2023-06-18:
By obfucating the entire MAC address of your BT device, you made it impossible for me to tell what device you have. The first 3 sets of numbers tell the manufacturer and the last 3 give the serial number. You only need to obfuscate the last 3.

I would also like to see the Bluetooth part of the output of 'lsusb'. In addition, run the command 'sudo dmesg -t > dmesg.txt' and attach dmesg.txt to this issue.

gonciarz, 2023-06-18:
Wow, so fast response. Thank you @lwfinger. Give me a moment, I'll pass raw data.

gonciarz, 2023-06-18:
```
bluetoothctl
Agent registered
[CHG] Controller E8:FB:1C:AF:C0:97 Pairable: yes
[bluetooth]# show
Controller E8:FB:1C:AF:C0:97 (public)
	Name: red
	Alias: red
	Class: 0x006c0104
	Powered: yes
	Discoverable: no
	DiscoverableTimeout: 0x000000b4
	Pairable: yes
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v1D6Bp0246d0542
	Discovering: no
	Roles: central
	Roles: peripheral
Advertising Features:
	ActiveInstances: 0x00 (0)
	SupportedInstances: 0x0a (10)
	SupportedIncludes: tx-power
	SupportedIncludes: appearance
	SupportedIncludes: local-name
	SupportedSecondaryChannels: 1M
	SupportedSecondaryChannels: 2M
	SupportedSecondaryChannels: Coded

```

```
lsusb -d 13d3:3571 -v

Bus 001 Device 002: ID 13d3:3571 IMC Networks Bluetooth Radio
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.00
  bDeviceClass          224 Wireless
  bDeviceSubClass         1 Radio Frequency
  bDeviceProtocol         1 Bluetooth
  bMaxPacketSize0        64
  idVendor           0x13d3 IMC Networks
  idProduct          0x3571 
  bcdDevice            0.00
  iManufacturer           1 Realtek
  iProduct                2 Bluetooth Radio
  iSerial                 3 00e04c000001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00b1
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              500mA
[... 198 lines cut ...]
      bInterfaceProtocol      1 Bluetooth
      iInterface              4 Bluetooth Radio
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0031  1x 49 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0031  1x 49 bytes
        bInterval               1
Device Status:     0x0001
  Self Powered
```

gonciarz, 2023-06-18:
[dmesg.txt](https://github.com/lwfinger/rtw89/files/11780921/dmesg.txt)

gonciarz, 2023-06-18:
Just in case I also attach a dmesg from upstream kernel. A previous one was from zen flavor. The result was identical of course.
[dmesg_upstream.txt](https://github.com/lwfinger/rtw89/files/11780959/dmesg_upstream.txt)

I addition to that, I've blocked temporally your module and tried another bluetooth (usb dongle) and it was working find with gnome-bluetooth on my Archlinux. So I would eliminate distribution state as a factor here (it is a fresh installation).

lwfinger, 2023-06-18:
Note that the RTW8852BE package consists of two devices - a PCIe wifi unit and a USB controller for Bluetooth. They share the 2.4 GHz radio and antenna, but are otherwise separate.

The code to recognize your BT device will be in kernel 6.4, but it has not been backported to any older kernels, thus your kernel does not see your device and no driver is loaded. Not too surprising that it soes not work! :)

You need to clone the repo at https://github.com:lwfinger/rtw89-BT.git, make and install it. You will then have a driver that should work.

gonciarz, 2023-06-18:
Thank you for clarification, and sorry for additional trouble. It will be more convenient for me if I use:
https://aur.archlinux.org/packages/linux-mainline
Please let me know if your changes were already merged to upstream kernel.
The source of 'linux-mainline' package is https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git and 'v6.4-rc6' tag. I believe I'll find your latest changes there. If not then I'll compile your driver. Thank you.

lwfinger, 2023-06-19:
Yes, that change is in v6.4-rcX. If you think that building a kernel is easier that a 20 second compile of an out-of-kernel driver, then be my guest.

closed 2023-06-19 by lwfinger

gonciarz, 2023-06-19:
@lwfinger From developer's point of view I agree with you. I hate when I have to spend more than few seconds to compile my projects and run all my unit tests, but from distro management one I prefer to build a kernel once and stay for few weeks with a compiled kernel, then to switch to official or zen ones.
I also very rarely had to compile some extra modules, usually I use prepared packages to do that for me so I need to explore that path. Even though it might be obvious like: make configure; make compile; make install; 

Anyway, I've smoke-tested your changes and I was able to discover a nearby Bluetooth devices and connect with my headset.
Thank you for your help once again.