lwfinger/rtw89 #253: Hi, I'm trying to use rtl8852be from yours or hrex39.

DOOMMARINE117, 2023-06-30:
Hi, I'm trying to install rt18852be from you or hrex. 

But, it needs password.

Is there a way around that?

I'm typing git clone https.

lwfinger, 2023-06-30:
First of all, you should use rtw89 (this repo), rather than the vendor driver. If it wants a password, it is your GitHub password. Yhey stopped anonymous downloads about 6 months ago

DOOMMARINE117, 2023-06-30:
> First of all, you should use rtw89 (this repo), rather than the vendor driver. If it wants a password, it is your GitHub password. Yhey stopped anonymous downloads about 6 months ago

Do I need to be in root privileges?

If so what is the command?

DOOMMARINE117, 2023-06-30:
It keeps saying failed authentication.

lwfinger, 2023-06-30:
You should NOT be in root mode. Everything involved in cloning a repo and building the driver is done in user mode. Only the installation needs to be in root mode using the 'sudo make install'. The reason for that is thatthat command writes files into directories where only root has write privilege.

The command should be 'git clone https://github.com/lwfinger/rtw89.git', just the way it says in README.md, which can be read on the web site.

DOOMMARINE117, 2023-07-01:
> You should NOT be in root mode. Everything involved in cloning a repo and building the driver is done in user mode. Only the installation needs to be in root mode using the 'sudo make install'. The [...]

[image]

DOOMMARINE117, 2023-07-01:
Went ahead ordered a wifi adapter for Linux mint. My Linux distro.

lwfinger, 2023-07-01:
For some reason, I cannot load your photo. Whenever possible, you should copy the text from the screen and paste it into the comment on the issue.

closed 2023-07-01 by lwfinger