lwfinger/rtw89 #256: Still Facing wifi issues on ubuntu 20.04 LTS after following README.md

shulnak09, 2023-07-15:
Hi all,

I recently got a new lenovo legion and dual boot the system with ubunto 20.04 LTS. I followed the steps sequentially as per README.md but still face issues for the realtek driver rtw_8852ce.

My kernel version is:  5.15.0-76-generic
My realtek_adapter is : rtw_8852ce

I followed these steps sequentially:

```
sudo apt-get update
sudo apt-get install make gcc linux-headers-$(uname -r) build-essential git
git clone https://github.com/lwfinger/rtw89.git
cd rtw89
make
sudo make install
```

After that, I tried to  " sudo modprobe -v rtw". These are the drivers that were listed and it did not list  **rtw_8852ce**
rtw_8851b     rtw_8852ae    rtw88_8723d   rtw88_8821ce  rtw88_8822c   rtw88_pci
rtw_8851be    rtw_8852b     rtw88_8723de  rtw88_8822b   rtw88_8822ce  
rtw_8852a     rtw_8852be    rtw88_8821c   rtw88_8822be  rtw88_core 

Further, to see the error, I used the following command: " sudo dmesg | grep -e 8852 -e rtw", and got the following error:
```
[    1.500187] rtw89core: loading out-of-tree module taints kernel.
[    1.500317] rtw89core: module verification failed: signature and/or required key missing - tainting kernel
[    1.513050] rtw89_8852ce 0000:04:00.0: Direct firmware load for rtw89/rtw8852c_fw.bin failed with error -2
[    1.513055] rtw89_8852ce 0000:04:00.0: failed to early request firmware: -2
[    1.513162] rtw89_8852ce 0000:04:00.0: Direct firmware load for rtw89/rtw8852c_fw.bin failed with error -2
[    1.514349] rtw89_8852ce 0000:04:00.0: failed to wait firmware completion
[    1.514373] rtw89_8852ce 0000:04:00.0: failed to setup chip information
[    1.514688] rtw89_8852ce: probe of 0000:04:00.0 failed with error -22
[    2.968446] Bluetooth: hci0: RTL: examining hci_ver=0c hci_rev=000c lmp_ver=0c lmp_subver=8852
[    2.968449] Bluetooth: hci0: RTL: unknown IC info, lmp subver 8852, hci rev 000c, hci ver 000c

```

** I am new to this and have been struggling and for any any help from the community, I will be really grateful.**

Regards
Anshul

lwfinger, 2023-07-15:
[    1.513050] rtw89_8852ce 0000:04:00.0: Direct firmware load for rtw89/rtw8852c_fw.bin failed with error -2

You are missing the firmware for the device. Ubuntu should provide it, but you can download it from https:lwfinger.com/download. Copy file rtw8852c_fw.bin (as root) to /lib/firmware/rtw88/.

shulnak09, 2023-07-15:
Hi lwfinger,

Thank you so much for your help. I remember doing this step but was copying the file to some other location.

Initially, I tried copying the   rtw8852c_fw.bin (as root) to /lib/firmware/rtw88, it did not work, then I copied the file to rtw89 folder inside /lib/firmware and my wifi now works.

Thank a lot again for all your help. It means  a lot. I have been struggling with this issue for past 4 days.

Regards
Anshul

lwfinger, 2023-07-15:
That was my error. Copying to /lib/firmware/rtw89/ was correct.

You should have posted earlier. It should not take 4 days.

closed 2023-07-15 by lwfinger

hankpharris, 2025-03-19:
hello, sorry to necro, but i believe its appropriate, very new to this stuff. I'm experiencing a very similiar issue. I dont see the bin file in the depository or any reference to it in the read me. I did notice the portion which explicitly mentions lenovo devices, and executed the instructions to the best of my ability. 
The command you suggest running first doesnt appear to do anything (sudo cp 70-rtw89.conf /etc/modprobe.d/). Should I be running these commands at a specific point in the process as compared to after everything else? I then also unloaded and reloaded the kernel module. I also tried running the dracut command but was told it couldn't access boot details. You also suggest using some lines which require dev, driver, and driver_option names, but I'm a little confused on what these should be or how I can find them.
I believe my chipset is rtw89_8852ce
On boot I get get the message that the chipset failed to early request firmware, wait firmware completion, and setup chip info.
Let me know if any further info or logs would be beneficial