lwfinger/rtw89 #257: ifi keeps dropping out.

DOOMMARINE117, 2023-07-17:
is there a way to have it not to do that?

lwfinger, 2023-07-18:
Not without more details. What chip do you have? What signal strength? What shows in the dmesg log?

None of my rtw89 chips ever drop their connection. On occasion, I will get a reconnect, but those are rare.

DOOMMARINE117, 2023-07-18:
> Not without more details. What chip do you have? What signal strength? What shows in the dmesg log?
> 
> None of my rtw89 chips ever drop their connection. On occasion, I will get a reconnect, but those are rare.

It's a realtek chip I believe. 

Where do I find the dmesg log? 

Sounds like I may be missing something else if I'm dropping connection.

DOOMMARINE117, 2023-07-18:
my signal level is 36.

lwfinger, 2023-07-18:
The 5.785 GHz is the frequency, not a signal strength. A strength of 36 dBm is way too large - abour 4 watts. For a strong signal it should be about -40 dBm of 10^-7 watts. The scale is logarithmic.

Run the command 'sudo dmesg -t > dmesg.txt' and attach the file dmesg.txt to this issue. Once I see that, I will likely have more commands for you to run.

lwfinger, 2023-07-19:
The dmesg, there are lines like

[UFW BLOCK] IN=wlo1 OUT= MAC=10:68:38:78:0f:d4:f4:69:42:b3:0a:e4:08:00 SRC=18.190.141.247 DST=192.168.1.141 LEN=83 TOS=0x00 PREC=0x00 TTL=236 ID=0 DF PROTO=TCP SPT=443 DPT=43428 WINDOW=119 RES=0x00 ACK PSH URGP=0 

These come from your your firewall. They say that IP address 18.190.141.247 is trying to reach 192.168.1.141, which is on your LAN. I have no idea what machine has the local address, but nslookup says that source address belongs to AWS. There are a number of different sources. The first 3 translate to the following:

247.141.190.18.in-addr.arpa     name = ec2-18-190-141-247.us-east-2.compute.amazonaws.com
120.120.132.3.in-addr.arpa      name = ec2-3-132-120-120.us-east-2.compute.amazonaws.com.
239.159.92.190.in-addr.arpa     name = server.valknuden.com.

After a number of those, it appears that you rebooted the system.

I have no idea what you are trying to do with source port 443, which is redirested to a number of destination ports; however, every indication is that the wifi driver is working fine. It is your setup that is interferring with traffic.

DOOMMARINE117, 2023-07-19:
> The dmesg, there are lines like
>
> [UFW BLOCK] IN=wlo1 OUT= MAC=10:68:38:78:0f:d4:f4:69:42:b3:0a:e4:08:00 SRC=18.190.141.247 DST=192.168.1.141 LEN=83 TOS=0x00 PREC=0x00 TTL=236 ID=0 DF PROTO=TCP [...]

really? is there a way to fix this?  i tried to setup manually way before. but didnt workout. 

how to have it stop interfering with it?

DOOMMARINE117, 2023-07-19:
> No site should be reaching back into your computer unless it is responding to your requests for information. I do not know if that is what is happening here, but if it is, the firewall should ignore them.
> 
> Try disabling your firewall as a test and determine what happens. I do not know what distro you are using, and I know nothing of configuring UFW. I have used Linux for many years, and only use the firewall in my router. I have never had any problems.

i went ahead and disable firewall. that increase it up to signal level 43

lwfinger, 2023-07-19:
Disabling the firewall cannot increase the signal level.

I forgot to tell you how to measure signal level. Run the command
sudo iw dev wlo1 scan | egrep "signal|SSID|freq"

You will see matched lines such as
        freq: 5745
        signal: -71.00 dBm
        SSID: NETGEAR81-5G
Those 3 belong to a single AP/band combination. Find your SSID and report all 3 lines for that entry.

DOOMMARINE117, 2023-07-19:
> i found them and it shows 

	freq: 5745
	signal: -58.00 dBm
	SSID: 
	freq: 5785
	signal: -37.00 dBm
	SSID: SpectrumSetup-E3
		 * SSID List
		 * center freq segment 1: 155
		 * center freq segment 2: 0

DOOMMARINE117, 2023-07-19:
> Disabling the firewall cannot increase the signal level.
>
> I forgot to tell you how to measure signal level. Run the command sudo iw dev wlo1 scan | egrep "signal|SSID|freq"
>
> You will see matched [...]

i found them.

	freq: 5745
	signal: -58.00 dBm
	SSID: 
	freq: 5785
	signal: -37.00 dBm
	SSID: SpectrumSetup-E3
		 * SSID List
		 * center freq segment 1: 155
		 * center freq segment 2: 0

DOOMMARINE117, 2023-07-20:
so what to do next?

lwfinger, 2023-07-21:
The only reason that I have seen for "dropping" out was caused by the rest of your system. I have no idea what distro you are using, but I would get rid of it. The network-handling of user space seems to be really buggy.

DOOMMARINE117, 2023-07-21:
> The only reason that I have seen for "dropping" out was caused by the rest of your system. I have no idea what distro you are using, but I would get rid of it. The network-handling of user space seems to be really buggy.

sounds like its on the distro then. im using mint 21.2 should i switch over to a different one?

DOOMMARINE117, 2023-07-21:
i also realzied. i was viewing my drivers list. 

i have device-2 oon with 8812au/8821au installed cause i wasnt sure at the time when doing linux of finding the right driver at the time.

how do you remove device-2?

this is when i type in inxi-Nn

lwfinger, 2023-07-21:
I have no idea what you did. If you have two different drivers for the same device, you need to uninstall one of them.

DOOMMARINE117, 2023-07-21:
> I have no idea what you did. If you have two different drivers for the same device, you need to uninstall one of them.

i have tried to remove it. but it doesnt appear on the terminal. 

is there a way around it?

DOOMMARINE117, 2023-07-21:
if i were to switch to a different distro, what distro should i go too? MX Linux, https://novos-wiki.novusy.cf/ , RebornOS, Drauger os and endeavour os?

lwfinger, 2023-07-21:
None of those. All of them are niche products that do not get as much usage as more mainline distros. What desktop are you using? 

I would go with one of the more-mainline distros such as Fedora, Ububtu, Debian, or openSUSE.   The latter is my distro of choice. OpenSUSE's main advantage is that they offer a number of desktops (KDE, Gnome, XFCE, etc.) at installation time, not by choosing a different download file. In fact, if you were to choose openUSE Tumbleweed, your wifi would be handled "out-of-the box" with no further actions required. Tumbleweed is a rolling release with daily updates, which may not be good for you, but it just works. :) OpenSUSE's Leap 15.5 version has a repo that has the rtw89 drivers, thus they also work with very little effort.

DOOMMARINE117, 2023-07-21:
> None of those. All of them are niche products that do not get as much usage as more mainline distros. What desktop are you using?
>
> I would go with one of the more-mainline distros such as Fedora, [...]

I'm using a amd Ryzen 7 laptop. I switched from Nvidia.

lwfinger, 2023-07-21:
The hardware is compatible with Linux and all desktops. Your desktop is likely Cinnamon. If you install openSUSE Leap15.5 and select the XFCE desktop at installation, you will be getting a desktop very near to the one you are using.

I know the networking user-mode software works fine with openSUSE. It definitely was not working in Mint.

DOOMMARINE117, 2023-07-21:
> 

> The hardware is compatible with Linux and all desktops. Your desktop is likely Cinnamon. If you install openSUSE Leap15.5 and select the XFCE desktop at installation, you will be getting a desktop [...]

i see. so sounds like linux mint needs more work than i thought. many said it was good. but, they never mention the buggy part.

so is opensuse a operating system too?

lwfinger, 2023-07-21:
It is a Linux distro. They package the various desktops and other things just like other distros do. I think they take a lot more precautions to catch bugs before they reach the user.

OpenSUSE supports gaming just like most others. Good gaming is a matter of the video driver. For best performance, you will need to use the drivers from nVidia. BTW, openSUSE packages them too.

DOOMMARINE117, 2023-07-21:
> It is a Linux distro. They package the various desktops and other things just like other distros do. I think they take a lot more precautions to catch bugs before they reach the user.
>
> OpenSUSE [...]

do they support amd?

lwfinger, 2023-07-22:
You are becoming a pain. Of course they support AMD processors. ALL 64-bit cpus execute the same instruction set. I have 3 machines running openSUSE - 2 with Intel cpus and 1 with AMD.

closed 2023-07-22 by DOOMMARINE117