lwfinger/rtw89 #258: Wifi adapter still not found after installation

bilginsenol, 2023-07-22:
Hello,

I got a brand new freedos laptop and tried installing Zorin Os 16.2 a few days ago. Upon installation and updates, I see no wifi adapter available on the settings menu. It says the Bluetooth is on and available, but I cannot see and connect to any devices around. I later tried installing a fresh Ubuntu 22.04 LTS and encountered the same issue. 

At some point, I tried installing Solus where the wifi worked out of the box and I identified that my laptop has rtw8852be (the reported driver there was "rtw89_8852be"). Currently I am back on Zorin 16.2 and I followed the installation procedure you have described on the readme.md page. After reboot, I still do not see available wifi adapters and it still says that the bluetooth is on and working while it certainly is not.

Here are some outputs that might be useful for diagnostics.

```
uname -r

5.15.0-56-generic
``` 
```
lspci

00:00.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir Root Complex
00:00.2 IOMMU: Advanced Micro Devices, Inc. [AMD] Renoir IOMMU
00:01.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir PCIe Dummy Host Bridge
00:02.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir PCIe Dummy Host Bridge
00:02.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir PCIe GPP Bridge
00:02.4 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir PCIe GPP Bridge
00:08.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir PCIe Dummy Host Bridge
00:08.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir Internal PCIe GPP Bridge to Bus
00:08.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir Internal PCIe GPP Bridge to Bus
00:14.0 SMBus: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller (rev 51)
00:14.3 ISA bridge: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge (rev 51)
00:18.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 166a
00:18.1 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 166b
00:18.2 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 166c
00:18.3 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 166d
00:18.4 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 166e
00:18.5 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 166f
00:18.6 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 1670
00:18.7 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 1671
01:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device b85b
02:00.0 Non-Volatile memory controller: Micron Technology Inc Device 5411 (rev 01)
03:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Device 15e7 (rev c4)
03:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Device 1637
03:00.2 Encryption controller: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 10h-1fh) Platform Security Processor
03:00.3 USB controller: Advanced Micro Devices, Inc. [AMD] Renoir USB 3.1
03:00.4 USB controller: Advanced Micro Devices, Inc. [AMD] Renoir USB 3.1
03:00.5 Multimedia controller: Advanced Micro Devices, Inc. [AMD] Raven/Raven2/FireFlight/Renoir Audio Processor (rev 01)
03:00.6 Audio device: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 10h-1fh) HD Audio Controller
04:00.0 SATA controller: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] (rev 81)
04:00.1 SATA controller: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] (rev 81)
```

```
lsmod | grep rtw

rtw_8852be             16384  0
rtw_8852b             389120  1 rtw_8852be
rtw89pci               61440  1 rtw_8852be
rtw89core             516096  2 rtw89pci,rtw_8852b
mac80211             1249280  2 rtw89pci,rtw89core
cfg80211              970752  3 rtw89core,mac80211,rtw_8852b
```

```
dmesg | grep rtw

[    5.272891] rtw89core: loading out-of-tree module taints kernel.
[    5.296092] rtw89core: module verification failed: signature and/or required key missing - tainting kernel
[    5.316746] rtw89_8852be 0000:01:00.0: Direct firmware load for rtw89/rtw8852b_fw-1.bin failed with error -2
[    5.316760] rtw89_8852be 0000:01:00.0: Direct firmware load for rtw89/rtw8852b_fw.bin failed with error -2
[    5.316762] rtw89_8852be 0000:01:00.0: failed to early request firmware: -2
[    5.316842] rtw89_8852be 0000:01:00.0: enabling device (0000 -> 0003)
[    5.316869] rtw89_8852be 0000:01:00.0: Direct firmware load for rtw89/rtw8852b_fw.bin failed with error -2
[    5.318995] rtw89_8852be 0000:01:00.0: failed to wait firmware completion
[    5.319000] rtw89_8852be 0000:01:00.0: failed to setup chip information
[    5.319572] rtw89_8852be: probe of 0000:01:00.0 failed with error -22
```

There are clearly some issues as reported by the last messages but I am not sure how to resolve them.

Thanks for the help.

bilginsenol, 2023-07-22:
Ok, I noticed I dropped the firmware downloaded from lwfinger.com/download to wrong location. My wifi now works after fixing that. But still, my bluetooth does not function as I have previously described: It is on and looks like trying to scan but nothing shows up even when there are discoverable devices. I am not sure how to diagnose the bluetooth problems.

lwfinger, 2023-07-22:
This driver does not handle Bluetooth.

closed 2023-07-22 by lwfinger