lwfinger/rtw89 #259: rtw89_8852be doesn't work as expected

ScaMar, 2023-07-23:
Hello everyone,
i have this card `03:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device b852` on a laptop i don't use every day.
I use it just when i'm not at home, so i'm not sure starting from which kernel this device stopped working properly.

I'm using `OpenSUSE Tumbleweed`. Usually i power on this device once in a week to update, to sync my nextcloud files, and to check everything i need is working. So, no intense testing. I notice this behaviour during my last travel, the network was really unreliable i was forced to use my 4G device via usb cable.

I read somewhere this issue happens when the device switch from 2.4 to 5GHz, so i've assigned two different names to 2.4GHz and 5GHz but issue is still there.

This is my laptop:
```txt
BIOS Information
        Vendor: HP
        Version: U85 Ver. 01.07.00
        Release Date: 03/24/2023
        Address: 0xF0000
        Runtime Size: 64 kB
        ROM Size: 32 MB
        Characteristics:
                PCI is supported
                PC Card (PCMCIA) is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Print screen service is supported (int 5h)
                8042 keyboard services are supported (int 9h)
                Serial services are supported (int 14h)
                Printer services are supported (int 17h)
                ACPI is supported
                USB legacy is supported
                Smart battery is supported
                BIOS boot specification is supported
                Function key-initiated network boot is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 7.0
        Firmware Revision: 33.97

Handle 0x0014, DMI type 1, 27 bytes
System Information
        Manufacturer: HP
        Product Name: HP ProBook 440 14 inch G9 Notebook PC
        Version:  
        Serial Number: xxxxxxx
        UUID: xxxxxxxxx
        Wake-up Type: Power Switch
        SKU Number: xxxxxxx
        Family: 103C_5336AN HP ProBook
```

This is my realtek card:
```txt
03:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device b852
        Subsystem: Hewlett-Packard Company Device 88e3
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 147
        Region 0: I/O ports at 4000 [size=256]
        Region 2: Memory at 80900000 (64-bit, non-prefetchable) [size=1M]
        Capabilities: [40] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=375mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [50] MSI: Enable+ Count=1/1 Maskable- 64bit+
                Address: 00000000fee005b8  Data: 0000
        Capabilities: [70] Express (v2) Endpoint, MSI 00
                DevCap: MaxPayload 256 bytes, PhantFunc 0, Latency L0s <4us, L1 <64us
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 10W
                DevCtl: CorrErr- NonFatalErr- FatalErr- UnsupReq-
                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop- FLReset-
                        MaxPayload 256 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr+ TransPend-
                LnkCap: Port #0, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <4us, L1 <64us
                        ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
                LnkCtl: ASPM L1 Enabled; RCB 64 bytes, Disabled- CommClk+
                        ExtSynch- ClockPM+ AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 2.5GT/s, Width x1
                        TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
                DevCap2: Completion Timeout: Not Supported, TimeoutDis+ NROPrPrP- LTR+
                         10BitTagComp- 10BitTagReq- OBFF Via message/WAKE#, ExtFmt- EETLPPrefix-
                         EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
                         FRS- TPHComp- ExtTPHComp-
                         AtomicOpsCap: 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR+ 10BitTagReq- OBFF Disabled,
                         AtomicOpsCtl: ReqEn-
                LnkCap2: Supported Link Speeds: 2.5GT/s, Crosslink- Retimer- 2Retimers- DRS-
                LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
                LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete- EqualizationPhase1-
                         EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
                         Retimer- 2Retimers- CrosslinkRes: unsupported
        Capabilities: [100 v2] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
                AERCap: First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
                        MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
                HeaderLog: 00000000 00000000 00000000 00000000
        Capabilities: [148 v1] Device Serial Number 00-e0-4c-ff-fe-88-52-01
        Capabilities: [158 v1] Latency Tolerance Reporting
                Max snoop latency: 3145728ns
                Max no snoop latency: 3145728ns
        Capabilities: [160 v1] L1 PM Substates
                L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
                          PortCommonModeRestoreTime=30us PortTPowerOnTime=60us
                L1SubCtl1: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+
                           T_CommonMode=0us LTR1.2_Threshold=106496ns
                L1SubCtl2: T_PwrOn=60us
        Kernel driver in use: rtw89_8852be
        Kernel modules: rtw89_8852be, rtw_8852be
```

The errors i'm facing is:
```txt
rtw89_8852be 0000:03:00.0: timed out to flush queues
```

What i did try until now is:

- rebuild the module from scratch from this git repository
- add those options to the modules:
```txt
options rtw89_pci disable_clkreq=y disable_aspm_l1=y disable_aspm_l1ss=y
options rtw89pci disable_clkreq=y disable_aspm_l1=y disable_aspm_l1ss=y
```

The firmware i'm using is:
```txt
[    9.929422] rtw89_8852be 0000:03:00.0: Direct firmware load for rtw89/rtw8852b_fw-1.bin failed with error -2
[    9.929522] rtw89_8852be 0000:03:00.0: Direct firmware load for rtw89/rtw8852b_fw.bin failed with error -2
[    9.929529] rtw89_8852be 0000:03:00.0: failed to early request firmware: -2
[    9.929765] rtw89_8852be 0000:03:00.0: enabling device (0000 -> 0003)
[    9.961473] rtw89_8852be 0000:03:00.0: Firmware version 0.27.32.1, cmd version 0, type 1
[    9.961494] rtw89_8852be 0000:03:00.0: Firmware version 0.27.32.1, cmd version 0, type 3
[   10.249367] rtw89_8852be 0000:03:00.0: chip rfe_type is 1
[   10.289346] rtw89_8852be 0000:03:00.0 wlp3s0: renamed from wlan0
```

Do you have any other suggestion?

Thank you

lwfinger, 2023-07-23:
There are some problems with compressed firmware (.xz) in loading firmware (https://bugzilla.suse.com/show_bug.cgi?id=1212808). We are working on this problem. You can eliminate it by decompressing the openSUSE-supplied firmware, or by downloading the uncompressed files from https://lwfinger.com/download and saving them in /lib/firmware/rtw89/.

The only other suggestioon I have is for you to install the version 0.29 firmware. The filename is /lib/firmware/rtw89/rtw8852b_fw-1.bin. It does make it better. Those module options do not address your problem, and should be eliminated unless other problems appear.

ScaMar, 2023-08-14:
I am now using  the module provided by the kernel `6.4.8-1-default` on 
```text
LSB Version:    n/a
Distributor ID: openSUSE
Description:    openSUSE Tumbleweed
Release:        20230809
Codename:       n/a
```
I was forced to set option `options rtw89_core disable_ps_mode=y` due to `unable to leave ps mode` error (and all sort of the following ones).
It seems working fine now, i will report errors if any.
Also i am using the firmware provided by OpenSuse. At the end of my sessions the only lines i can read in messages file are:
```bash
# dmesg | grep rtw
[    6.525138] rtw89_8852be 0000:03:00.0: loaded firmware rtw89/rtw8852b_fw-1.bin
[    6.525287] rtw89_8852be 0000:03:00.0: enabling device (0000 -> 0003)
[    6.528049] rtw89_8852be 0000:03:00.0: Firmware version 0.29.29.1, cmd version 0, type 5
[    6.528054] rtw89_8852be 0000:03:00.0: Firmware version 0.29.29.1, cmd version 0, type 3
[    6.854223] rtw89_8852be 0000:03:00.0: chip rfe_type is 1
[    6.890579] rtw89_8852be 0000:03:00.0 wlp3s0: renamed from wlan0
```

Regards