lwfinger/rtw89 #262: rtw89_8852be : wifi and bluetooth interfere

MK73DS, 2023-08-03:
Hello,
Recently, I noticed I couldn't use bluetooth headsets with my PC since the audio is very chopped. After some investigation, I found that the wifi is causing interference since both bluetooth and wifi use the same physical antenna. This finding was made clear since from the very moment I turn off my wifi (`iwctl adapter phy0 set-property Powered off`) the audio is immediately glitchless, and immediately when I turn it back on it starts skipping a lot. This is weird since I'm connected to a 5GHz network and interference shouldn't happen...

However, this wasn't always the case, and bluetooth and wifi were able to coexist without any issue even on a 2.4GHz network, so my hardware (or my BIOS) definitely isn't at fault. I tried to bisect the faulting commit, I only did it on a day by day basis and not on a commit by commit basis in order to save time, here are my findings (format is DD/MM/YYYY) : 
- 22/05/2023 (commit hash d25cd621a2272267b34cc299bf9526fb6c75e535 ) : No problem
- 26/05/2023 (commit hash 2286917a2d8c8ee4b02dab09cf08bdda8ee4737e ) : Wifi doesn't work
- 27/05/2023 (commit hash 0d763459bedf292fc83f97f530d3994f70728c8c ) : Wifi doesn't work
- 28/05/2023 (commit hash 8705056914a1e31d34a9b17c5b7df2e5ceb8792c ) : Wifi works but interference with bluetooth

I do not have any technical knowledge so I can't really be more precise on what introduced the issue, I am sorry. Also, I can't seem to find any log generated when the glitches happen during playback of audio...

Here are my system infos: 
```
System:
  Host: ?????? Kernel: 6.4.7-arch1-2 arch: x86_64 bits: 64 Desktop: KDE Plasma
    v: 5.27.7 Distro: Arch Linux
Machine:
  Type: Laptop System: LENOVO product: 82SN v: IdeaPad 5 Pro 16ARH7
    serial: <superuser required>
  Mobo: LENOVO model: LNVNB161216 v: SDK0K17763 WIN
    serial: <superuser required> UEFI: LENOVO v: J4CN36WW date: 12/29/2022
Battery:
  ID-1: BAT0 charge: 57.3 Wh (77.6%) condition: 73.8/75.0 Wh (98.4%)
CPU:
  Info: 8-core AMD Ryzen 7 6800HS Creator Edition [MT MCP] speed (MHz):
    avg: 1800 min/max: 1600/4784
Graphics:
  Device-1: AMD Rembrandt [Radeon 680M] driver: amdgpu v: kernel
  Device-2: Chicony Integrated Camera driver: uvcvideo type: USB
  Display: x11 server: X.Org v: 21.1.8 driver: X: loaded: amdgpu
    unloaded: modesetting dri: radeonsi gpu: amdgpu s-res: 2560x1600
    resolution: 2560x1600
  API: OpenGL v: 4.6 Mesa 23.3.0-devel (git-c9b2f83a2f) renderer: AMD
    Radeon Graphics (rembrandt LLVM 15.0.7 DRM 3.52 6.4.7-arch1-2)
Network:
  Device-1: Realtek driver: rtw89_8852be
Drives:
  Local Storage: total: 476.94 GiB used: 144.64 GiB (30.3%)
```

Thank you for helping.

_Note_ : I know my kernel comes with this driver. In fact, I noticed the issue first by using the kernel's driver. During my tests, I have removed the rtw89 driver from the kernel so that I'm sure it doesn't interfere with my bisect.

lwfinger, 2023-08-03:
Your so-called bisection is essentially useless. You need to specify what commit failed. As this driver is derived from the kernel driver in wireless-next, you may need to bisect to find the exact commit that failed.

The driver has coexistence routines to ensure that BT and wifi can coexist. As your wifi is 5GHz and BT always uses 2.4 GHz, they should not interfere.

MK73DS, 2023-08-03:
Thank you for your quick answer. I did bisect commit by commit. The ones where I didn't get any wifi were because of an error in the Makefile that was corrected in a later commit (this one 8705056914a1e31d34a9b17c5b7df2e5ceb8792c ), so what I did is checkout each commit and manually edit the Makefile while testing.

I found that up until this commit ee73802fd9c767a7a7b2b6daa508267c00c16509 everything works fine, but the next commit 2ae0853b32389e6b97e65a28562be893dc30bdaf introduced the bug I experience. I hope this will make it easier for you (or anyone in charge of this kernel) to find out the issue and hopefully fix it.

I agree with you that 5GHz wifi and bluetooth shouldn't interfere, but they do, I don't know why. I am definitely sure I am connected to a 5GHz network from the output of `iwctl station wlan0 show` and the fact that the only wifi I have at home is 5GHz.

lwfinger, 2023-08-03:
That info is a big help. If you have to do this kind of thing again, it is a lot faster to do a 'git bisect'. If necessary, you can make a patch such as was needed for Makefile.

Please do a further test for me. Using a bad version, do the following:

sudo modprobe -rv rtw_8852be
sudo modprobe -rv rtw89core
sudo modprobe -v rtw89core disable_ps_mode=y
sudo modprobe -v rtw_8852be

Does the result work? Please post the console output so that I can see exactly what it printed. Thanks.

MK73DS, 2023-08-03:
Thank you for the info, I am not knowledgeable at all when it comes to git and its more advanced commands. Here is the output of the commands you wanted me to try, I'm using the latest commit from the main branch since it is a bad version.

```
$ sudo modprobe -rv rtw_8852be
rmmod rtw_8852be
rmmod rtw89pci
rmmod rtw_8852b
$ sudo modprobe -rv rtw89core 
rmmod rtw89core
rmmod mac80211
$ sudo modprobe -v rtw89core disable_ps_mode=y
insmod /lib/modules/6.4.7-arch1-2/kernel/net/mac80211/mac80211.ko.zst 
insmod /lib/modules/6.4.7-arch1-2/kernel/drivers/net/wireless/realtek/rtw89/rtw89core.ko disable_ps_mode=y
$ sudo modprobe -v rtw_8852be
insmod /lib/modules/6.4.7-arch1-2/kernel/drivers/net/wireless/realtek/rtw89/rtw89pci.ko 
insmod /lib/modules/6.4.7-arch1-2/kernel/drivers/net/wireless/realtek/rtw89/rtw_8852b.ko 
insmod /lib/modules/6.4.7-arch1-2/kernel/drivers/net/wireless/realtek/rtw89/rtw_8852be.ko
```

I also confirm that after this, the issue is gone.
I also tried with the driver included in the linux 6.4 kernel, since I have the same issue with it. By replacing `rtw_8852be` with `rtw89_8852be` and `rtw89core` with `rtw89_core` in the commands you wanted me to try, the issue is also gone (as expected I guess).

I want to thank you very much once more. It is usually rare to get a response for an issue, especially ones that are that helpful.

lwfinger, 2023-08-03:
You are welcome. Helping finding the commit that causes a problem is the kind of thing I can help with.

I will report this problem to the person in charge of the driver at Realtek and see what he has to say. If you want to be kept in the loop, E-mail me a address to use. That will expose that address to an open mailing list, and I understand if you want to avoid that.

lwfinger, 2023-08-04:
FYI, this is what I wrote under the subject "Regression in rtw89":
A user of my GitHub repo has reported a regression beginning with commit 26a125f550a3 ("wifi: rtw89: correct PS calculation for SUPPORTS_DYNAMIC_PS").
The user finds interference between wifi and the BT device that is part of the package.

We were able to determine that loading rtw89_core with the "disable_ps_mode=y" option does fix the problem. Our discussion of the problem can be seen at https://github.com/lwfinger/rtw89/issues/262#issuecomment-1664658764.

Reading the patch in question, it seems that a vif count of 1 is not sufficient to enable PS, and that it is also necessary to make sure that BT is not active as well.

MK73DS, 2023-08-04:
Amazing, thank you very much. I don't think I will be of any use in the loop and I prefer to not share my email address publicly, sorry. I hope they will address this issue quickly.

lwfinger, 2023-08-04:
I understand your point, which is why I asked the question the way I did.

The Realtek developer and I have a good working arrangement, and he got back to me very quickly. He stated that this patch did not break PS, but that it was broken before. This change merely turned it on correctly, as it is essential for battery life in mobile devices. He also suggested a number of things to try:

1. Make sure your firmware is the latest. You will find the wifi firmware at https://lwfinger.com/download/. Copy the two rtw8852b_fw*.bin files to your local computer, and then move them (as root) to /lib/firmware/rtw89/. For the BT firmware, copy all the files from https://lwfinger.com/rtl_bt/ to your computer, and move  them (as root) to /lib/firmware/rtl_bt/. After these firmware files are updated, reboot and test the setup for BT interfereence.
2. If it still fails, open two terminal windows and tile them so that both can be seen. In one, enter the command 'sudo iw wlan0 power_save off'.  Change the "wlan0" to the name actually used for your device. If you do not already know this, you can get it from the 'sudo dmesg | grep renamed'. At this point, BT should be working.
3. In the second terminal, create a file named script.sh containing the following:
#!/bin/sh
while [ true ]
do
    cat /sys/kernel/debug/ieee80211/phy0/rtw89/btc_info
    date
    sleep 2
done
4. Run that script using 'sudo bash script.sh'. It will print every 2 seconds.
5. In the first terminal window, toggle between 
    'sudo iw wlan0 power_save on'
and
    'sudo iw wlan0 power_save off'
while running your head phones. Allow about 10 seconds between the changes.
6. When failure happens, note the time in the second window where the problem started, and post the output from the second terminal with the time where the problem started.

MK73DS, 2023-08-05:
Thank you again. 

1. Downloading and copying the files didn't work. Maybe because the firmware files I have on my system are compressed .zst files. So I then removed all the .zst files corresponding to the firmware files I downloaded and rebooted. The issue is still present. 
2. As you said, as soon as I disable power_save, the issue immediately disappear. At that point, everything works fine.
3. (nothing particular to say, I created the script)
4. (nothing particular to say, I ran the script)
5. As said before, the effect of enabling/disabling the power_save feature is immediate on my bluetooth audio. I still waited the 10 seconds like you suggested.
6. I will post three times the output of the second terminal.

<details>
<summary>First is when the problem started. It happened as soon as I enabled the power_save feature on my wifi card.</summary>

```
=========================================
WL FW / BT FW           0.29.29.1 / NA
manual                  0
=========================================

 [bt_info]       : raw_data[49 00 c5 00 04 7f] (type:auto/cnt:1576/same:1)
=========================================
========== [BTC COEX INFO (1)] ==========
 [coex_version]  : Coex:7.0.1(branch:19), WL_FW_coex:7.8.0(branch:29)(Match, desired:5.5.0), BT_FW_coex:7(Match, desired:5)
 [sub_module]    : WL_FW:0.29.29.1, BT_FW:0x1fc6b20f(patch)
 [hw_info]       : cv:1, rfe_type:0x1, ant_iso:10, ant_pg:2, 3rd_coex:0, dbcc:0, tx_num:2, rx_num:2
========== [WL Status] ==========
 [status]        : link_mode:9, rf_off:0, power_save:1, scan:N(band:1/phy_map:0x0), connecting:N, roam:N, 4way:N, init_ok:Y
 [port_0]        : role=1(phy-0), connect=2(client_cnt=-1), mode=3, center_ch=42, bw=2, mac_id=0, max_tx_time=0us, max_tx_retry=0
 [port_0]        : rssi=-59dBm(51), busy=0, dir=DL, tx[rate:260/busy_level:0], rx[rate:280/busy_level:0/drop:0]
========== [BT Status] ==========
 [status]        : enable:Y, btg:Y, connect:Y, igno_wl:N, mailbox_avl:Y, rfk_state:0x0
 [profile]       : profile:A2DP, multi-link:N, role:Master, ble-connect:N, CQDDR:N, A2DP_active:Y, PAN_active:N
 [link]          : rssi:-53dBm, tx_rate:3M,  acl_busy!! afh_map[ffff_ffff_ffff_f3f1_fffe], wl_ch_map[en:0/ch:0/bw:32]
 [stat_cnt]      : retry:0, relink:1, rate_chg:11, reinit:0, reenable:1, role-switch:0, afh:0, inq_page:1(inq:0/page:1), igno_wl:1
 [A2DP]          : type:Legacy, bit-pool:63, flush-time:0, vid:0xa, Dev-name:0x2d4857, sut_pwr:0, golden-rx:0
 [bt_info]       : raw_data[49 00 c5 00 04 7f] (type:auto/cnt:1576/same:1)
 [trx_req_cnt]   : Hi-rx = 178, Hi-tx = 0, Lo-rx = 546, Lo-tx = 546 (bt_polut_wl_tx = 0)(BG:0-0/INIT:0-0/LE:0-0)
========== [Mechanism Status (Auto)] ==========
 [status]        : type:shared, reason:NTFY_RADIO_STATE(), action:WL_OFF(), ant_path:5, run_cnt:2220
 [dm_steps]      : -> UPDATE_BT_INFO      -> WL_5G          -> OFF_EQ0      -> NTFY_WL_STA         -> WL_5G          -> OFF_EQ0      
 [dm_steps]      : -> UPDATE_BT_INFO      -> WL_5G          -> OFF_EQ0      -> UPDATE_BT_INFO      -> WL_5G          -> OFF_EQ0      
 [dm_steps]      : -> UPDATE_BT_INFO      -> WL_5G          -> OFF_EQ0      -> UPDATE_BT_INFO      -> WL_5G          -> OFF_EQ0      
 [dm_steps]      : -> UPDATE_BT_INFO      -> WL_5G          -> OFF_EQ0      -> UPDATE_BT_INFO      -> WL_5G          -> OFF_EQ0      
 [dm_steps]      : -> NTFY_WL_STA         -> WL_5G          -> OFF_EQ0      -> NTFY_RADIO_STATE    -> WL_OFF         -> OFF_BT       
 [dm_flag]       : wl_only:0, bt_only:0, igno_bt:1, free_run:0, wl_ps_ctrl:0, wl_mimo_ps:0, leak_ap:0, fw_offload:N
 [trx_ctrl]      : wl_rssi_lvl:3, para_lvl:0, wl_tx_pwr:orig, wl_rx_lvl:0, bt_tx_pwr_dec:0, bt_rx_lna:7(Ori-tbl), wl_btg_rx:0
 [dm_ctrl]       : wl_tx_limit[en:0/max_t:5280us/max_retry:31], bt_slot_reg:0-TU, bt_scan_rx_low_pri:0
 [tdma_policy]   : type:0, rx_flow_ctrl:0, tx_pause:0, wl_toggle_n:0, leak_n:0, ext_ctrl:0, policy_type:0
 [slot_list]     :   OFF[100/0xe5555555/0],   B2W[005/0xea5a5a5a/1],    W1[070/0xea5a5a5a/1],    W2[015/0xea5a5a5a/1],   W2B[015/0xea5a5a5a/1]
 [slot_list]     :    B1[250/0xe5555555/0],    B2[007/0xea5a5a5a/0],    B3[005/0xe5555555/0],    B4[050/0xe5555555/0],    LK[020/0xea5a5a5a/1]
 [slot_list]     :   BLK[500/0x55555555/0],   E2G[000/0xea5a5a5a/0],   E5G[000/0xffffffff/1],   EBT[000/0xe5555555/0], ENULL[000/0xaaaaaaaa/1]
 [slot_list]     :   WLK[250/0xea5a5a5a/0], W1FDD[050/0xffffffff/1], B1FDD[050/0xffffdfff/1]
========== [HW Status] ==========
 [scoreboard]    : WL->BT:0x00004083(cnt:443), BT->WL:0x4001051b(total:219, bt_update:14)
 [reg]           : 0_0xda24=0xffffffff, 0_0xda28=0xffffffff, 0_0xda2c=0xf0ffffff, 0_0xda30=0x00000008, 0_0xda4c=0x0000000d, 0_0xda10=0x00000100
 [reg]           : 0_0xda20=0x00000002, 0_0xda34=0x00000100, 0_0xcef4=0x00000000, 0_0x8424=0x00000000, 0_0xd200=0x00000000, 0_0xd220=0x00000000
 [reg]           : 1_0x0980=0x10182250, 1_0x4738=0x8325c500, 1_0x4688=0x1a10ff04, 1_0x4694=0x04fa2a2a
========== [Statistics] ==========
 [summary]       : h2c_cnt=886(fail:1), c2h_cnt=0
 [notify_cnt]    : total=821, show_coex_info=162, power_on=0, init_coex=1, power_off=0, radio_state=398, role_info=2, wl_rfk=5, wl_sta=238
 [notify_cnt]    : scan_start=2, scan_finish=2, switch_band=2, special_pkt=5, timer=4, control=0, customerize=0sam. 05 août 2023 10:18:38 CEST
```
</details>

<details>
<summary>Second is the output right before the previous one, when power_save was disabled and no issues were present.</summary>

```
[... cut ...]
 [reg]           : 1_0x0980=0x10182250, 1_0x4738=0x8325c500, 1_0x4688=0x1a10ff04, 1_0x4694=0x04fa2a2a
========== [Statistics] ==========
 [summary]       : h2c_cnt=904(fail:1), c2h_cnt=0
 [notify_cnt]    : total=841, show_coex_info=172, power_on=0, init_coex=1, power_off=0, radio_state=406, role_info=2, wl_rfk=5, wl_sta=240
 [notify_cnt]    : scan_start=2, scan_finish=2, switch_band=2, special_pkt=5, timer=4, control=0, customerize=0sam. 05 août 2023 10:18:48 CEST
```
</details>

I hope this contains all the informations you need.

lwfinger, 2023-08-05:
The driver would load the uncompressed version first, thus the .zst stuff would have been ignored, but deleting it did not hurt.

I sent your logs upstream and will let you know when I hear anything.

At least you know how to disable power_save when you want to use BT.

lwfinger, 2023-08-06:
Please do two more things for me. Run the command 'lsusb' and post the output. In addition, run the command 'sudo dmesg > dmesg.txt' and attach dmesg.txt to this issue. Thanks.

MK73DS, 2023-08-07:
Here's the output of `lsusb` :

```
$ lsusb
Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 002: ID 04f2:b758 Chicony Electronics Co., Ltd Integrated Camera
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 0fce:320d Sony Ericsson Mobile Communications AB XQ-BC52
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 0bda:4853 Realtek Semiconductor Corp. Bluetooth Radio
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

And here is the dmesg : [dmesg.txt](https://github.com/lwfinger/rtw89/files/12282211/dmesg.txt)

lwfinger, 2023-08-07:
I think I know what your problem is - the system is using the wrong BT driver.

BT will use a generic driver unless your specific device is located in a "blacklist" contained within the driver. Having your USB ID in that badly-named table will blacklist the generic driver and load a specific driver. The IDs for your device are not in that table, thus the driver for Realtek BT devices is not loaded. That will be fixed, but it will take a while for the fix to propagate through the system.

As a temporary fix, 'git clone https://GitHub.com'lwfinger/rtw89-BT.git', 'cd rtw89-BT', make', and 'sudo make install'. That will install a BT driver that will recognize your device. After rebooting, rerun the power_save on/off part of the test. If that still fails, then run the script that dumps btc_info. Send me the output when on and off, and a new dmesg.txt.

MK73DS, 2023-08-07:
I can't compile the BT driver, I get this error. Maybe I'm missing a dependency but I checked the readme and it seems I have everything.
```
make -C /lib/modules/6.4.8-arch1-1/build M=/home/me/rtw89-BT modules
  CC [M]  /home/me/rtw89-BT/btusb.o
/home/me/rtw89-BT/btusb.c: In function ‘btusb_intel_diagnostics’:
/home/me/rtw89-BT/btusb.c:2436:26: error: ‘INTEL_TLV_TYPE_ID’ undeclared (first use in this function)
 2436 |         if (tlv->type != INTEL_TLV_TYPE_ID)
      |                          ^~~~~~~~~~~~~~~~~
/home/me/rtw89-BT/btusb.c:2436:26: note: each undeclared identifier is reported only once for each function it appears in
/home/me/rtw89-BT/btusb.c:2440:14: error: ‘INTEL_TLV_SYSTEM_EXCEPTION’ undeclared (first use in this function)
 2440 |         case INTEL_TLV_SYSTEM_EXCEPTION:
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/me/rtw89-BT/btusb.c:2441:14: error: ‘INTEL_TLV_FATAL_EXCEPTION’ undeclared (first use in this function)
 2441 |         case INTEL_TLV_FATAL_EXCEPTION:
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/me/rtw89-BT/btusb.c:2442:14: error: ‘INTEL_TLV_DEBUG_EXCEPTION’ undeclared (first use in this function); did you mean ‘INTEL_TLV_DEBUG_LOCK’?
 2442 |         case INTEL_TLV_DEBUG_EXCEPTION:
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~
      |              INTEL_TLV_DEBUG_LOCK
/home/me/rtw89-BT/btusb.c:2443:14: error: ‘INTEL_TLV_TEST_EXCEPTION’ undeclared (first use in this function)
 2443 |         case INTEL_TLV_TEST_EXCEPTION:
      |              ^~~~~~~~~~~~~~~~~~~~~~~~
/home/me/rtw89-BT/btusb.c: In function ‘btusb_setup_realtek’:
/home/me/rtw89-BT/btusb.c:2582:13: error: implicit declaration of function ‘btrealtek_test_flag’; did you mean ‘btintel_test_flag’? [-Werror=implicit-function-declaration]
 2582 |         if (btrealtek_test_flag(data->hdev, REALTEK_ALT6_CONTINUOUS_TX_CHIP))
      |             ^~~~~~~~~~~~~~~~~~~
      |             btintel_test_flag
/home/me/rtw89-BT/btusb.c:2582:45: error: ‘REALTEK_ALT6_CONTINUOUS_TX_CHIP’ undeclared (first use in this function)
 2582 |         if (btrealtek_test_flag(data->hdev, REALTEK_ALT6_CONTINUOUS_TX_CHIP))
      |                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/me/rtw89-BT/btusb.c: In function ‘btusb_probe’:
/home/me/rtw89-BT/btusb.c:4238:37: error: invalid application of ‘sizeof’ to incomplete type ‘struct btrealtek_data’
 4238 |                 priv_size += sizeof(struct btrealtek_data);
      |                                     ^~~~~~
/home/me/rtw89-BT/btusb.c:4321:23: error: too many arguments to function ‘btintel_configure_setup’
 4321 |                 err = btintel_configure_setup(hdev, btusb_driver.name);
      |                       ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/me/rtw89-BT/btusb.c:26:
/home/me/rtw89-BT/btintel.h:209:5: note: declared here
  209 | int btintel_configure_setup(struct hci_dev *hdev);
      |     ^~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:252: /home/me/rtw89-BT/btusb.o] Error 1
make[1]: *** [Makefile:2032: /home/me/rtw89-BT] Error 2
make: *** [Makefile:32: all] Error 2
```

If this is just a temporary fix before the ID of my device is in the blacklist table, then I'm okay with disabling the power saving mode when I use bluetooth for the time being.

lwfinger, 2023-08-07:
The code would build on kernel 6.5, but not with 6.4. I have fixed that issue.

I want you to test this fix to see if it is correct, or if we have a problem with the Realtek wifi and/or BT driver.

MK73DS, 2023-08-07:
Alright. I installed the bluetooth driver. However, the issue still persisted. Here's the result of the power_save on/off part.
<details>
<summary>After enabling power_save and audio glitches happened</summary>

```
=========================================
WL FW / BT FW           0.29.29.1 / NA
manual                  0
=========================================

 [bt_info]       : raw_data[49 00 c3 00 04 7f] (type:auto/cnt:63/same:0)
=========================================
========== [BTC COEX INFO (1)] ==========
 [coex_version]  : Coex:7.0.1(branch:19), WL_FW_coex:7.8.0(branch:29)(Match, desired:5.5.0), BT_FW_coex:7(Match, desired:5)
 [sub_module]    : WL_FW:0.29.29.1, BT_FW:0x1fc6b20f(patch)
 [hw_info]       : cv:1, rfe_type:0x1, ant_iso:10, ant_pg:2, 3rd_coex:0, dbcc:0, tx_num:2, rx_num:2
========== [WL Status] ==========
 [status]        : link_mode:9, rf_off:0, power_save:1, scan:N(band:1/phy_map:0x0), connecting:N, roam:N, 4way:N, init_ok:Y
 [port_0]        : role=1(phy-0), connect=2(client_cnt=-1), mode=3, center_ch=42, bw=2, mac_id=0, max_tx_time=0us, max_tx_retry=0
 [port_0]        : rssi=-63dBm(47), busy=0, dir=DL, tx[rate:260/busy_level:0], rx[rate:277/busy_level:0/drop:0]
========== [BT Status] ==========
 [status]        : enable:Y, btg:Y, connect:Y, igno_wl:N, mailbox_avl:Y, rfk_state:0x0
 [profile]       : profile:A2DP, multi-link:N, role:Master, ble-connect:N, CQDDR:N, A2DP_active:Y, PAN_active:N
 [link]          : rssi:-55dBm, tx_rate:3M,  acl_busy!! afh_map[ffff_ffff_ffff_ffff_fffe], wl_ch_map[en:0/ch:0/bw:32]
 [stat_cnt]      : retry:0, relink:0, rate_chg:3, reinit:0, reenable:1, role-switch:0, afh:0, inq_page:0(inq:0/page:0), igno_wl:1
 [A2DP]          : type:Legacy, bit-pool:63, flush-time:0, vid:0xa, Dev-name:0x2d4857, sut_pwr:0, golden-rx:0
 [bt_info]       : raw_data[49 00 c3 00 04 7f] (type:auto/cnt:63/same:0)
 [trx_req_cnt]   : Hi-rx = 199, Hi-tx = 0, Lo-rx = 536, Lo-tx = 537 (bt_polut_wl_tx = 0)(BG:0-0/INIT:0-0/LE:0-0)
========== [Mechanism Status (Auto)] ==========
 [status]        : type:shared, reason:NTFY_RADIO_STATE(), action:WL_OFF(), ant_path:5, run_cnt:181
 [dm_steps]      : -> UPDATE_BT_INFO      -> WL_5G          -> OFF_EQ0      -> UPDATE_BT_INFO      -> WL_5G          -> OFF_EQ0      
 [dm_steps]      : -> UPDATE_BT_INFO      -> WL_5G          -> OFF_EQ0      -> UPDATE_BT_INFO      -> WL_5G          -> OFF_EQ0      
 [dm_steps]      : -> UPDATE_BT_INFO      -> WL_5G          -> OFF_EQ0      -> NTFY_WL_STA         -> WL_5G          -> OFF_EQ0      
 [dm_steps]      : -> NTFY_WL_STA         -> WL_5G          -> OFF_EQ0      -> NTFY_RADIO_STATE    -> WL_OFF         -> OFF_BT       
 [dm_steps]      : -> NTFY_RADIO_STATE    -> WL_5G          -> OFF_EQ0      -> NTFY_RADIO_STATE    -> WL_OFF         -> OFF_BT       
 [dm_flag]       : wl_only:0, bt_only:0, igno_bt:1, free_run:0, wl_ps_ctrl:0, wl_mimo_ps:0, leak_ap:0, fw_offload:N
 [trx_ctrl]      : wl_rssi_lvl:3, para_lvl:0, wl_tx_pwr:orig, wl_rx_lvl:0, bt_tx_pwr_dec:0, bt_rx_lna:7(Ori-tbl), wl_btg_rx:0
 [dm_ctrl]       : wl_tx_limit[en:0/max_t:5280us/max_retry:31], bt_slot_reg:0-TU, bt_scan_rx_low_pri:0
 [tdma_policy]   : type:0, rx_flow_ctrl:0, tx_pause:0, wl_toggle_n:0, leak_n:0, ext_ctrl:0, policy_type:0
 [slot_list]     :   OFF[100/0xe5555555/0],   B2W[005/0xea5a5a5a/1],    W1[070/0xea5a5a5a/1],    W2[015/0xea5a5a5a/1],   W2B[015/0xea5a5a5a/1]
 [slot_list]     :    B1[250/0xe5555555/0],    B2[007/0xea5a5a5a/0],    B3[005/0xe5555555/0],    B4[050/0xe5555555/0],    LK[020/0xea5a5a5a/1]
 [slot_list]     :   BLK[500/0x55555555/0],   E2G[000/0xea5a5a5a/0],   E5G[000/0xffffffff/1],   EBT[000/0xe5555555/0], ENULL[000/0xaaaaaaaa/1]
 [slot_list]     :   WLK[250/0xea5a5a5a/0], W1FDD[050/0xffffffff/1], B1FDD[050/0xffffdfff/1]
========== [HW Status] ==========
 [scoreboard]    : WL->BT:0x00004083(cnt:72), BT->WL:0x4001051b(total:58, bt_update:11)
 [reg]           : 0_0xda24=0xffffffff, 0_0xda28=0xffffffff, 0_0xda2c=0xf0ffffff, 0_0xda30=0x00000008, 0_0xda4c=0x0000000d, 0_0xda10=0x00000100
 [reg]           : 0_0xda20=0x00000002, 0_0xda34=0x00000100, 0_0xcef4=0x00000000, 0_0x8424=0x00000000, 0_0xd200=0x00000000, 0_0xd220=0x00000000
 [reg]           : 1_0x0980=0x10182250, 1_0x4738=0x8325c500, 1_0x4688=0x1a10ff04, 1_0x4694=0x04fa2a2a
========== [Statistics] ==========
 [summary]       : h2c_cnt=242(fail:2), c2h_cnt=0
 [notify_cnt]    : total=201, show_coex_info=84, power_on=0, init_coex=1, power_off=0, radio_state=82, role_info=2, wl_rfk=5, wl_sta=15
 [notify_cnt]    : scan_start=1, scan_finish=1, switch_band=2, special_pkt=5, timer=3, control=0, customerize=0lun. 07 août 2023 22:31:52 CEST
```

</details>

<details>
<summary>When power_save was disabled and everything was fine</summary>

```
=========================================
WL FW / BT FW           0.29.29.1 / NA
manual                  0
=========================================

 [bt_info]       : raw_data[49 00 c3 00 04 7f] (type:auto/cnt:63/same:0)
=========================================
========== [BTC COEX INFO (1)] ==========
 [coex_version]  : Coex:7.0.1(branch:19), WL_FW_coex:7.8.0(branch:29)(Match, desired:5.5.0), BT_FW_coex:7(Match, desired:5)
 [sub_module]    : WL_FW:0.29.29.1, BT_FW:0x1fc6b20f(patch)
[... 20 lines cut ...]
 [dm_flag]       : wl_only:0, bt_only:0, igno_bt:0, free_run:0, wl_ps_ctrl:0, wl_mimo_ps:0, leak_ap:0, fw_offload:N
[... cut ...]
 [mailbox]       : send_ok:62, send_fail:0, recv:1147, A2DP_empty:1034(stop:0, tx:4251, ack:4006, nack:31)
 [RFK/LPS]       : wl_rfk[req:5/go:5/reject:0/tout:0], bt_rfk[req:0], AOAC[RF_on:388/RF_off:388]
 [notify_cnt]    : total=190, show_coex_info=78, power_on=0, init_coex=1, power_off=0, radio_state=79, role_info=2, wl_rfk=5, wl_sta=13
 [notify_cnt]    : scan_start=1, scan_finish=1, switch_band=2, special_pkt=5, timer=3, control=0, customerize=0lun. 07 août 2023 22:31:46 CEST
```

</details>

And finally, the dmesg : [dmesg.txt](https://github.com/lwfinger/rtw89/files/12285085/dmesg.txt)

lwfinger, 2023-08-08:
Realtek would like to know the make and model of your BT headset.

MK73DS, 2023-08-08:
It's a Sony WH-1000XM3. It has no issue on any other device I used it on (multiple smartphones, Nintendo Switch, another laptop) so I would be surprised if it is defective.

Another finding I noticed. I'm using KDE Plasma and I have an app on my smartphone, KDE Connect, that allows me to remotely control the mouse cursor via wifi. When wifi power_save is on and without any device connected via bluetooth, the motion of the cursor is very choppy. When power_save is off, it becomes smooth again. So it seems like wifi reception is also affected, not only bluetooth (I guess it makes sense since they share the same antenna).

samuelskottenborg, 2023-08-08:
If it's any help, I have the exact same problem with rtw89_8852be as described earlier in this thread on my Sony WH-1000XM3 and AirPods Pro.

lwfinger, 2023-08-08:
@samuelskottenborg - That does help. What does 'lsusb | grep Bluetooth' show?

samuelskottenborg, 2023-08-08:
Great.

It shows 
Bus 003 Device 004: ID 0bda:4853 Realtek Semiconductor Corp. Bluetooth Radio

I'm naturally interested in getting this solved too, so I'm happy to provide more if you like :)

lwfinger, 2023-08-08:
Could you please post the output from 'sudo cat /sys/kernel/debug/usb/devices' that pertains to the BT device? I need that to submit a patch to add the device to the kernel's tables. Thanks.

samuelskottenborg, 2023-08-08:
T:  Bus=03 Lev=01 Prnt=01 Port=09 Cnt=03 Dev#=  4 Spd=12   MxCh= 0
D:  Ver= 1.00 Cls=e0(wlcon) Sub=01 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=0bda ProdID=4853 Rev= 0.00
S:  Manufacturer=Realtek
S:  Product=Bluetooth Radio
S:  SerialNumber=00e04c000001
C:* #Ifs= 2 Cfg#= 1 Atr=e0 MxPwr=500mA
I:* If#= 0 Alt= 0 #EPs= 3 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=81(I) Atr=03(Int.) MxPS=  16 Ivl=1ms
E:  Ad=02(O) Atr=02(Bulk) MxPS=  64 Ivl=0ms
E:  Ad=82(I) Atr=02(Bulk) MxPS=  64 Ivl=0ms
I:* If#= 1 Alt= 0 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=   0 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=   0 Ivl=1ms
I:  If#= 1 Alt= 1 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=   9 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=   9 Ivl=1ms
I:  If#= 1 Alt= 2 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  17 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  17 Ivl=1ms
I:  If#= 1 Alt= 3 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  25 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  25 Ivl=1ms
I:  If#= 1 Alt= 4 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  33 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  33 Ivl=1ms
I:  If#= 1 Alt= 5 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  49 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  49 Ivl=1ms

lwfinger, 2023-08-08:
Thanks. The patch adding your BT device to the kernel tables has been sent upstream.

lwfinger, 2023-08-09:
@QuakeString - What does 'lsusb | grep Bluetooth' show for your device?

QuakeString, 2023-08-10:
Here is the output
```
lsusb | grep Bluetooth
Bus 003 Device 005: ID 0cb8:c559 Opticis Co., Ltd Bluetooth Radio
```

QuakeString, 2023-08-10:
But for me `disable_ps_mode=y` did not work.

lwfinger, 2023-08-10:
Are you using the in-kernel driver, or the one from this repo? If you have a question, run the command 'lsmod | grep rtw' and make certain that your options command matches the one that is loaded.

If you are using this repo, did you unload the driver before you ran the modprobe command you gave above. One other potential problem is that command would only apply until you rebooted, or reloaded the driver.

Run the command 'sudo dmesg | grep renamed', and use the new name for wlan0 in the command
'sudo iw wlan0 power_save off'. Does that cure the problem? If it does, then your modprobe command was not in effect.

QuakeString, 2023-08-10:
my system:
```
Operating System: Manjaro Linux 
KDE Plasma Version: 5.27.6
KDE Frameworks Version: 5.108.0
Qt Version: 5.15.10
Kernel Version: 6.1.41-1-MANJARO (64-bit)
Graphics Platform: X11
Processors: 12 × 12th Gen Intel® Core™ i5-12450H
Memory: 15.3 GiB of RAM
Graphics Processor: Mesa Intel® Graphics
Manufacturer: TIMI
Product Name: Xiaomi NoteBook Pro
```
This kerner has no drive of this wifi device.
so I installed it saperately

```
lsmod | grep rtw                                                                                                                  ✔ 
rtw_8852be             16384  0
rtw_8852b             352256  1 rtw_8852be
rtw89pci               81920  1 rtw_8852be
rtw89core             585728  2 rtw89pci,rtw_8852b
mac80211             1310720  2 rtw89pci,rtw89core
cfg80211             1138688  3 rtw89core,mac80211,rtw_8852b
```

```
$sudo modprobe -rv rtw_8852be                                                                            
[sudo] password for mozammel: 
rmmod rtw_8852be
rmmod rtw89pci
rmmod rtw_8852b
$ sudo modprobe -rv rtw89core                                                                            
rmmod rtw89core
rmmod mac80211
$ sudo modprobe -v rtw89core disable_ps_mode=y                                                            
insmod /lib/modules/6.1.41-1-MANJARO/kernel/net/mac80211/mac80211.ko.zst 
insmod /lib/modules/6.1.41-1-MANJARO/updates/dkms/rtw89core.ko.zst disable_ps_mode=y
$ sudo modprobe -v rtw_8852be                                                                             
insmod /lib/modules/6.1.41-1-MANJARO/updates/dkms/rtw89pci.ko.zst 
insmod /lib/modules/6.1.41-1-MANJARO/updates/dkms/rtw_8852b.ko.zst 
insmod /lib/modules/6.1.41-1-MANJARO/updates/dkms/rtw_8852be.ko.zst 
```

I confirm now after following above step Bluetooth worked properly.

lwfinger, 2023-08-10:
@MK73DS: It seems that I dod not completely understand the Realtek requests. What you should do is start the script running with power_save turned on. After a couple of cycles, note the time, turn power_save off, and let the script run for 10 sec. Copy the entire output into a file named power_save.txt, and attach that file here. Appparently, it takes some time for the changes to propagate, and they want to see the whole process.

AnakTeka, 2023-08-12:
>@MK73DS: It seems that I dod not completely understand the Realtek requests. What you should do is start the script running with power_save turned on. After a couple of cycles, note the time, turn power_save off, and let the script run for 10 sec. Copy the entire output into a file named power_save.txt, and attach that file here. Appparently, it takes some time for the changes to propagate, and they want to see the whole process. Also indicate which interval has power_save turned off.

I have same issue where the BT headset sound is choppy when connected to a WI-FI, I created a script with instructions that you mention

```
#!/bin/sh

> power_save.txt

iw wlan0 set power_save on

for i in {1..5}; do
    cat /sys/kernel/debug/ieee80211/phy0/rtw89/btc_info >> power_save.txt
    echo "---- date now $(date) ----" >> power_save.txt
    sleep 2
done

echo "---- Turning OFF power_save at $(date) ----" >> power_save.txt
sudo iw wlan0 set power_save off

for i in {1..5}; do
    cat /sys/kernel/debug/ieee80211/phy0/rtw89/btc_
info >> power_save.txt
    echo "---- date now $(date) ----" >> power_save.txt
    sleep 2
done

echo "End of script" >> power_save.txt
```

Let me know if I misunderstood your instructions or if there's anything else you'd like me to adjust in the script. This is the generated power_save.txt

[power_save.txt](https://github.com/lwfinger/rtw89/files/12328139/power_save.txt)

4javier, 2023-09-16:
Hi.
I have the exact same issue on an HP 470 G9
My bluetooth headphones works flawlessly on the same laptop on Win and on an old Thinkpad.
Here on Arch on the HP if I don't suppress powersaving audio is stuttering, despite being connected to a 5GHz network.
This is the usb identifier for my chip `0bda:b85c`, I already checked inside the infamous table and it's not listed.
Would you mind asking Realtek devs to add it, please?
Just to assure you the issue is the same here's the kernel driver in use for the NIC
```

QuakeString, 2023-09-17:
> Hi. I have the exact same issue on an HP 470 G9 My bluetooth headphones works flawlessly on the same laptop on Win and on an old Thinkpad. Here on Arch on the HP if I don't suppress powersaving audio is stuttering, despite being connected to a 5GHz network. This is the usb identifier for my chip `0bda:b85c`, I already checked inside the infamous table and it's not listed. Would you mind asking Realtek devs to add it, please? Just to assure you the issue is the same here's the kernel driver in use for the NIC
> 
> ```
> 03:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device b852
> 	Subsystem: Hewlett-Packard Company Device 88e3
> 	Kernel driver in use: rtw89_8852be
> 	Kernel modules: rtw89_8852be, wl
> ```
> 
> Thanks.

I put these on a .sh file and run it on startup as cronjob. 
`@reboot /path-to-your-script/_realtek_WiFi_Bluetooth_tempFix.sh
`
```
sleep 10
sudo modprobe -rv rtw89_8852be
sudo modprobe -rv rtw89_core
sudo modprobe -v rtw89_core disable_ps_mode=y
sudo modprobe -v rtw89_8852be
```
it does fix my issue.

AnakTeka, 2023-09-17:
> > Hi. I have the exact same issue on an HP 470 G9 My bluetooth headphones works flawlessly on the same laptop on Win and on an old Thinkpad. Here on Arch on the HP if I don't suppress powersaving audio is stuttering, despite being connected to a 5GHz network. This is the usb identifier for my chip `0bda:b85c`, I already checked inside the infamous table and it's not listed. Would you mind asking Realtek devs to add it, please? Just to assure you the issue is the same here's the kernel driver in use for the NIC
> > ```
> > 03:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device b852
> > 	Subsystem: Hewlett-Packard Company Device 88e3
> > 	Kernel driver in use: rtw89_8852be
> > 	Kernel modules: rtw89_8852be, wl
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Thanks.
> 
> I put these on a .sh file and run it on startup as cronjob.
> 
> ```
> $ sudo crontab -e
> @reboot /path-to-your-script/_realtek_WiFi_Bluetooth_tempFix.sh
> ```
> 
> ```
> sleep 10
> sudo modprobe -rv rtw89_8852be
> sudo modprobe -rv rtw89_core
> sudo modprobe -v rtw89_core disable_ps_mode=y
> sudo modprobe -v rtw89_8852be
> ```
> 
> It does fix my issue.

Instead of a cron script where you unload and reunload the modules, you should instead put the module options in `/etc/modprobe.d/` as it will be easier to maintain any kernel modules options in a single place.

for example mine on 

`/etc/modprobe.d/rtw89_core.conf` 

```
options rtw89_core disable_ps_mode=y
```

and `/etc/modprobe.d/rtw89core.conf`

```
options rtw89core disable_ps_mode=y
```

And apparently its there on the README.md https://github.com/lwfinger/rtw89#4-option-configuration-

4javier, 2023-09-17:
Thanks guys, I know how to do that, and I add the simplest way would be adding that module param to kernel commandline in your bootloader config.
But even if I currently don't need powersaving too much, these kind of workarounds are those things that got put there and forgotten, until in a year they will bite back for some reason when you can't even recall why you put them there.
That's why I asked for the id to be added to upstream driver whitelist.
Thanks for your interest, you've been both really kind.

4javier, 2023-09-28:
In the end I cloned https://github.com/lwfinger/rtw89-BT , patched btusb.c adding my device's id and packaged it in dkms.
It looks like it gets recognized, but I still got interference that goes away when I disable wifi powersaving.
```
set 29 01:10:25 hp-work kernel: btrtl: loading out-of-tree module taints kernel.
set 29 01:10:25 hp-work kernel: btrtl: module verification failed: signature and/or required key missing - tainting kernel
set 29 01:10:25 hp-work kernel: usbcore: registered new interface driver uvcvideo
set 29 01:10:25 hp-work kernel: ************** Entering btusb_probe, driver_info 0x0
set 29 01:10:25 hp-work kernel: **************** output of usb_match_id 00000000add3c5eb
set 29 01:10:25 hp-work kernel: ************** driver_info 0x210000
set 29 01:10:25 hp-work kernel: ***************** configuring Realtek callbacks
set 29 01:10:25 hp-work kernel: usbcore: registered new interface driver btusb
set 29 01:10:25 hp-work kernel: *************** Entering btrtl_setup_realtek
set 29 01:10:25 hp-work kernel: i915 0000:00:02.0: vgaarb: deactivate vga console
set 29 01:10:25 hp-work kernel: i915 0000:00:02.0: [drm] Using Transparent Hugepages
set 29 01:10:25 hp-work kernel: Bluetooth: hci0: RTL: examining hci_ver=0b hci_rev=000b lmp_ver=0b lmp_subver=8852
set 29 01:10:25 hp-work kernel: Bluetooth: hci0: RTL: rom_version status=0 version=1
set 29 01:10:25 hp-work kernel: Bluetooth: hci0: RTL: loading rtl_bt/rtl8852bu_fw.bin
set 29 01:10:25 hp-work systemd[1]: Finished Create Volatile Files and Directories.
set 29 01:10:25 hp-work kernel: mousedev: PS/2 mouse device common for all mice
set 29 01:10:25 hp-work kernel: Bluetooth: hci0: RTL: loading rtl_bt/rtl8852bu_config.bin
set 29 01:10:25 hp-work kernel: Bluetooth: hci0: RTL: cfg_sz 6, total sz 58003
set 29 01:10:25 hp-work kernel: i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
set 29 01:10:25 hp-work kernel: mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_ops [i915])
set 29 01:10:25 hp-work kernel: i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc.bin (v2.20)
set 29 01:10:25 hp-work kernel: input: SYNA32C3:00 06CB:CE17 Mouse as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-19/i2c-SYNA32C3:00/0018:06CB:CE17.0004/input/input13
set 29 01:10:25 hp-work kernel: input: SYNA32C3:00 06CB:CE17 Touchpad as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-19/i2c-SYNA32C3:00/0018:06CB:CE17.0004/input/input14
set 29 01:10:25 hp-work kernel: hid-multitouch 0018:06CB:CE17.0004: input,hidraw3: I2C HID v1.00 Mouse [SYNA32C3:00 06CB:CE17] on i2c-SYNA32C3:00
set 29 01:10:25 hp-work kernel: i915 0000:00:02.0: [drm] GT0: GuC firmware i915/adlp_guc_70.bin version 70.5.1
set 29 01:10:25 hp-work kernel: i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
set 29 01:10:25 hp-work systemd[1]: Starting Load Kernel Module dm_mod...
set 29 01:10:25 hp-work kernel: i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads
set 29 01:10:25 hp-work kernel: i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
set 29 01:10:25 hp-work kernel: i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
set 29 01:10:25 hp-work kernel: i915 0000:00:02.0: [drm] GT0: GUC: RC enabled
set 29 01:10:25 hp-work systemd[1]: Starting Load Kernel Module loop...
set 29 01:10:25 hp-work systemd[1]: Set Up Additional Binary Formats was skipped because no trigger condition checks were met.
set 29 01:10:25 hp-work systemd[1]: Update Boot Loader Random Seed was skipped because no trigger condition checks were met.
set 29 01:10:25 hp-work systemd[1]: First Boot Wizard was skipped because of an unmet condition check (ConditionFirstBoot=yes).
set 29 01:10:25 hp-work systemd[1]: First Boot Complete was skipped because of an unmet condition check (ConditionFirstBoot=yes).
set 29 01:10:25 hp-work systemd[1]: Rebuild Hardware Database was skipped because no trigger condition checks were met.
set 29 01:10:25 hp-work systemd[1]: Starting Rebuild Journal Catalog...
set 29 01:10:25 hp-work systemd[1]: Commit a transient machine-id on disk was skipped because of an unmet condition check (ConditionPathIsMountPoint=/etc/machine-id).
set 29 01:10:25 hp-work systemd[1]: TPM2 PCR Machine ID Measurement was skipped because of an unmet condition check (ConditionPathExists=/sys/firmware/efi/efivars/StubPcrKernelImage-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f).
set 29 01:10:25 hp-work kernel: mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
set 29 01:10:25 hp-work kernel: i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
set 29 01:10:25 hp-work systemd[1]: Starting Load/Save RF Kill Switch Status...
set 29 01:10:25 hp-work systemd[1]: Starting Network Time Synchronization...
set 29 01:10:25 hp-work systemd[1]: Starting Record System Boot/Shutdown in UTMP...
set 29 01:10:25 hp-work systemd[1]: Finished Rebuild Dynamic Linker Cache.
set 29 01:10:25 hp-work systemd[1]: Started Load/Save RF Kill Switch Status.
set 29 01:10:25 hp-work systemd[1]: modprobe@dm_mod.service: Deactivated successfully.
set 29 01:10:25 hp-work kernel: rtw89_8852be 0000:03:00.0: loaded firmware rtw89/rtw8852b_fw-1.bin
set 29 01:10:25 hp-work kernel: rtw89_8852be 0000:03:00.0: enabling device (0000 -> 0003)
[... cut ...]
> 						     BTUSB_WIDEBAND_SPEECH },
> 	// 4javier
> 	{ USB_DEVICE(0x0bda, 0xb85c), .driver_info = BTUSB_REALTEK |
> 						     BTUSB_WIDEBAND_SPEECH },
> 
> 	/* Realtek Bluetooth devices */
> 	{ USB_VENDOR_AND_INTERFACE_INFO(0x0bda, 0xe0, 0x01, 0x01),
> 	  .driver_info = BTUSB_REALTEK },

And  I wonder if this is the right fw it's loading

> set 29 01:10:25 hp-work kernel: Bluetooth: hci0: RTL: rom_version status=0 version=1
> set 29 01:10:25 hp-work kernel: Bluetooth: hci0: RTL: loading rtl_bt/rtl8852bu_fw.bin

lwfinger, 2023-09-29:
Do not worry about that match ID. Those ***** prints weree added to keep track of the progress of probing the BT hardware. Yours look normal.

The latest firmware for the 8852b BT devices have the following md5sum output:
3eb96095411a33c339b3da2a7e9d94f3  /lib/firmware/rtl_bt/rtl8822b_config.bin
861d9b6792410223e5b17cb5d5627b02  /lib/firmware/rtl_bt/rtl8822b_config.bin.xz
547896dc612f5e4a704e4bff8dfb908b  /lib/firmware/rtl_bt/rtl8822b_fw.bin
30b6f8a757a9d1f2cdbfc0712bba61c3  /lib/firmware/rtl_bt/rtl8822b_fw.bin.xz

You are missing the config file. My log shows
[153365.739130] Bluetooth: hci0: RTL: loading rtl_bt/rtl8852bu_fw.bin
[153365.739492] Bluetooth: hci0: RTL: loading rtl_bt/rtl8852bu_config.bin
Your distro should be providing that file.

I will submit your USB ID to the mainline kernel, but I need one other piece of info. Please run 
sudo cat /sys/kernel/debug/usb/devices
and post the section here that matches your BT device. Thanks.

4javier, 2023-09-29:
Thanks for your interest

> You are missing the config file. My log shows
> [153365.739130] Bluetooth: hci0: RTL: loading rtl_bt/rtl8852bu_fw.bin
> [153365.739492] Bluetooth: hci0: RTL: loading rtl_bt/rtl8852bu_config.bin
> Your distro should be providing that file.
> 

Mine shows that too, lt's just split by a couple uninteresting lines

> set 29 01:10:25 hp-work kernel: Bluetooth: hci0: RTL: loading rtl_bt/rtl8852bu_fw.bin
> set 29 01:10:25 hp-work systemd[1]: Finished Create Volatile Files and Directories.
> set 29 01:10:25 hp-work [...]

This is the cat you asked for
```
T:  Bus=01 Lev=01 Prnt=01 Port=09 Cnt=05 Dev#=  6 Spd=12   MxCh= 0
D:  Ver= 1.00 Cls=e0(wlcon) Sub=01 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=0bda ProdID=b85c Rev= 0.00
S:  Manufacturer=Realtek
S:  Product=Bluetooth Radio
S:  SerialNumber=00e04c000001
C:* #Ifs= 2 Cfg#= 1 Atr=e0 MxPwr=500mA
I:* If#= 0 Alt= 0 #EPs= 3 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=81(I) Atr=03(Int.) MxPS=  16 Ivl=1ms
E:  Ad=02(O) Atr=02(Bulk) MxPS=  64 Ivl=0ms
E:  Ad=82(I) Atr=02(Bulk) MxPS=  64 Ivl=0ms
I:* If#= 1 Alt= 0 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=   0 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=   0 Ivl=1ms
I:  If#= 1 Alt= 1 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=   9 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=   9 Ivl=1ms
I:  If#= 1 Alt= 2 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  17 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  17 Ivl=1ms
I:  If#= 1 Alt= 3 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  25 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  25 Ivl=1ms
I:  If#= 1 Alt= 4 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  33 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  33 Ivl=1ms
I:  If#= 1 Alt= 5 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  49 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  49 Ivl=1ms
```
Unfortunately, if as you say the driver is already correctly loaded, adding the usbid in the mainline will not solve the issue, but at least I could file a bugreport to them

Thanks  again

lwfinger, 2023-09-29:
Your device has been in the kernel code since v6.4. No need to change the kernel code.

closed 2023-09-29 by lwfinger

4javier, 2023-09-30:
Can't find any reference to `b85c` anywhere in btusb.c whitelist in mainline.
Anyway, do you have any suggestion on how to correctly file a bugreport to ralink?
My device still suffers of the interference when wifi powersaving is on.
It looks like the kind of problem they solved on Windows with `1.10.1061.3001 Rev.S` update of the driver in April.