lwfinger/rtw89 #263: Wi-Fi works for few seconds or minutes and then disconnects indefinitely (RTL8852BE)

mgsanava, 2023-08-10:
Hello Larry. Laptop's processor is AMD Ryzen™ 5-5625U. And laptop is HP Pavilion 14-ec1019au if it helps.
From Windows I could see wireless hardware details as Realtek RTL8852BE Wi-Fi 6 802.11ax PCIe Adapter.

I first tried Ubuntu 23.04 in which Wifi was working for a minute or two and then disconnects indefinitely (Kernel 6.2).
Faced system crashes in Ubuntu and hence I moved to Zorin OS (Kernel 5.15) and installed drivers from your repo.
Could not find success with signed driver so I installed a normal driver (SB turned off) using the instructions exactly as mentioned in the README. Everything goes fine and Wifi hardware was also detected. BUT in a minute or two it disconnects and won't come back. Same problem when using `"sudo apt install rtw89-dkms"` from a `"ppa:kelebek333/kablosuz"`.

Note: I tried a distro with 6.4 kernel hoping that Wifi driver will work flawlessly. Same behaviour Wifi comes and goes down indefinitely. Some other info that might help:

```
sudo lshw -c network
[sudo] password for mani: 
  *-generic                 
       description: Wireless interface
       product: Realtek Semiconductor Co., Ltd.
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0
       bus info: pci@0000:01:00.0
       logical name: wlo1
       version: ff
       serial: 10:68:38:c3:2a:2c
       width: 32 bits
       clock: 66MHz
       capabilities: bus_master vga_palette cap_list ethernet physical wireless
       configuration: broadcast=yes driver=rtw89_8852be driverversion=5.19.0-32-generic firmware=N/A latency=255 link=no maxlatency=255 mingnt=255 multicast=yes wireless=IEEE 802.11
       resources: irq:81 ioport:f000(size=256) memory:fcf00000-fcffffff
```
**Note: Above commands prints out IP also if connected to internet but by the time I issued the command Wifi is gone.**

```
inxi -Na
Network:
  Device-1: Realtek driver: rtw89_8852be v: N/A modules: rtw_8852be pcie:
    speed: Unknown lanes: 63 link-max: gen: 6 speed: 64 GT/s port: N/A
    bus-ID: 01:00.0 chip-ID: 10ec:b852 class-ID: 0280
```
Please let me know if any other information is required.

chm46e, 2023-08-10:
Experiencing the same issue here with rtl8852be.
This still occurs with the currently latest kernel release 6.5.0-rc5.
dmesg reports: https://pastebin.com/7LzqJRUC [full_output](https://github.com/lwfinger/rtw89/files/12312434/rtwerr.txt)

This should probably be taken to the maintainer and/or mailing list:
Ping-Ke Shih <pkshih@realtek.com>
linux-wireless@vger.kernel.org

mgsanava, 2023-08-10:
@chm46e - are you asking me to report this to maintainers or have you done it already?
I can write an email but not sure how to do it via mailing list.

chm46e, 2023-08-10:
It's better to be sent to the maintainer imo, less chance to be ignored.
Haven't personally sent anything yet.
Pretty sure you need to subscribe to linux-wireless to do that. http://vger.kernel.org/majordomo-info.html
Would be nice to have it in 6.6.

lwfinger, 2023-08-10:
I act as triage for the maintainer. You can go to him directly if you want, but he and I would prefer to have tested all options before you do.

The first thing to test is if the problem is related to the PCIe bus in your machine. Note that no one has ever reported this problem before, thus I am suspicious, particularly when the issue seems to suddenly be a problem in communicating with the device. Vendors are shipping horrible BIOS code that causes all sorts of problems.

Create a file (as root) named /lib/modules.d/70-rtw8852be.conf and add the following:
options rtw89_pci disble_aspm_l1=y disble_aspm_l1ss=y
options rtw89pci disble_aspm_l1=y disble_aspm_l1ss=y

Then reboot. Note, there are two versions - one for the in-kernel driver, and one for the driver from this repo.

If the wifi still has the same problem, then run the command 'sudo dmesg > dmesg.txt' and attach the resulting dmesg.txt to this issue.

mgsanava, 2023-08-11:
"Vendors are shipping horrible BIOS code that causes all sorts of problems." - True and we need to live with it until businessmen and businesses exists.

I created /lib/modules.d/70-rtw8852be.conf and added those 2 lines. Rebooted and no change.

Attached the dmesg logs:

1. Look into dmesg_this_repo.txt for log with lwfinger driver
2. Hrex39 is another repo on GitHub - Wifi is working with it although Bluetooth is wonky

[dmesg5_hrex39.txt](https://github.com/lwfinger/rtw89/files/12319159/dmesg5_hrex39.txt)
[dmesg_this_repo.txt](https://github.com/lwfinger/rtw89/files/12319160/dmesg_this_repo.txt)

lwfinger, 2023-08-11:
I know the Hrex39 repo. It contains the vendor driver, which is really awful. One main difference, besides crappy code,  is that it uses really old firmware. If you do a 'ls /lib/firmware/rtw89/', you will see two different files for rtw8852b. On my system I get

rtw8851b_fw.bin.xz  rtw8852a_fw.bin.xz  rtw8852b_fw-1.bin.xz  rtw8852b_fw.bin.xz  rtw8852c_fw.bin.xz

Your system is using rtw8852b_fw-1.bin or its compressed version as shown by in the dmesg output:

[    3.611449] rtw89_8852be 0000:01:00.0: loaded firmware rtw89/rtw8852b_fw-1.bin
[    3.611541] rtw89_8852be 0000:01:00.0: enabling device (0000 -> 0003)
[    3.613691] rtw89_8852be 0000:01:00.0: Firmware version 0.29.29.1, cmd version 0, type 5
[    3.613695] rtw89_8852be 0000:01:00.0: Firmware version 0.29.29.1, cmd version 0, type 3

Run the command,
sudo mv ls /lib/firmware/rtw89/rtw8852b_fw-1.bin* .

That will save the newer firmware file in your current directory, and force the driver to use the older version (without the -1). Unload and reload the driver, or reboot. Does that help?

mgsanava, 2023-08-13:
In my system ls `/lib/firmware/rtw89/` returned `rtw8851b_fw.bin  rtw8852a_fw.bin  rtw8852b_fw.bin  rtw8852c_fw.bin rtw8852b_fw-1.bin`
There are no .xz files at all.
I anyway did what you suggested -> `sudo mv /lib/firmware/rtw89/rtw8852b_fw-1.bin .` (corrected as per my system)
### And voila, Wifi works now after a reboot!
**But there is a caveat.**
I left the laptop for couple hours in lid closed state and then when I opened the lid Wifi is gone. Tested only once though.
Of course a reboot helped me getting back Wifi. It's okay though. I'm not sure if we can close the issue since there is a lot of workaround and also considering that Wifi is not working out of the box even though Wifi driver is present in latest kernels. What can I submit to investigate further (or) Should we just blame HP for this?

Please suggest...

h3fang, 2023-08-13:
Hello,  I am using ASUS B650m Wifi Plus mother board and rtw89 from latest linux-firmware. 

With default settings, it just refuses to connect to my 5G WPA3 WIFI network.

Setting
```
options rtw89_pci disable_clkreq=y disable_aspm_l1=y disable_aspm_l1ss=y
options rtw89_core disable_ps_mode=y
```
makes it connect 1~3 minutes after a fresh reboot.

Removing `/usr/lib/firmware/rtw89/rtw8852b_fw-1.bin.zst` and reboot makes it connect immediately.

lwfinger, 2023-08-14:
Before I push this problem upstream, I would like you to try one more thing.

Today, I upgraded this repo to match what will be in kernel 6.6. Please do a 'git pull', 'make', and 'sudo make install'. Then restore the rtw8852b_fw-1 firmware in whatever form you have, and retry the wifi. Let me know the results.  Thanks.

mgsanava, 2023-08-14:
Updated the repo & installed.
I restored rtw8852b_fw-1.bin in its previous location. 
Then rebooted and **Wifi still seems to work fine**.
I streamed 4K videos to test. No problem found.
Btw, my kernel is at 6.
Thanks for the help.
Can we expect the change from Kernel 6.6 onwards?

lwfinger, 2023-08-14:
Yes. Until then, you will need to use this repo.

mgsanava, 2023-08-15:
Yesterday, I was updating Plymouth screen with `sudo update-initramfs -u`, out of the blue Wifi disconnected.
Rebooted the laptop and it kept disconnecting inconsistently. It was late already hence I left it like that.
Today it got disconnected twice. So, I moved that rtw8852b_fw-1.bin from its location again to check and rebooted.
Wifi still got disconnected. Rebooted again without making any changes. Now it seems okay.
I'm typing all this while Wifi is connected. I'm not sure of the behavior but the log when Wifi is gone is something like:
```
[   93.218567] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   93.218600] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   93.218633] rtw89_8852be 0000:01:00.0: write rf busy swsi
[   93.218665] rtw89_8852be 0000:01:00.0: write rf busy swsi
```
Somehow I feel rtw8852b_fw-1.bin is causing problem during laptop sleep/suspend/resume states.
In line 1143 there is a reference to rtw8852b_fw-1.bin in today's log. [dmesg.txt](https://github.com/lwfinger/rtw89/files/12341147/dmesg.txt)

Note: I have added "options rtw89_core disable_ps_mode=y" in /lib/modules.d/70-rtw8852be.conf as h3fang suggested just to check if that solves the problem. Still testing it.

lwfinger, 2023-08-15:
The reference to rtw8852b_fw-1.bin is merely stating that it tried to load it first, but it was not available.

Are you up tp date with the code? The command 'git log' should have commit a90a30a0eddca as the top entry. If not, then all your testing shows nothing.

mgsanava, 2023-08-16:
Sure - I'm up to date with the code.
```
gmk@zorinCore:~/drivers/rtw89$ git log
commit a90a30a0eddca05ccb2496b106daca4a76b89b11 (HEAD -> main, origin/main, origin/HEAD)
Author: Larry Finger <Larry.Finger@lwfinger.net>
Date:   Sun Aug 13 20:51:43 2023 -0500

    rtw89: fix build errors for kernels older than 6.5
    
    Signed-off-by: Larry Finger <Larry.Finger@lwfinger.net>

commit 0437ffbfb9093a263817e56db7f1378ef508880a
Author: Larry Finger <Larry.Finger@lwfinger.net>
Date:   Sun Aug 13 20:39:17 2023 -0500

    rtw89: Fix some merge errors
    
    This brings this repo up to the wireless-next repo as on August 13, 2023.
    
    Caution: Despite my best efforts, you will not be able to bisect through
    the previous commits.
    
    Signed-off-by: Larry Finger <Larry.Finger@lwfinger.net>

commit 34a5e6719781bf81550345c2577e968279a5d389
:...skipping...

```
**Something is better than nothing.** So, can we close the ticket?

closed 2023-08-16 by lwfinger

mgsanava, 2023-08-17:
Just making some notes here -
*Today Wifi was disconnected when I simply plugged in the charger. And I think Wifi disconnection problem is specific to this laptop model only (HP Pavilion 14-ec1019au).

mgsanava, 2023-08-17:
> The power supply in your laptop has some kind of fault

Very very unlikely. This is a brand new laptop with ZERO problems if I'm on Windows.
> and the system might have all kinds of other problems

Yeah, same as above, Windows is fine again in all aspects. So, this might be related to Linux OS as a whole not the driver.

lwfinger, 2023-08-17:
The situation is that HP has modified Windows code and drivers to handle the quirks of their product. Windows is their market. They spend zero time doing the same for Linux. Yes, there may be quirks that need to be applied to Linux, but no one has the hardware, the expertise, and the inclination to solve such problems.

Personally, I would never buy an HP or Lenovo laptop. My experience is that Dell or Toshiba are much better engineered. They "just work".

mgsanava, 2023-08-18:
>Should we just blame HP for this?

Yeah, like I said before, it's a BIG yes.
I bought this laptop only because I got a good deal on it.
Dell is not on my list at all because I'm not aware that they are engineered better. Toshiba's are not available here.

All I have to remember is not to do any payment activities while on Linux because God knows when Wifi will go off.

Another quirk is that this F*g HP laptop cannot detect detect my mobile via USB, both in Linux and Windows. Due to which I cannot tether via USB. I always borrow somebody else's phone when Wifi is gone. I never understood why all phones work except my own [mobile](https://www.gsmarena.com/xiaomi_11_lite_5g_ne-11101.php).

mgsanava, 2023-08-31:
Wifi is okay from the past 2 weeks. No disconnection issue so far. 
Wifi Signal strength is fair. Bluetooth is working. 
I held the Kernel version at 5.15.x so that I don't have to recompile the driver often.
This is good to go into Kernel 6.6 as per previous comments.

System: Zorin OS 16 Core based on Ubuntu 20.04 LTS

mgsanava, 2023-11-24:
I switched to Ubuntu 23.10 hoping that Wi-Fi driver will come prebuilt with no dropping issues, which is not the case.
I also swicthed to mainline kernel 6.6.1. Wi-Fi was still dropping.

_**Then I understood that HP is complete sh!t of a laptop for Linux.**_

**The amazing file that solved my issue:**
`sudo nano /usr/lib/modprobe.d/70-rtw89.conf`

**What's inside the file:**
`options rtw89_pci disable_clkreq=y disable_aspm_l1=y disable_aspm_l1ss=y`

Wi-Fi was working absolutely fine after doing this.
I need to give credits to a guy who posted this somewhere online which I did not bookmark.
Note that I had a similar file in /etc/modprobe.d/ previously which did not solve the issue.