lwfinger/rtw89 #264: rtw89_8852be: bluetooth doesn't work on archlinux fresh install kernel 6.4.10-arch1-1

AnakTeka, 2023-08-12:
Hi,

I have `09:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device b852` and I'm trying to connect my bluetooth headphone to the computer but apparently the bluetooth is not being detected by KDE.

The realtek card:
```
09:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device b852
	Subsystem: AzureWave Device 5471
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 104
	IOMMU group: 19
	Region 0: I/O ports at c000 [size=256]
	Region 2: Memory at fc800000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=375mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [50] MSI: Enable+ Count=1/1 Maskable- 64bit+
		Address: 00000000fee00000  Data: 0000
	Capabilities: [70] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s <4us, L1 <64us
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 26W
		DevCtl:	CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop- FLReset-
			MaxPayload 256 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr+ TransPend-
		LnkCap:	Port #0, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <4us, L1 <64us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM L1 Enabled; RCB 64 bytes, Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 2.5GT/s, Width x1
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Not Supported, TimeoutDis+ NROPrPrP- LTR+
			 10BitTagComp- 10BitTagReq- OBFF Via message/WAKE#, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- TPHComp- ExtTPHComp-
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR+ 10BitTagReq- OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkCap2: Supported Link Speeds: 2.5GT/s, Crosslink- Retimer- 2Retimers- DRS-
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
		LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- Ad
vNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Capabilities: [148 v1] Device Serial Number 00-e0-4c-ff-fe-88-52-01
	Capabilities: [158 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [160 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=30us PortTPowerOnTime=60us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_C
ommonMode=0us LTR1.2_Threshold=284672ns
		L1SubCtl2: T_PwrOn=150us
	Kernel driver in use: rtw89_8852be
	Kernel modules: rtw89_8852be
```

dmesg attached
[result_dmesg.txt](https://github.com/lwfinger/rtw89/files/12327722/result_dmesg.txt)

lsmod:

```
[user@host ~]$ lsmod | grep rtw
rtw89_8852be           12288  0
rtw89_8852b           360448  1 rtw89_8852be
rtw89_pci              86016  1 rtw89_8852be
rtw89_core            622592  2 rtw89_8852b,rtw89_pci
mac80211             1527808  2 rtw89_core,rtw89_pci
cfg80211             1286144  3 rtw89_8852b,rtw89_core,mac80211
```

```
[user@host ~]$ sudo modinfo rtw89_8852be
filename:       /lib/modules/6.4.10-arch1-1/kernel/drivers/net/wireless/realtek/rtw89/rtw89_8852be.ko.zst
license:        Dual BSD/GPL
description:    Realtek 802.11ax wireless 8852BE driver
author:         Realtek Corporation
srcversion:     87FC3D1B28540DD0400027C
alias:          pci:v000010ECd0000B85Bsv*sd*bc*sc*i*
alias:          pci:v000010ECd0000B852sv*sd*bc*sc*i*
depends:        rtw89_pci,rtw89_8852b
retpoline:      Y
intree:         Y
name:           rtw89_8852be
vermagic:       6.4.10-arch1-1 SMP preempt mod_unload 
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        5E:D9:7D:20:B9:8A:BA:B3:93:B4:CA:A1:5C:34:5E:E5:69:18:40:85
sig_hashalgo:   sha512
signature:      30:64:02:30:4F:93:FA:BA:BD:68:F4:A9:B0:B7:2A:1A:B0:C8:0C:07:
                48:B6:DB:70:D8:E5:79:BA:AE:B8:47:B6:FF:56:E6:03:26:24:C7:58:
                90:35:3C:A5:1A:47:EC:FF:D6:EF:C0:AF:02:30:7C:B2:9E:FA:7C:99:
                F5:DF:9F:FD:59:D8:B7:D7:1E:56:66:A4:41:EF:C6:54:99:C9:BD:CB:
                7D:CE:B0:DB:CE:E1:C0:26:70:32:C2:9D:7F:A5:FC:F4:0F:81:18:C0:
```

lsusb:
```
[user@host ~]$ lsusb
Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 002: ID 0951:1666 Kingston Technology DataTraveler 100 G3/G4/SE9 G2/50
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 002: ID 046d:c52f Logitech, Inc. Unifying Receiver
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 25a7:fa70 Areson Technology Corp 2.4G Wireless Receiver
Bus 003 Device 004: ID 0b05:19af ASUSTek Computer, Inc. AURA LED Controller
Bus 003 Device 003: ID 258a:008a SINO WEALTH Bluetooth Keyboard
Bus 003 Device 002: ID 13d3:3571 IMC Networks Bluetooth Radio
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Let me know if there are additional logs that needs to be provided.

lwfinger, 2023-08-12:
First of all, let me reiterate that the wifi and Bluetooth devices are separate devices that are packaged together, and share a 2.4 GHz radio. Having a wifi driver loaded does nothing for BT. In addition, I know very little about BT.

Your dmesg shows that the BT device is recognized and that the firmware is being loaded. As far as I can see, BT should be functioning correctly. I see you have a BT keyboard. If that has a separate BT controller, perhaps there is some interference.

There have been reports of wifi power_save hurting BT output, but not connectivity. To check that, use the command

sudo iw wlan0 power_save off

Does that help?

One other thing to try: run 'bluetoothctl' and enter 'list'. Does that show a controller? If it does, enter 'scan on'. Does that show your headphones?

We have now exhausted my knowledge of connecting to BT. You will need to get info somewhere else.

AnakTeka, 2023-08-12:
Thanks a lot for the answer, after realizing that the BT firmware is loaded successfully based on your answer, I check the bluetooth service itself and its not loaded.

For reference for another reader, in my case, a simple `sudo systemctl enable bluetooth.service` fixed a non-existent BT device (in GNOME and KDE).

Although I can now connect to my Bluetooth headphones, the sound is inaudible unless I disconnect from my 5GHz Wi-Fi. Once disconnected, the sound is perfect.

closed 2023-08-12 by AnakTeka