lwfinger/rtw89 #265: resolve_btfids ：not found

inuyasha-660, 2023-08-14:
i use On Debian12,At the beginning, there was "skipping BTF generation xxxx to unavailability of vmlinux,and no error was reported ,but after i modprobe rtw_8852ae the wifi was still not work ,so i try to rm the vmlinux from the kernel. now ,As the title says, it reports resolve_btfids: notfound. 
how can i fix it , thank u

closed 2023-08-14 by inuyasha-660

lwfinger, 2023-08-14:
You removed a critical part of the kernel. I am surprised that it would still boot.

Reinstall your kernel AS SOON AS POSSIBLE!!!!!!!

The BTF messages have to do with including debug information in the kernel modules. They have nothing to do with your wifi problems. You need to check the log using 'sudo dmesg | less' to see what is happening.

inuyasha-660, 2023-08-15:
Thank you for you reply , i have reinstalled yesterday.