lwfinger/rtw89 #268: More stable than package in linux kernel 6.2.0-26-generic

Mblakey, 2023-08-19:
Not an issue per say, and apologies if this is the wrong place to put this question. 

After updating to 6.2.0-26-generic, Ubuntu 22.04 i kept having internet drops where the device would state *-network DISABLED on `lshw -C network`. This would occur maybe after 3 hours ish of successful connection, and a computer restart would fix the issue temporarily. The card i have uses 8852be. 

After removing the supplied driver through apt and using this one. I have not experienced a connection drop but unsure why? Are there differences between this driver and the supplied one in apt? 

I suspect after reading around that fundamentally this is a BIOS issue, was wondering what you think?

lwfinger, 2023-08-19:
With the in-kernel driver, you were using the driver from kernel 6.2.

This repo uses the driver from kernel 6.6, which will not be released for another 3 months! It is not a BIOS issue. If that were true, neither driver would work correctly.

What you are seeing is progress!!! :)

closed 2023-08-19 by lwfinger

Mblakey, 2023-08-20:
Ah amazing, thank you for the great work you've done and made open to the community! very much appreciate it!