lwfinger/rtw89 #269: weired issue on the latest kernel on fedora & build error in xanmod kernel

ghostinsidehost, 2023-09-03:
### I have been using this driver on fedora 36 for  1 year i had no issue thank you a ton for making this.
recently switched to **fedora 38 and on the 6.4 kernel after installing this the wifi works but after 2 minutes the wifi disconnecets and doesnt work, even after reboot the problem persists.**

**but on the 6.2 kernel works fine.**

Now I want to use the xanmod kernel https://copr.fedorainfracloud.org/coprs/rmnscnce/kernel-xanmod/

I tried the xanmod edge & lts kernel but I am facing build error please check the log. I am not a technical personal please co operate and help. 

```
[saikat@fedora ~]$ cd localrepo/rtw89/
[saikat@fedora rtw89]$ uname -r
6.0.8-xm1.0.fc38.x86_64
[saikat@fedora rtw89]$ make 
make -C /lib/modules/6.0.8-xm1.0.fc38.x86_64/build M=/home/saikat/localrepo/rtw89 modules
make[1]: Entering directory '/home/saikat/localrepo/rtw89'
make[1]: *** /lib/modules/6.0.8-xm1.0.fc38.x86_64/build: No such file or directory.  Stop.
make[1]: Leaving directory '/home/saikat/localrepo/rtw89'
make: *** [Makefile:96: all] Error 2
[saikat@fedora rtw89]$ 

```

saikatm, 2023-09-03:
@lwfinger please help sir

lwfinger, 2023-09-03:
If the make step cannot find /lib/modules/.../build, you have not installed the kernel headers. I had never heard of the xanmod kernel. Even though their websight says the kernel headers are being installed, they are not.

My guess is that you are getting conflicts between this driver and the one built into kernel 6.4 on Fedora. Install the xanmod kernel version 6.5, and do not use this repo. That in-kernel driver is very nearly the same as this repo.

ghostinsidehost, 2023-09-04:
@lwfinger the xanmod kernel 6.5 in not available for fodera and the available version is too old. also let's keep aside xanmod.

how to fix the driver conflict on latest feodra kernel? please help I'm not technical any logs needed i can provide let me know the steps 🙏🏻🙏🏻 my computer is unusable without this driver

lwfinger, 2023-09-04:
You need to create a file to blackllst the kernel version of your device. You have never told me what device you actually have, thus I cannot give you the exact commands.

Do a web search on blacklisting drivers and create a file that blacklists the rtw89_XXXXYe driver, where XXXX is the bumber of the device, and Y is the letter that follows the name.

saikatm, 2023-09-05:
### update:  blacklisted the built in module

```
/etc/modprobe.d/wifi.conf
# Do not load the 'rtw89_8852be' module on boot.
blacklist rtw89_8852be

```

now the **rtw_8852be** module loads and the wifi works few minutes but the same issue happens the wifi vanishes . 
should I try removing the built in kernel module **rtw89_8852be** instead of blacklisting it?? 

**please check this logs**:

[dmesg.txt](https://github.com/lwfinger/rtw89/files/12526448/dmesg.txt)
[lsmod.txt](https://github.com/lwfinger/rtw89/files/12526449/lsmod.txt)
[lspci-k.txt](https://github.com/lwfinger/rtw89/files/12526450/lspci-k.txt)

lwfinger, 2023-09-05:
Your lsmod output shows the correct modules are loaded.
From your dmesg output, the following two lines are surprising:

[  103.082430] rfkill: input handler enabled
[  104.840979] rfkill: input handler disabled

It seems that something tried to disable wireless and then stopped. We may want to disable the rfkill. Try a 
'sudo rfkill unblock all', or better yet, remove rfkill using dnf.

The next thing that shows in dmesg is
[  162.774649] rtw89_8852be 0000:01:00.0: failed to pre-release fwcmd

That failure destroys the driver and leads to all the messages that follow. I would like to get rid of the rfkill anomaly before we tackle this one.

ghostinsidehost, 2023-09-05:
`sudo rfkill unblock all` did nothing the problem persist. 
can't uninstall rfkill as it may break gnome shell!? 

I don't understand why the built in driver loads even though I blacklisted it. 

also if the driver is broken ehy does it includes in the linux kernel? 

what to do next should I provide more logs or anything to do in the BIOS? 

please help me 😭😭😭

lwfinger, 2023-09-05:
The built-in driver is not loaded. Do not pay any attention to the error messages. They say rtw89_8852be even though rtw_8852be is loaded. Look at the output of 'lsmod | grep rtw'.

The in-kernel sdriver is just older than the one we have in this repo. The in-kernel one is not broken, there have been improvements.

Getting rid of rfkill will not break gnome. I booted a Fedora VM and found a command that works:
sudo mv /usr/libexec/gsd-rfkill /usr/libexec/gsd-rfkill.save

That will disable rfkill.

closed 2023-10-02 by lwfinger

ghostinsidehost, 2023-10-02:
sorry for late replay I used ubuntu and this [patch](https://github.com/lwfinger/rtw89/issues/240#issuecomment-1519118474)  fixed it. 
also I booted the latest fedora 39 beta and wifi works OBB and had no issue after using for a long period of time.

I want to express my heartfelt gratitude for maintaining this repository. I'm not sure who created the driver in the Linux kernel, but I truly appreciate it. Thank you so much.

@lwfinger