lwfinger/rtw89 #270: rtl8852be : Activation of connection timeout

chetan453, 2023-09-26:
When I start my Linux the wifi is connected for around 2-3 minutes, but after a pop-up shows stating "Activation of connection timeout", and then all the Wi-Fi networks disappear.  I had dual boot and wifi driver is working fine in windows. 
`nmcli dev status` output
```
DEVICE           TYPE      STATE         CONNECTION         
enx36e6c04f935f  ethernet  connected     Wired connection 2 
wlo1             wifi      disconnected  --                 
virbr0           bridge    unmanaged     --                 
lo               loopback  unmanaged     --                 
p2p-dev-wlo1     wifi-p2p  unmanaged     --    
``` 
`lshw -C network` output
```
  *-network DISABLED        
       description: Wireless interface
       product: Realtek Semiconductor Co., Ltd.
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0
       bus info: pci@0000:01:00.0
       logical name: wlo1
       version: 00
       serial: 50:c2:e8:c2:72:ed
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress cap_list ethernet physical wireless
       configuration: broadcast=yes driver=rtw89_8852be driverversion=6.2.0-33-generic firmware=N/A latency=0 link=no multicast=yes wireless=IEEE 802.11
       resources: irq:77 ioport:f000(size=256) memory:fcf00000-fcffffff
  *-network
       description: Ethernet interface
       physical id: a
       bus info: usb@1:2
       logical name: enx36e6c04f935f
       serial: 36:e6:c0:4f:93:5f
       capabilities: ethernet physical
       configuration: autonegotiation=off broadcast=yes driver=rndis_host driverversion=6.2.0-33-generic duplex=half firmware=RNDIS device ip=192.168.130.29 link=yes multicast=yes port=twisted pair
```
What I had tried till now(that didn't works )

- I installed the updated version of rtw89 repository after uninstalling the old one.
- I deleted the wifi driver from the windows and then resets the PC setting to get the driver back. 
-  I also tried to restart the network-manager through `systemctl restart NetworkManager` in the hope that it would restart the wifi but it didn't works as I expected, no wifi networks were displaying after that. 

I had an HP pavilion laptop, and after dual boot their is no wifi driver in my linux system so I installed rtw89, but wifi drive is rtl8852be.
This is the details of my linux system
```
Linux chetan 6.2.0-33-generic #33~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Sep  7 10:33:52 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

lwfinger, 2023-09-26:
There is too little info here to know what is happening. After you get this failure, run the command 'sudo dmesg > dmesg.txt' and attach the file dmesg.txt to this issue.

The fact you have an HP laptop suggests a few other things to try. The BIOS in many late-model HPs is defective. As root, create a file named /lib/modprobe.d/70-rtw89.conf and add the following lines:
options rtw89core disable_ps_mode=y
options rtw89_core disable_ps_mode=y
options rtw89pci disable_clkreq=y disable_aspm_l1=y disable_aspm_l1ss=y
options rtw89_pci disable_clkreq=y disable_aspm_l1=y disable_aspm_l1ss=y

The Windows driver appears to work around these problems using techniques like this.

chetan453, 2023-09-27:
Here is the attached ouput of dmesg 
[dmesg.txt](https://github.com/lwfinger/rtw89/files/12739221/dmesg.txt)
Also I tried the adding of 70-rtw89.conf but this also didn't worked.

lwfinger, 2023-09-27:
Unfortunately, your dmesg output is useless as your buffer is too small, and it had been wrapped around after only 110 sec.

You will need to increase the buffer size at boot. When GRUB is just starting, you will need to edit the startup line and add "log_buf_len=131072" to the kernel command line and then start the boot process.

If you would prefer, you can edit (as root) file /boot/grub2/grub.cfg and find the line that starts with "linux" or "linuxefi", then add the above string to the end of the line.

chetan453, 2023-09-29:
The issue has been resolved by adding the lines in 70-rtw89.conf file. I see that today, the wifi is working fine. 
I had done this earlier and rebooted the system, which is getting connection timed out. After that, I did nothing extra, but now wifi is working fine. 
Thanks for your help.

closed 2023-09-29 by chetan453