lwfinger/rtw89 #271: Lenovo Thinkbook AMD CPU rtw89_8852be stopped working after a period of time

whao, 2023-09-28:
Here is the system info
```
                   -`                    whao@thinkbook-14 
                  .o+`                   ----------------- 
                 `ooo/                   OS: Arch Linux x86_64 
                `+oooo:                  Host: 21D0 ThinkBook 14 G4+ ARA 
               `+oooooo:                 Kernel: 6.1.55-1-lts 
               -+oooooo+:                Uptime: 22 mins 
             `/:-:++oooo+:               Packages: 1341 (pacman) 
            `/++++/+++++++:              Shell: zsh 5.9 
           `/++++++++++++++:             Resolution: 2880x1800 
          `/+++ooooooooooooo/`           DE: Plasma 5.27.8 
         ./ooosssso++osssssso+`          WM: kwin 
        .oossssso-````/ossssss+`         Theme: [Plasma], Breeze [GTK3] 
       -osssssso.      :ssssssso.        Icons: [Plasma], breeze [GTK2/3] 
      :osssssss/        osssso+++.       Terminal: kitty 
     /ossssssss/        +ssssooo/-       CPU: AMD Ryzen 7 6800H with Radeon Graphics (16) @ 3.200GHz 
   `/ossssso+/:-        -:/+osssso+-     GPU: AMD ATI Radeon 680M 
  `+sso+:-`                 `.-/+oso:    Memory: 4582MiB / 13662MiB
```
My issue is pretty similar to https://github.com/lwfinger/rtw89/issues/226 and https://github.com/lwfinger/rtw89/issues/240. I have used AUR to install the latest version and tested with or without modifying `/usr/lib/modprobe.d/70-rtw89.conf`
```
options rtw89_pci disable_clkreq=y disable_aspm_l1=y disable_aspm_l1ss=y
options rtw89pci disable_clkreq=y disable_aspm_l1=y disable_aspm_l1ss=y
```
while still the wlan reported errors after using a certain period of time
```
Sep 29 00:47:39 thinkbook-14 kernel: rtw89core: loading out-of-tree module taints kernel.
Sep 29 00:47:39 thinkbook-14 kernel: rtw89core: module verification failed: signature and/or required key missing - tainting kernel
Sep 29 00:47:39 thinkbook-14 kernel: rtw89_8852be 0000:03:00.0: loaded firmware rtw89/rtw8852b_fw-1.bin
Sep 29 00:47:39 thinkbook-14 kernel: rtw89_8852be 0000:03:00.0: enabling device (0000 -> 0003)
Sep 29 00:47:39 thinkbook-14 kernel: rtw89_8852be 0000:03:00.0: Firmware version 0.29.29.1, cmd version 0, type 5
Sep 29 00:47:39 thinkbook-14 kernel: rtw89_8852be 0000:03:00.0: Firmware version 0.29.29.1, cmd version 0, type 3
Sep 29 00:47:40 thinkbook-14 kernel: rtw89_8852be 0000:03:00.0: chip rfe_type is 1
Sep 29 00:47:40 thinkbook-14 kernel: rtw89_8852be 0000:03:00.0 wlo1: renamed from wlan0
Sep 29 00:47:40 thinkbook-14 NetworkManager[626]: <info>  [1695919660.5892] rfkill3: found Wi-Fi radio killswitch (at /sys/devices/pci0000:00/0000:00:02.2/0000:03:00.0/ieee80211/phy0/rfkill3) (driver rtw89_8852be)
Sep 29 00:59:44 thinkbook-14 kernel: rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 0
Sep 29 00:59:45 thinkbook-14 kernel: rtw89_8852be 0000:03:00.0: timed out to flush queues
Sep 29 00:59:45 thinkbook-14 kernel: rtw89_8852be 0000:03:00.0: timed out to flush pci txch: 0
```
I have to reload rtw module to make interface work again. I also tested with the mainline kernel but I got the same errors.
I read that it is caused by some buggying BIOS settings while I didn't find options that relate to low power PCI or something else :(. I have been experiencing this months though I can used wired connection :p. I wonder if I did anything wrong.

Here is my module files
```
➜  ~ ls -l /lib/firmware/rtw89/rt*
-rw-r--r-- 1 root root 223286 Aug 10 04:33 /lib/firmware/rtw89/rtw8851b_fw.bin.zst
-rw-r--r-- 1 root root 234465 Aug 10 04:33 /lib/firmware/rtw89/rtw8852a_fw.bin.zst
-rw-r--r-- 1 root root 226343 Aug 10 04:33 /lib/firmware/rtw89/rtw8852b_fw-1.bin.zst
-rw-r--r-- 1 root root 181184 Aug 10 04:33 /lib/firmware/rtw89/rtw8852b_fw.bin.zst
-rw-r--r-- 1 root root 250691 Aug 10 04:33 /lib/firmware/rtw89/rtw8852c_fw.bin.zst
```

lwfinger, 2023-09-28:
First of all, do a 'git pull' as there have been recent changes.

If that still fails, modify /usr/lib/modprobe.d/70-rtw89.conf and add these two lines:
options rtw89_core disable_ps_mode=y
options rtw89core disable_ps_mode=y

The late model HP and Lenovo laptops have been a lot of trouble as their BIOS seems to be exceptionly buggy.

kolbma, 2023-09-29:
I've a __Lenovo Ideapad 5 15ABA7__ and using __Linux 6.5.5__ with the official drivers I've to add 
```
options rtw89_pci disable_aspm_l1=y disable_aspm_l1ss=y
```
to not get broken __wifi__ after some network traffic until next reboot  
and
```
options rtw89_core disable_ps_mode=y
```
for the __bluetooth__ part. Without it my BT mouse is often laggy and from time to time a BT scan doesn't find anything any longer until reboot.

The `disable_clkreq=y` is not needed with my Ideapad 5.

@lwfinger Do you know if there will be more fixes to the official kernel to get the hardware to work without the driver options? And is it possible for you to give a more detailed explanation what might be the problem with the Realtek hardware in Lenovo devices? Thx

kolbma, 2023-09-29:
Ah, and you need at least __firmware__ version __0.29.29.3__.  
The older 0.29.29.1 also doesn't work with the driver options added.

closed 2023-09-29 by lwfinger

whao, 2023-10-03:
> Ah, and you need at least **firmware** version **0.29.29.3**. The older 0.29.29.1 also doesn't work with the driver options added.

Hey, how can I upgrade the firmware? I have `git clone` and manually installed the latest driver but the firmware number is still **0.29.29.1**.