lwfinger/rtw89 #274: Can't make for kernel version 5.10.0

ivansteezy, 2023-10-13:
I can't build the driver, when I try to `make` I got the following compile errors:

```
/rtw89/core.c: In function ‘rtw89_track_work’:
/rtw89/core.c:3022:2: error: implicit declaration of function ‘rtw89_tas_track’ [-Werror=implicit-function-declaration]
 3022 |  rtw89_tas_track(rtwdev);
      |  ^~~~~~~~~~~~~~~

/rtw89/core.c: In function ‘rtw89_core_start’:
/rtw89/core.c:3900:2: error: implicit declaration of function ‘rtw89_tas_reset’; did you mean ‘rtw89_hci_reset’? [-Werror=implicit-function-declaration]
 3900 |  rtw89_tas_reset(rtwdev);
      |  ^~~~~~~~~~~~~~~
      |  rtw89_hci_reset

/rtw89/core.c: In function ‘rtw89_core_init’:
/rtw89/core.c:4020:2: error: implicit declaration of function ‘rtw89_tas_init’; did you mean ‘rtw89_mac_init’? [-Werror=implicit-function-declaration]
 4020 |  rtw89_tas_init(rtwdev);
      |  ^~~~~~~~~~~~~~
      |  rtw89_mac_init
```
My specifications: 

[image]

SirMaik, 2023-10-14:
@ivansteezy Please let me know if you manage to fix this.

lwfinger, 2023-10-14:
I have now fixed the problem. Do a 'git pull' and try again.

closed 2023-10-14 by lwfinger

ivansteezy, 2023-10-14:
Thanks @lwfinger  that did the fix.